pybricks/support #147: [Question]

KoenTimmermans, 2020-10-12:
**Question**
Can the NXT IRLink Sensor (NIL1046) and NXT Acceleration / Tilt Sensor (NAC1040) work well with PyBricks and EV3?

**Context**
The NXT IRLink Sensor (NIL1046) and NXT Acceleration / Tilt Sensor (NAC1040) from HiTechnic are listed on the page below as supported. However, on the website where I can buy it I come across the following: "Although theoretically compatible, latency and loss of information can occur when using this sensor with an EV3 brick."

I know the sensors were originally released for the NXT, but is the potential problem related to the standard EV3 software or hardware? In other words, could ev3dev / PyBricks be able to handle this?

The page I referred to: http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensors.html

An additional question, a bit more off topic, but does anyone happen to know whether the NXT Acceleration / Tilt Sensor (NAC1040) indeed functions as a LEGO gyro sensor, but in 3 axes?

**Screenshots**
Does not apply.

Thanks in advance for the response.

dlech, 2020-10-12:
We don't have specialized sensor classes for 3rd party sensors in Pybricks like we do for the official LEGO sensors. But any sensors in the list you linked can be used with the [Ev3dev sensor class](https://pybricks.github.io/ev3-micropython/iodevices.html#ev3dev-sensors). Some of these sensors have issues with the official LEGO EV3 firmware for EV3, but tend to work better with ev3dev. In fact, you can have full control over I2C devices with the [I2CDevice class](https://pybricks.github.io/ev3-micropython/iodevices.html#i2c-device).

The NAC1040 only measure acceleration, not rotational velocity, so it is not like the EV3 gyro sensor at all.

KoenTimmermans, 2020-10-12:
Thanks for the quick response! Nice to hear that the sensors can be used with ev3dev.

As for the NAC1040, I get it. I was thinking about a balancing robot and I would think that with the help of measuring acceleration on three axes it should work. I also read that the sensor can measure a slope from the horizontal, uing gravity. I think that should be usable for a segway principle as well.

dlech, 2020-10-12:
Don't forget that gravity is not the only force acting on the robot. The angle derived from the accelerometer will only be correct if the robot is at rest or moving at a constant velocity.

KoenTimmermans, 2020-10-12:
Thanks for that insight. I should have thought about that better. In that case the case becomes more difficult. Perhaps it is better to use the EV3 gyro sensors then, one per axis.

dlech, 2020-10-12:
FYI, Mindsensors also makes a [3-axis gyro](http://www.mindsensors.com/ev3-and-nxt/15-gyro-multisensitivity-accelerometer-and-compass-for-nxt-or-ev3) if you need more one axis. But simple segways only have one axis of rotation.

KoenTimmermans, 2020-10-12:
I know, thank you. I saw someone's NAC1040 and NIL1046 for sale for a good price, so that's why. ; )

Thanks for your help!

closed 2020-10-12 by KoenTimmermans