pybricks/support #148: I have two computers with two very different behaviors.  The easiest to describe is my desktop:

dlech, 2020-10-13:
I have two computers with two very different behaviors.  The easiest to describe is my desktop:
Lenovo Razor ideacentre Y900 RE-341SZ Signature Edition
Windows 10 Home version 2004 build 19041.508
Chrome browser Version 85.0.4183.121 (Official Build) (64-bit)

When I go to code.pybricks.com and hit <CTRL> Refresh to be sure I have the latest firmware, and try to download to my City Hub, I get the message "No compatible devices found."

I've tried to pair the City Hub outside of Pybricks by just using the Windows "Add Bluetooth Device" utility, and the City Hub is never seen.  So it's like my internal Bluetooth radio isn't compatible with the City Hub.  When I look at Bluetooth in Device Manager, there are a lot of things listed and I can't tell which of them is the internal Bluetooth radio.

I'll list my laptop behavior in #136 because it's similar to things discussed there.

_Originally posted by @Walt-White in https://github.com/pybricks/support/issues/48#issuecomment-707423733_

dlech, 2020-10-13:
Are you trying to download and run a program or trying to update the firmware? For updating the firmware, are you following the instructions here? https://docs.pybricks.com/en/latest/start_pup.html#installing-the-pybricks-firmware

Walt-White, 2020-10-13:
Thanks for the quick response.  Both running a program and updating the firmware fail.  Both report "No compatible devices found".  I'm following the quoted instructions and never see the LEGO Bootloader as a device as described in the 7th bullet.  Only "No compatible devices found".

I'm using Bluetooth for a mouse the whole time, so my desktop Bluetooth radio is working.  I've successfully paired multiple EV3 bricks.  The City Hub just never appears.  

I've been successful with the City Hub on my laptop and have run programs, but now it fails, as well.  So I know the Hub is working.  The laptop has more details from the Chrome console, which I will post sometime tomorrow morning.

laurensvalk, 2020-10-13:
> I'm using Bluetooth for a mouse the whole time, so my desktop Bluetooth radio is working. I've successfully paired multiple EV3 bricks. The City Hub just never appears.

EV3 uses Bluetooth classic. The hubs all use Bluetooth Low Energy, which is somewhat newer. Could it be that your desktop has only Bluetooth classic? If so, perhaps you could try a USB Bluetooth Low Energy dongle.

@dlech, if this is the case, are we able to detect this so we can notify the user?

dlech, 2020-10-13:
> @dlech, if this is the case, are we able to detect the presence/absence of a BLE radio so we can notify the user?

It is on the TODO list.

Walt-White, 2020-10-13:
@laurensvalk That's it!  The computer is only two years old but has an antique version of Bluetooth that Microsoft reports as "withdrawn".  I'll try the dongle.  I wound up using a dongle on my laptop to connect to two City Hubs when I was programming a train bypass using the Brick Automation Project.  I'll dig it out and use it on my desktop.
Thanks much.

Walt-White, 2020-10-14:
@dlech I'm able to connect to the City Hub now.  

Current stats:
Lenovo Razor ideacentre Y900 RE-341SZ Signature Edition
Windows 10 Home version 2004 build 19041.572
Broadcom BCM20702 Bluetooth 4.0 USB Device - Firmware Version: HCI 7.5860 / LMP 7.8718
Chrome browser Version 85.0.4183.121 (Official Build) (64-bit)

Now I can pair and connect with the City Hub and download the latest firmware.

But it still fails when I run a program:
`from pybricks.pupdevices import ColorDistanceSensor
from pybricks.parameters import Port, Color
from pybricks.tools import wait

# Initialize the sensor.
sensor = ColorDistanceSensor(Port.B)
print("Hello Pybricks")
# Repeat forever.
while True:

    # If the sensor sees an object nearby.
    if sensor.distance() <= 5:

        # Then blink the light red/blue 5 times.
        for i in range(5):
            sensor.light.on(Color.RED)
            wait(30)
            sensor.light.on(Color.BLUE)
            wait(30)
    else:
        # If the sensor sees nothing
        # nearby, just wait briefly.
        wait(10)`

The logs describe a GATT Server problem, and when I searched Issues I found someone else reporting the exact same problem.  You asked that person to generate and submit Bluetooth logs which I've done and will try to drag & drop here.

Thanks for your (and laurensvalk's) help.  It really makes a difference.
Walt
[bth_hci.zip](https://github.com/pybricks/support/files/5381411/bth_hci.zip)

dlech, 2020-10-15:
I forgot... this is a know issue with the City hub: https://github.com/pybricks/support/issues/48#issuecomment-706463541

dlech, 2020-10-15:
If you don't want to wait a few weeks for the next update, you can download [this file](https://github.com/pybricks/pybricks-micropython/actions/runs/303953717) and click and drag it onto the firmware update button in code.pybricks.com (don't unzip the file).

Walt-White, 2020-10-15:
It's working.  I'm using the version command to print the firmware version at the start of my program.  My other problem was a misunderstanding of the distance() function.  I was used to the WeDo 2.0 projects I did in a classroom where the phone/tablet app displayed the distance in centimeters.  

Now with the ColorDistanceSensor, I was using a distance of 2 or 3 which is how far it is in centimeters to the Great Ball Contraption balls I'm sensing.  I read the fine manual on Signals and Units and found that it's [relative, not specific](https://docs.pybricks.com/en/latest/signaltypes.html#relativedistance).  Using the example value of 40, it works.

mwinkler, 2020-11-03:
First of all, thank you for all the hard work you do with this project.

I am also able to connect to the city hub now.
For a first test, I plug a train motor in port A and load the example 1 program from [Motors without Rotation Sensors](https://docs.pybricks.com/en/latest/pupdevices.html#motors-without-rotation-sensors).
The code editor prints following error:
```
Traceback (most recent call last):
  File "main.py", line 7, in <module>
OSError: [Errno 19] ENODEV: 

A sensor or motor is not connected to the specified port:
--> Check the cables to each motor and sensor.
--> Check the port settings in your script.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.
```
I tried also the other port, but always the same error.

A press on the connect button, prints the following
```
('cityhub', '3.0.0a9', 'v3.0.0a9-25-g2541b8e4 on 2020-10-13')
Hello, World!
```
Regards

laurensvalk, 2020-11-04:
Thanks @mwinkler. Glad to hear you can connect to the hub now. We moved your other question to a dedicated issue so we can better help you.

dlech, 2020-12-11:
This has been resolved.

closed 2020-12-11 by dlech