pybricks/support #149: [Project] Small Line Follower with Prime Hub and mixed Powered Up elements

JorgePe, 2020-10-13:
**Description**

Just what the title says: a very small line follower with very few pieces to demonstrate Pybricks' potential
I flashed my Prime Hub from the 51515 Robot Inventor set with v3.0.0a9 so it now does dual boot and I can use pybricksdev with it through the USB cable.

Since I didn't want to disassemble my current 'Blast' configuration, I used just the hub with a few other pieces I had available - particularly 2 large Technic motors (identical to the large motors from the SPIKE set) and 1 Color/Distance sensor from the BOOST set.

[image]

**Source Code**

Code is very simple and quick tuned for running over a common sheet of paper. At this early stage of pybricks support for the Prime Hub, I have to run the code in thethered mode so a small closed circuit is enough for demonstration with a short USB cable connected to my laptop.

```python
from pybricks.hubs import PrimeHub
from pybricks.parameters import Port, Color
from pybricks.pupdevices import Motor, ColorDistanceSensor
from pybricks.tools import wait

hub = PrimeHub()
mL = Motor(Port.E)
mR = Motor(Port.F)
s = ColorDistanceSensor(Port.A)

SPEED = 40    # duty cycle applied to motors (percentage)
TIME = 2         # time executing each step
THRESHOLD = 50    # threshold value for detecting the line
LIMIT = 5   # number of times a deviation is allowed before stepping a bit back

count = 0   

while True:
   # move forward
    mL.dc(-SPEED)
    mR.dc(SPEED)
    wait(TIME)
    v = s.reflection()
    print(v)
    if v < THRESHOLD :
        # move left
        mL.dc(SPEED)
        mR.dc(SPEED)
        count = 0
    else :
        if count < LIMIT :
            # move right
            mL.dc(-SPEED)
            mR.dc(-SPEED)
            count = count +1
        else:
            # move back to prevent deviation
            mL.dc(SPEED)
            mR.dc(-SPEED)
            wait(3*TIME)
            count = 0
    wait(2*TIME) 
```

**Building instructions**
I believe there is no need for instructions as anyone can build this toy in less than 5 minutes

laurensvalk, 2020-10-14:
Thanks for sharing. I think you are the first user in the world to dual boot the Inventor Hub! 

It's cool to see that these motors are already working as well - I don't have them.

If you like, here are some ideas for further exploration:
- If you do `mL = Motor(Port.E, Direction.COUNTERCLOCKWISE)`, positive speeds/duty will make it go forward too, making the rest of the script easier to interpret
- You could try `mR.dc(SPEED + steering)` and `mL.dc(SPEED - steering)` with `steering = (v - THRESHOLD)*KP`

Disconnecting USB used to work, I think. We'll have to try that again. We might even change the way USB works after we enable Bluetooth. At that point, it could work just the like other hubs, leaving the USB free for other things like fast file transfer. We're not sure yet, so feel free to share your wishes and feature requests :)

JorgePe, 2020-10-14:
Well you already said BT - using the Chrome IDE would be good for beginners since using text editor and pybricksdev might scare them. For me BT would also have the advantage of using the Nordic UART (if supported on this firmware) to cooperate with other systems until full messaging is available.

Can't remember any particular feature I might need... I am still trying to figure how to use most of the API like Grid and Speaker (I think still missing?). Generally speaking, having support for all internal devices of the PU family hubs would be great but it's already on the roadmap.

Thanks for the tips, will try them.

laurensvalk, 2020-10-14:
Speaker is not implemented yet, but we're investigating to include (very basic) speech synthesis :nerd_face: .

There is some experimental IMU functionality in there already, so have a look at the `experimental` module, but we're going to have to rewrite it entirely to make it non-blocking before we will officially support it.

JorgePe, 2020-10-14:
I already found the experimental IMU, had been using it with Control+ HUB for a wristband-based HID controller - sending the output of the IMU to a "master" device on a bakcpack where it calculates the orientation of the wrist. But my math is very rust and only implemented 2 DOF.

TechnicBRICKs, 2020-10-14:
Have also tried to install the Pybricks firmware too, but regardless of not so smooth feedback progress at the console and the success message in the end, it always end booting with the LEGO firmware.

laurensvalk, 2020-10-14:
That's a good sign! It means the update was successful.

This feature is not ready for average user yet, though. It's more like a proof of concept for now :)

Please use the Technic Hub for real projects for now. Switching over will be easy later since motors and sensors work the same across all hubs.

TechnicBRICKs, 2020-10-14:
Yeah! Not a real project, just wanted to give it a try! :)

laurensvalk, 2020-10-14:
It's great to have your feedback, we appreciate it. This gives us more confidence in the procedure. It works just like the official update procedure, but still :) There's already a version on the way which will show progress using the pixels on the screen. When you reach 100%, all lights will be on.

As you have found, turning on the Pybricks firmware still need some work. The reason is this: by the time we check the button state, the power is not on yet, so it cannot read the button state.

JorgePe, 2020-10-19:
Small update: thanks to a Laurens suggestion the robot no longer needs to be tethered so I can now let it run full speed on the floor.
I made some minor adjustments to the code so it now uses proportional control (and the code is also much simpler)

```
# with Laurens Valk tips
# for Proportional control and COUNTERCLOCKWISE option

from pybricks.hubs import PrimeHub
from pybricks.parameters import Port, Color, Direction
from pybricks.pupdevices import Motor, ColorDistanceSensor
from pybricks.tools import wait

hub = PrimeHub()
mL = Motor(Port.E, Direction.COUNTERCLOCKWISE)
mR = Motor(Port.F)
s = ColorDistanceSensor(Port.A)

SPEED = 95
TIME = 1
THRESHOLD = 55
KP = 2.5

# time to go downstairs - waiting for a button would be much clever
wait(60000)

while True:
    wait(TIME)
    v = s.reflection()
    print(v)
    steering = (v - THRESHOLD) * KP
    mR.dc(SPEED + steering)
    mL.dc(SPEED - steering)
```

not sure about that "wait(TIME)" line... forgot to comment it but it probably works fine without it.
Had to adjust THRESHOLD a bit because the color/distance sensor was detecting the separation on my white floor tiles as lines and following them.

laurensvalk, 2020-10-19:
Nice and clean :smile:  Do you want to share the video link as well?

By the way, here's a another trick you can use to wait for any button:
```python
# wait for any button to be pressed
while not any(hub.buttons.pressed()):
    wait(10)
```

JorgePe, 2020-10-19:
I thought you would  never ask :D 
https://youtu.be/oHwqm5GqPTU

TechnicBRICKs, 2020-10-19:
Do you have new app script to flash the FW that supports unthetered
operation?

Best,
Fernando Correia
Sent without an iPhone... 😁

JorgePe, 2020-10-19:
No, the firmware is the same.
I edited 2 lines in "connections.py" in the pybricksdev tool, it caused connection to keep stablished.
I can tell you the lines to comment but I think it is better if you wait for a new release of the pybricksdev took, then you just updated it fully with a single command

JorgePe, 2020-10-19:
Since you you are already using REPL, you can copy&paste the whole script into it.

JorgePe, 2020-10-19:
yes it did.
The "pybricksdev run" command now ends and releases the connection:
```
Connecting to /dev/ttyACM0
2020-10-19 10:26:12,923:    INFO: Sending: 17%
2020-10-19 10:26:13,027:    INFO: Sending: 33%
2020-10-19 10:26:13,130:    INFO: Sending: 50%
2020-10-19 10:26:13,233:    INFO: Sending: 67%
2020-10-19 10:26:13,336:    INFO: Sending: 83%
2020-10-19 10:26:13,439:    INFO: Sending: 100%
2020-10-19 10:26:13,544:    INFO: Disconnected by client.
```
before the change, with our without the "print" command, i never saw that last line.

JorgePe, 2020-10-19:
but before the change, I could re-run it again (as long as never removed the USB cable).
Now it gives me a checksum error, I need to first restart the hub.

closed 2020-12-16 by dlech