pybricks/support #150: [Doc] Pybricks Exception Index?

devinbreise, 2020-10-14:
Is there a list somewhere of the exceptions that can be raised by Pybricks?  We've noted that trying to create an EV3 device will throw an uerrno.ENODEV and misuse of the Motor/Drivebase APIs can result in a uernoo.EPERM.  Are there others?

Thank you

laurensvalk, 2020-10-14:
Exceptional point, thanks for raising it.

laurensvalk, 2020-10-14:
A list of errors sounds like a good idea. I think we can also look at adding the possible exceptions to each method in the docs. I've refrained from doing so thus far to keep the docs concise without making them too complicated for beginners.

One alternative would be to make the error messages themselves more explicit, but this would require more firmware space which (apart from EV3) we don't have much of, so it is probably going to be in the docs.