pybricks/support #153: Time Machine

kueden, 2020-10-19:
**Description**
A Tellurion depicts how day, night, and the seasons as well as the lunar phases are caused by the rotation and orientation of earth on its axis and its orbit around the sun. The Time Machine is a further modification of this meachanism to introduce the full Powered up functionality. For further information on the astronimical details visit Tellurion-II.
The Technic Hub is way more than a battery box for remote control. The Powered Up Hub can be converted into an autonomous controller that does no longer require a computer or mobile device for operation.
This makes the Tellurion a Time Machine:
-Input time and date present on the Tellurion into the contol software (20.6.2020 21:44 when built according to the lxf file)
-Set the desired time and date in the past or the future ( usually you would choose your present time)
-Start the software and the model will run as long as the batteries last (for increased operation time use the 9V Adapter)

**Video**
https://youtu.be/e-JSfwyZt6c

**Source Code**
https://www.bricksafe.com/files/kueden/time-machine/TMrun.txt

**Building instructions**
https://rebrickable.com/mocs/MOC-54177/kueden/time-machine/#details

laurensvalk, 2020-10-19:
This is very cool! Thanks a lot for sharing both the building instructions and the code.

Since Pybricks is still in beta, we'd love to hear more about your experience with it. Were there particular features you really liked for this model? Were there perhaps things that didn't work very well yet? Did you find anything missing?

kueden, 2020-10-20:
Thanks for your feedback! In my view the code.pybricks.com page works very well. I did not attemt to save the code via that site since the description indicated some issues. But simply copy/pasting it took perfectly well care of it. There is not really a compaint from my side. Whenever I ran into issues, they turned out to be linked to my limited competence in Python. 

I am having fun with Mindstorms EV3 for about 4 years now. I find the possibilities absolutely thrilling. However, the application of the EV3 code to the Powered Up has in my view the highest potential for the following reason:
With mindorms it used to difficult or impossible to include the EV3 Brick into existing models and the fact of having incompatible motors for Power Functions and Mindstorms was a nuisance as well. The latter is now overcome by the new Mindorms version that is on the market right now. It uses the same motors, the same connections and is fully compatible.

With Powered Up any Technic Model with Control+ is instantaneously converted into an autonomous robot with the help of your tools. This is totally cool. At present I see the following limitations, that you guys could probably easiliy overcome:

1. The blinking blue light when the bluetooth connection is lost is anoying. Essentially this forced me to remove the Time Machine from the living room  ;-)
This issue is already listed and well known to you guys.
2. The Technic Control+ Hub has a three axis gyrosensor. Connecting an external one to the already limited 4 inputs decreases the possibilites dramatically. If you could make those internal gyrosensors accessible by code would be extremely cool
3. I could not figure out how to integrate bluetooth connection for remote control into the code. So we are looking at the same issue as with the EV3: Either bluetooth remote control without any brains or full funcionality by code combined with an IR remote control only. Making the bluetooth connection possibly by your code would be extremely helpful in converting models into robots. It turns out that autonomous robots are nice, but the play factor is way higher in the case of Python augmented remote control. In this case the computer takes care of all the tidious anoying work and the operator gets all the fun part.

These are not really issues, and I am perfectly happy with what you are providing to us at present. Nevertheless, adding this functionality would dramatically increase the potential of your tools. If you consider these issues worthwile for listing as such, please let me know. I can happily do that. However, I do not want to steal any of your time.

laurensvalk, 2020-10-20:
> With Powered Up any Technic Model with Control+ is instantaneously converted into an autonomous robot with the help of your tools. This is totally cool. 

This was exactly our goal, so it's great that people are now starting to use it.

Thanks for the insights on point 3 --- this is still in the exploration stage. When you talk about remote control augmenting autonomy, do you mean one hub communicating with another? This is indeed the direction we'd like to go in.

Have you seen the [messaging](https://docs.pybricks.com/en/latest/messaging.html) module for EV3? Does this meet the requirements for your use case? We're thinking of doing the same for the Powered Hub hubs eventually.

kueden, 2020-10-20:
Indeed I am aware of the messaging function and that will actually take care of my comment 3. In particular converting my soon jobless EV3 Bricks into remote controls would be a nice option.

Just to give you an idea with what I mean with Mindstorms augmented remote control: The [TWIST3R](https://youtu.be/vwqFy5J-gqA) and the [SPINN3R](https://youtu.be/DTtD5tVK7aU) are impossible to manually control. But with the help of Mindstorms they turn into fun vehicules. Obviously they could be operated with the Technic Control+, but the limited number of ports prevents me to add Gyro Sensor or IR-Receiver. Therefore profiting from bluetooth and the internal Gyro Sensor would be fantastic. 
Again these are just inputs and I do not want to push you in any directions.

ZPhilo, 2020-10-20:
Awesome! Also love your (almost purist) 9V adapter for C+!!!

kueden, 2020-10-20:
Thanks, indeed I am very happy with it.

closed 2020-10-23 by kueden