pybricks/support #155: [Feature] Transfer focus to I/O window after starting program

laurensvalk, 2020-10-23:
**Is your feature request related to a problem? Please describe.**
When pressing the run button in Pybricks Code, the code runs and the cursor/keyboard isn't focused in any pane in particular.

**Describe the solution you'd like**
It would be nice to have it focused in on the standard i/o pane, so that programs that use `input()` or `getchar()` #154  work right away.

**Describe alternatives you've considered**
Manually clicking on it every time.

JorgePe, 2020-10-23:
Was thinking the same

dlech, 2020-12-11:
Will this confuse users who don't use the terminal for input? They might expect if they start typing that what they are typing would still be in the code pane (although the cursor would be gone which gives some hint).

Should the program stopping also switch focus back to the code pane?

JorgePe, 2020-12-12:
It might confuse. Is it possible to define it as an advanced setting, disabled by default?

laurensvalk, 2020-12-12:
Well, the cursor isn’t currently focused in the editor after clicking run either. It would be nice if it was though.

There’s another problem, which is that the download and run logic is affected by the I/O window for now. So, let’s not change focus to the terminal for now.

closed 2020-12-12 by laurensvalk