pybricks/support #156: LPE RC car

Dadibom, 2020-10-23:
Hey everyone, I'm currently working on my childhood dream project: an lpe (lego pneumatic engine) powered rc car.

I just got the new mindstorms set because I thought it would solve the issues I had with the control+ hub, but I have been disappointed. There seems to be no bluetooth functions in the official mindstorms app.

I am looking to control several functions of the car with a bluetooth gamepad. With the control+ hub I could barely even get the steering working smoothly. In the end I had to write my own PID controller with node-poweredup but since the logic runs on my computer, the performance is still not top notch (and i would need another fast acting servo motor for the throttle). Sending any more than 10-20 commands per second causes the control+ hub to go crazy, then freeze.

What I am looking for now is a way to control my motors as fast and smooth servos using micropython. Ideally the controller would connect straight to the hub but I am fine with the hub just receiving bluetooth input from my laptop. The controls need to be analog so I can't really use the keyboard thing posted here recently.

TLDR:
Any pointers to how I can use bluetooth communication in python programs running on the mindstorms hub would be appreciated.

laurensvalk, 2020-10-23:
In addition to the [online IDE](https://www.code.pybricks.com/) we provide a [Python package](https://github.com/pybricks/pybricksdev) called `pybricksdev` to download and run scripts on the hub, and communicate with it once it is running.

So your setup could work if you don't mind that your gamepad is connected to your computer. 

We've recently simplified the process of sending things to the hub in #154 . Combined with `pybricksdev`, your project might work really well in Jupyter Notebook. As it happens, we were in the process of updating these demos [https://github.com/pybricks/pybricksdev-demo](https://github.com/pybricks/pybricksdev-demo) for use with the new messaging. We'll need a few days for that though.

In the mean time, if you install `pybricksdev`, and if you can work out how to read the buttons of your gamepad in regular Python on your PC, I can give you some more pointers after we update the examples.

Dadibom, 2020-10-23:
Thanks. Unfortunately `poetry install` fails.

```
 • Installing twisted (20.3.0): Failed

  EnvCommandError

  Command ['/home/david/pybricksdev/.venv/bin/pip', 'install', '--no-deps', '/home/david/.cache/pypoetry/artifacts/22/83/81/8c46054ce174600810f2224fc2616421aa0cf88c93d0d1e3d9807f0c9b/Twisted-20.3.0.tar.bz2'] errored with the following return code 2, and output: 
  Processing /home/david/.cache/pypoetry/artifacts/22/83/81/8c46054ce174600810f2224fc2616421aa0cf88c93d0d1e3d9807f0c9b/Twisted-20.3.0.tar.bz2
  ERROR: Exception:
  Traceback (most recent call last):
    File "/home/david/.pyenv/versions/3.8.2/lib/python3.8/tarfile.py", line 1691, in bz2open
      from bz2 import BZ2File
    File "/home/david/.pyenv/versions/3.8.2/lib/python3.8/bz2.py", line 19, in <module>
      from _bz2 import BZ2Compressor, BZ2Decompressor
  ModuleNotFoundError: No module named '_bz2'
```

Dadibom, 2020-10-23:
Got it installed now. Had to uninstall python, install libz2-dev and then reinstall python.

Dadibom, 2020-10-23:
Could you please confirm if the mindstorms hub uses the same firmware as spike prime before i reflash? Wouldn't want to brick my $400 brick by reflashing... :D @laurensvalk

Dadibom, 2020-10-23:
Oof, the link in the pybricksdev readme to original firmware files is broken.. this is a bit scary

dlech, 2020-10-23:
I've installed Pybricks on an Inventor hub already. However, we currently don't have low battery protection, so you will need to be careful not to damage your battery by running it too low.

https://github.com/gpdaniels/spike-prime still has various firmware files. The files were just moved.

Dadibom, 2020-10-23:
Great, thanks. I see that gpdaniels has separate firmware files for spike and mindstorms, but I assume they use the same pybricks firmware?

dlech, 2020-10-23:
Yes, same Pybricks firmware.

Dadibom, 2020-10-23:
`ValueError: No DFU device found`
:(
Running ubuntu 18

dlech, 2020-10-23:
Did you set up udev rules? This is my `/etc/udev/rules.d/99-lego.rules`:

```ini
# SPIKE Prime hub in firmware update mode
SUBSYSTEM=="usb", ATTRS{idVendor}=="0694", ATTRS{idProduct}=="0008", SYMLINK+="spike-prime-dfu-%k", MODE="0666"

# SPIKE Prime hub
SUBSYSTEM=="usb", ATTRS{idVendor}=="0694", ATTRS{idProduct}=="0009", SYMLINK+="spike-prime-%k", MODE="0666"

# MINDSTORMS Inventor hub
SUBSYSTEM=="usb", ATTRS{idVendor}=="0694", ATTRS{idProduct}=="0010", SYMLINK+="mindstorms-inventor-%k", MODE="0666"

# MINDSTORMS Inventor hub in firmware update mode
SUBSYSTEM=="usb", ATTRS{idVendor}=="0694", ATTRS{idProduct}=="0011", SYMLINK+="mindstorms-inventor-dfu-%k", MODE="0666"
```

Dadibom, 2020-10-23:
I did not. Was that in some guide that i missed? Tried just adding your file and restarting and it still throws the same error unfortunately.

dlech, 2020-10-23:
It will take me some time to look into that. But as a workaround:

    sudo apt update && sudo apt install dfu-util
    dfu-util --list

If that lists the device, then we know the problem is in `pybricksdev`.

Dadibom, 2020-10-23:
without hub plugged in, it shows me nothing. when connected in flash mode, i get this:

```
Found DFU: [0694:0011] ver=0100, devnum=5, cfg=1, intf=0, path="1-1", alt=3, name="UNKNOWN", serial="325B38813338"
Found DFU: [0694:0011] ver=0100, devnum=5, cfg=1, intf=0, path="1-1", alt=2, name="UNKNOWN", serial="325B38813338"
Found DFU: [0694:0011] ver=0100, devnum=5, cfg=1, intf=0, path="1-1", alt=1, name="UNKNOWN", serial="325B38813338"
Found DFU: [0694:0011] ver=0100, devnum=5, cfg=1, intf=0, path="1-1", alt=0, name="@LEGO LES HUB /0x08000000/02*016Ka,02*016Kg,01*064Kg,07*128Kg/0x10000000/01*1Ma/0x10100000/31*1Ma", serial="325B38813338"
```

laurensvalk, 2020-10-23:
Wait, didn’t the original post mention Control+ (technic hub)? That’s what I would recommend for now. That’s the most user friendly to install at this point.

There’s no Bluetooth support for Pybricks on SPIKE and MINDSTORMS yet, so I don’t really see this being useful for this project just yet.

Dadibom, 2020-10-23:
yeah i did use a control+ hub before but the result wasn't very good so i bought the mindstorms hub today because of the advertised bluetooth support, which wasn't there :(

dlech, 2020-10-23:
Pybricks works on the Technic Control+ hub too. Just go to https://code.pybricks.com. You could use that for now and upgrade the the Inventor hub when we get Bluetooth working there. The same program should be easy to copy from one to the other with minimal changes.

Dadibom, 2020-10-23:
When i asked several weeks ago, bluetooth in pybricks didn't seem like it was gonna happen. That's why I got the mindstorms hub :(

Oh well, I guess I will at least have two extra ports available whenever support for that hub is added. I will have to convince myself that alone was worth $400 :joy: 

I'm like 90% there with gamepad handling so I'm eagerly awaiting the update :+1:

laurensvalk, 2020-10-23:
I went ahead and created a quick demo to try this. This already works with the latest `pybricksdev` and the latest firmware via #132 . This probably requires some explanation, but I thought I'd share this in case you're already quite familiar with Python.

This runs a script on the hub (cell 2 below) and an on the PC (cell 3). The PC generates a sine pattern (you could replace this by your gamepad) and the hub makes the motor track it. This works fairly smoothly on my system.

You'll probably still want to reduce the data flow to keep good performance. But in this setup you can balance what runs on the hub and what runs on the PC. All the real-time loops should probably run on the hub while you occasionally send some high level control signals from the PC. What's best will depend on your application.

[image]

laurensvalk, 2020-10-23:
I should also add that this is very much beta stuff you're seeing here, so you might find yourself occasionally taking the batteries out to try again. The stable interface will only be [Pybricks Code](https://www.code.pybricks.com/) for a while, but hackers are welcome to play with this extra stuff already :)

Dadibom, 2020-10-23:
Thanks @laurensvalk 
This is exactly what i was after. Just being able to send whatever data i want is perfect. Will try tomorrow!

Is the plan to allow communication between hubs too?

JorgePe, 2020-10-24:
> Is the plan to allow communication between hubs too?

Yes it is on the roadmap #29 but not for the short run

We started using 'output' and 'input' for some interesting things with laptop or even MINDSTORMS EV3 as 'clients' but now with 'getchar' being non-blocking (input waits for the 'client' to send a carriage return) things are easier but still not hub-to-hub, you will need a master device to coordinate them all

Dadibom, 2020-10-24:
@laurensvalk do we have to use notebook for this?
When running in notebook, the evdev module i used for bluetooth can't be found.  :(

Dadibom, 2020-10-24:
https://youtu.be/TxPCer8samw

Thanks everyone. It's now working!

laurensvalk, 2020-10-24:
> do we have to use notebook for this?

No you don't have to, but that's not related to the problem. Jupyter Notebook is mostly just an editor. And pretty convenient for this purpose.

> When running in notebook, the evdev module i used for bluetooth can't be found

`poetry` is kind of like `pip`, but it installs all dependencies locally in your project folder. If you prefer to install everything globally with `pip`, that's possible too, but that would be much more work to update when we make changes.

To add `evdev` to your project, just do:

```
poetry add evdev
```

laurensvalk, 2020-10-24:
> https://youtu.be/TxPCer8samw
> 
> Thanks everyone. It's now working!

Very nice!

And if you need ideas to make it smoother, feel free to open a dedicated issue for that.

The communication is probably already as fast as it gets, so then it might come down to changing a few parameters on the hub side. Depending on what you want to do, you can limit the acceleration, so that sudden changes to your signal smooth out.

Dadibom, 2020-10-24:
Ah, adding the package with poetry will certainly be cleaner than pip in the notebook.

I am currently limiting to 20 messages per second, will experiment with different intervals once i got the other motors set up too. I'll play with the pid values and acceleration once i got some load on the servo :)

Hoping that one day I can communicate directly between the mindstorms hub and controller so that latency will be reduced further!

Dadibom, 2020-10-24:
What is preventing us from using the ubluetooth module on the control+ or spike/mindstorms hubs?

laurensvalk, 2020-10-25:
Nothing technical, just a lot of work and time, and some design aspects.

It's not as trivial as simply enabling a MicroPython module. In case of SPIKE/MINDSTORMS, bluetooth needs to be implemented first. It's a different chip than the one in Control+. We've got most of that step figured out by now, but there's some implementation work to be done here.

And we'll need to look at how to integrate it with the rest of Pybricks. We use MicroPython a bit differently than most MicroPython boards. On most of those, when you stop MicroPython, everything stops. In our case, we want the "system code" to keep running, so you can at least still run your next program without having to reset.

Dadibom, 2020-10-26:
Thanks for the clarification. Let me know if there's anything I can do to help. I am very keen on getting this to work on the mindstorms hub.

Would also be nice if the default telemetry was optional to reduce load.

laurensvalk, 2020-10-26:
Cool. We'll keep you posted and thanks for taking the time try these new features.

> Would also be nice if the default telemetry was optional to reduce load.

Which telemetry? At the moment, there is no telemetry while a program is running unless you `print` something.

Dadibom, 2020-10-26:
I was just referring to the spike telemetry. It constantly sends sensor data, but perhaps that is not the case with your firmware.

laurensvalk, 2020-10-26:
Ah, got it. We don't send any data at the moment, but if we ever do, making it optional sounds like a good idea :+1: .

Dadibom, 2020-10-26:
Sending the number `3` breaks the program. All other numbers between 0 and 255 seem to work fine.

[image]

laurensvalk, 2020-10-26:
Ah, I forgot to mention that. Yes, 3 is equivalent to sending CTRL+C, which stops the program.

This will be resolved once program run logic and messaging use separate “channels”. Right now, there is just a single serial input/output stream used for everything.

Dadibom, 2020-10-26:
Hehe okay, great :)

Dadibom, 2020-11-02:
I have now hit a wall regarding the number of io ports. Will either need to hook up a second control+ hub or use the technic large hub.
I assume the technic large hub is pretty far off atm so I'll focus on a second control+ hub for now.

I got three questions:
* Will the new communication api support sending data back to the computer in the same way?

* What do I need to do to connect two hubs using your method @laurensvalk ? I'm assuming I need to rename one of the hubs in some way and then do something like
```
hub1 = BLEPUPConnection()
await hub1.connect(await find_device("Pybricks hub 1"))
hub2 = BLEPUPConnection()
await hub2.connect(await find_device("Pybricks hub 2"))
```
Is this assumption correct? What do I need to do to rename the hub? Or do all hubs have some unique value like a mac address I can use instead?

* The PC script has a sleep call at the beginning. Having a too short sleep value causes the program to fail. Whenever I'm adding more code to the hub, I need to increase the sleep value. Is there some better way to do this? I'm thinking something along the lines of `await hub.waitUntilReady()`

laurensvalk, 2020-11-02:
> Or do all hubs have some unique value like a mac address I can use instead?

Yes. So print the result of `find_device` one time and then just use it directly like this 

```python
hub_2 = BLEPUPConnection()
await hub_2.connect('90:84:2B:4A:2B:75')
```

> Will the new communication api support sending data back to the computer in the same way?

If you mean the mailboxes in `pybricks.messaging`, then yes.

I've also just pushed something new you can use already. You can now read e.g. `hub_2.output`, which is a list of lines printed by the hub. So this provides an easy way of sending things back to the PC already.

> The PC script has a sleep call at the beginning. Having a too short sleep value causes the program to fail. Whenever I'm adding more code to the hub, I need to increase the sleep value. Is there some better way to do this? I'm thinking something along the lines of await hub.waitUntilReady()

Pretty close. Try this:

```python
await hub.wait_until_state(hub.RUNNING)
```

Pybricksdev has mainly been a testing tool thus far and hasn't been optimized for performance. Using the PC as the middle man might not get great performance at this point, so keep that in mind :)

closed 2020-12-16 by dlech