pybricks/support #157: [Project] The incomplete list of cool Pybricks Projects

laurensvalk, 2020-10-26:
This is an incomplete collection of awesome creations that users have made with Pybricks so far. Lots of inspiration for your next project! :bulb: :bulb: 

We don't currently have time to maintain a dedicated web page for this, but we don't want your creations to be forgotten either. There's already a few that I could no longer find. So for now, let's collect them here and see how it goes. They're listed in no particular order. Just post a comment below if you'd like your creation to be added, or if you found a project online that's worth sharing.

# Official sets - now with Pybricks
- @albant: [8094 Plotter](https://www.youtube.com/watch?v=gLFPl4FGzL8)
- @jorgepe: [Hedwig the Owl](https://www.youtube.com/watch?v=VQ_j7HJwABw)
- @Patlepirate01: [Hedwig the Owl](https://www.facebook.com/Pybricks/posts/662400101062140)
- @Patlepirate01: [Haunted house](https://www.facebook.com/groups/PoweredUp/permalink/2364060840562040/)
- Pybricks team: [Roller coaster](https://www.facebook.com/Pybricks/posts/632324460736371)
- Pybricks team: [Self balancing 42099 Off Road Truck](https://youtu.be/qi6MRyLSrsI)
- Pybricks team: [EV3 Education Robots](https://pybricks.github.io/ev3-micropython/examples/color_sorter.html)
- @AntoniLuongPham: [EV3 Home Edition Robots](https://github.com/pybricks/pybricks-projects)

# Bluetooth / remote control
- Dimitri: [SPIKE SwirlBot remote control](https://www.youtube.com/watch?v=l_aQ-F2_25M)
- @jorgepe: [Augmented remote control using a keyboard](https://www.youtube.com/watch?v=esE_6oiFy20)

# Trains:
- Pybricks team: [Automated freight train and switches](https://www.facebook.com/Pybricks/posts/601700160465468)
- Pybricks team: [Train color sorter](https://www.facebook.com/Pybricks/posts/564500934185391)
- Karin: [Automated train barrier](https://www.youtube.com/watch?v=79LWUEo21dg)
- Technicmaster0: [Automated railway crossing](https://www.youtube.com/watch?v=jDTvF2vR95w)

# Machines
- @zphilo: [Ball color sorter](https://youtu.be/YzaEvIgPkdY)
- @kueden: [Time machine](https://www.youtube.com/watch?v=e-JSfwyZt6c)

# Line followers
- @zphilo: [Off road line follower](https://www.youtube.com/watch?v=TFhKRbFzMF0)
- @jorgepe: [Mixed platform line follower](https://www.youtube.com/watch?v=oHwqm5GqPTU)

# Great Ball Contraption
- rjbrickbuilds: [Entry for Great Ball Pit's 2020 GBC Challenge](https://rebrickable.com/mocs/MOC-55141/rjbrickbuilds/rjs-thirteen)

# Games
- @jorgepe: [Google T-Rex game](https://www.youtube.com/watch?v=1HaFRPFWk_I)
- Dimitri: [Mario Piranha Powerslide](https://www.youtube.com/watch?v=JotTs3OWKdU)

# Multi-hub projects
- @jorgepe: [Hive test](https://www.youtube.com/embed/6-3q4zq-a7w)

XilefTech, 2020-10-26:
Is this irl-projects connected to a robot of some sort, or just gereally pybricks-coded stuff.

If it is the last one, you can maybe add [CharlieOSX](https://github.com/XilefTech/CharlieOSX)
(might wannt to add a WIP label to it ;) )

laurensvalk, 2020-10-26:
All Pybricks projects are welcome. I'll add it, thanks!

TheVinhLuong102, 2020-10-26:
This list is amazing! Love seeing these creations from the PyBricks community!

TheVinhLuong102, 2020-10-31:
@laurensvalk may I check if PyBricks-API / PyBricks-Projects participate in DigitalOcean's [Hacktoberfest](https://hacktoberfest.digitalocean.com)? Could you make the PyBricks repos Hacktoberfest-participating repos so that we can get Hacktoberfest T-Shirts? :)

dlech, 2020-10-31:
I added the `hacktoberfest-accepted` label on the PRs from @AntoniLuongPham so hopefully that works.

TheVinhLuong102, 2020-10-31:
@dlech thank you so much!

closed 2020-12-16 by dlech