pybricks/support #158: [Bug] Technic Hub accelerometer has poor performance

azzieg, 2020-10-27:
The accelerometer data from IMU.accel() is very slow to respond to orientation changes, it takes over a second for it to settle after rotating the brick. The performance of the same brick in the official PoweredUp hub is much better.

I used the following code:

from pybricks.experimental import IMU
from pybricks.tools import wait
imu = IMU()
while True:
  print(imu.accel())
  wait(100)

If you compare with the accelerometer sensor brick in the official Powered Up app (I believe it's the orange one with XYZ axes) the responsiveness is way better, the values settle immediately after rotating the brick.

What I think might be the culprit is how the sensor is initialized in pybricks-micropython/pybricks/experimental/pb_module_experimental.c. The accelerometer seems to be using a low sampling rate of 12.5 Hz (LSM6DS3TR_C_XL_ODR_12Hz5) and two low pass filters with a pretty aggressive setting (LSM6DS3TR_C_XL_LOW_NOISE_LP_ODR_DIV_100). Would it perhaps make sense to use a higher sampling rate (ODR) and/or less aggressive filtering settings, like LPF1-only or higher cut-off settings for LPF2? Also the applied setting setting for the analog low-pass filter seems to be more aggressive than the default, but not sure what role this plays (official documentation says it's only used for anti-aliasing at high ODRs).

I found https://www.st.com/resource/en/datasheet/lsm6ds3tr-c.pdf to be extremely valuable in understanding the filtering setup, especially Figure 7 (gives a high-level overview of filtering blocks), Figure 8 (explains the composite filter details including how various inputs affect the signal path) and Table 73 (presents how various combinations of settings impact the effective filtering bandwidth).

In general, does it actually make much sense to use digital filters in the sensor, or given the general purpose of Pybricks it would it make sense to provide as raw data as possible? While gyro sensor configuration applies lightweight LPF and HPF filtering, most filtering can also be skipped there (see Figure 6).

laurensvalk, 2020-10-27:
Nice find, and thanks for opening this issue! You're totally right, we haven't configured the accelerometer at all yet. That's why it is currently only available in the `experimental` module. We plan to rewrite this code completely, but it's not on the roadmap for the very next release #29 just yet.

The experimental code we have now is provided by the manufacturer (STMicroelectronics), but it uses *blocking* I2C operations. That sort of works, but you might find that other things do not work if they occur at exactly the same time. For example, sometimes the stop button does not work if you read the IMU in a fast loop. So this needs to be re-implemented in a non-blocking manner.

So in short, we have the code there because it was too much fun not to include it, but it needs some work. We left most values at the default values from ST. I did [tweak the Gyro](https://github.com/pybricks/pybricks-micropython/commit/27acc5cc81709dafde3d18fb2cea5452532fda01#diff-b6dd0e22f477a75e8d36f43f9ac53a93a70cd9e761c533c26cd7f7fafc92c0a6) just a bit to try out self-balancing performance, which works very well now.

If you're interested, we could do something similar for the accelerometer, so that it works well enough for quite a few applications until we thoroughly rewrite this code and make the filters configurable.

Do you want to try to [build the firmware yourself](https://github.com/pybricks/pybricks-micropython/blob/master/CONTRIBUTING.md)? If you want to try various filter settings and tell us which would be good for now, we'd be happy to take a pull request or change the values for you.

laurensvalk, 2021-01-05:
Most filters have been disabled in recent builds, so the sensor is now much more responsive.

Meanwhile, the code is non-blocking and a few methods are already available via the hub instead of via `experimental`.

You can already try this without building the code. To try the latest firmware, view the "Show/hide beta steps to save a program permanently" section on the [getting started page](https://docs.pybricks.com/en/work/start_pup.html).

closed 2021-07-13 by laurensvalk