pybricks/support #159: [Question] How do I calibrate the NXT Compass Sensor?

Boss-Chicken, 2020-11-03:
**Question:**
How do I calibrate the NXT Compass Sensor?

**Context:**
I was (and still am) trying to make a program that allows my robot to go in a straight line with the help of a PID. I noticed that my compass readings are inaccurate and attempted to recalibrate it to get better readings, but I don't know how to do so!

Any help will be greatly appreciated!

**Screenshots:**
[image]

Below is the code I have tried...

[image]

laurensvalk, 2020-11-03:
The `Ev3devSensor` class has no `write` method, but you can add one. Have a look at the **second example** [here](https://docs.pybricks.com/en/latest/iodevices.html#pybricks.iodevices.Ev3devSensor).

It shows how you can add a method `get_modes`. It sounds like you want to add a method like `write_command`, which could look something like this:

```python

    def write_command(self, command):
        """Write to the command attribute."""

        # Open the command file.
        with open(self.path + '/command', 'w') as command_file:

            # Write the command.
            command_file.write(command)
```

Boss-Chicken, 2020-11-03:
> The `Ev3devSensor` class has no `write` method, but you can add one. Have a look at the **second example** [here](https://docs.pybricks.com/en/latest/iodevices.html#pybricks.iodevices.Ev3devSensor).
> 
> It shows how you can add a method `get_modes`. It sounds like you want to add a method like `write_command`, which could look something like this (I have not tested this):
> 
> ```python
>     def write_command(self, command):
>         """Write to the command attribute."""
> 
>         # Open the command file.
>         with open(self.path + '/command', 'w') as command_file:
> 
>             # Write the command.
>             command_file.write(command)
> ```

I don't have much experience messing around with classes, can you please show me how to do it without the classes?

Thanks!

laurensvalk, 2020-11-03:
Have you looked at the example I linked to? You can add the method shown here just below the `get_modes` method.

Boss-Chicken, 2020-11-03:
I'll try it out and let you know how it goes, thanks!

dlech, 2021-07-06:
Is this still a problem?

closed 2022-06-03 by dlech