pybricks/support #160: Train motor is not detected on City hub

dlech, 2020-11-04:
First of all, thank you for all the hard work you do with this project.

I am also able to connect to the city hub now.
For a first test, I plug a train motor in port A and load the example 1 program from [Motors without Rotation Sensors](https://docs.pybricks.com/en/latest/pupdevices.html#motors-without-rotation-sensors).
The code editor prints following error:
```
Traceback (most recent call last):
  File "main.py", line 7, in <module>
OSError: [Errno 19] ENODEV: 

A sensor or motor is not connected to the specified port:
--> Check the cables to each motor and sensor.
--> Check the port settings in your script.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.
```
I tried also the other port, but always the same error.

A press on the connect button, prints the following
```
('cityhub', '3.0.0a9', 'v3.0.0a9-25-g2541b8e4 on 2020-10-13')
Hello, World!
```
Regards

_Originally posted by @mwinkler in https://github.com/pybricks/support/issues/148#issuecomment-721424031_

dlech, 2020-11-04:
Does the problem still happen after rebooting the hub?

mwinkler, 2020-11-04:
A reboot did the trick! It works now.
Thank you

laurensvalk, 2020-11-04:
Nice! See also the [note in the getting started guide](https://docs.pybricks.com/en/latest/start_pup.html) about device detection, in case you run into this again. It's a known issue on our backlog.

closed 2020-11-04 by laurensvalk