pybricks/support #161: [Question] How can we use the ev3dev2 in pybricks-python?

ChenCyril, 2020-11-07:
**Question**
For some reason, I need to use the WIFI to send the message from the EV3 to computer. And I can't find out the function of WIFI (Only see the Bluetooth) in pybricks. So I want to install the ev3dev2 in pybricks-python. But I was failed to do that. 

**Context**
I have tried  `sudo brickrun -r -- pybricks-upip install micropython-ev3dev2`  to install ev3dev2. But it tells me that  `Unexpected error: GDBus.Error:org.freedesktop.DBus.Error.AccessDenied: Rejected send message, 1 matched rules; type="method_call", sender=":1.52" (uid=0 pid=1655 comm="brickrun -r -- pybricks-upip install micropython-e") interface="org.ev3dev.ConsoleRunner" member="Start" error name="(unset)" requested_reply="0" destination="org.ev3dev.ConsoleRunner" (uid=1000 pid=397 comm="/usr/bin/conrun-server ")` .
And if I use the command without `sudo` , it tells me that `  File "upip/upip.py", line 314, in <module>
  File "upip/upip.py", line 307, in main
  File "upip/upip.py", line 204, in install
IndexError: string index out of range`

So, how to solve this problem to install micropython-ev3dev2? Or how to install pybricks in micropython?
Or, can I have the pybricks-api to let the EV3 to communicate with the PC and other EV3?
(I use the Edimax wifi dongle)
I will also appreciate that you can give other solutions to let the EV3 to communicate with the PC and other EV3 by WIFI.

laurensvalk, 2020-11-07:
`pybricks-micropython` uses a newer version of `micropython` compared to `ev3dev2`, so you cannot import `ev3dev2`.

> Or, can I have the pybricks-api to let the EV3 to communicate with the PC and other EV3?

Fortunately, yes! You can use the same tools of the `pybricks.messaging` module on your PC.

Here is an example project you can run on Windows/Mac/Linux. Be sure to download everything you see on this page, including the `pybricks` folder. It contains an implementation of `pybricks.messaging` that runs on your computer.
https://github.com/pybricks/pybricks-api/tree/master/examples/ev3/bluetooth_pc

It requires that you have a bluetooth Python library on your PC such as pybluez.

ChenCyril, 2020-11-09:
Thanks a lot. 
The bluetooth is not steady enough for me. Now I use the bluetooth to communicate for short time and use the the package `socket` in pybricks-micropython to communicate by network.

closed 2020-11-09 by ChenCyril