pybricks/support #163: The print command causes software crash

kueden, 2020-11-08:
**Describe the bug**
Repeated print commands cause crash of the Technic Hub

**To reproduce**
Steps to reproduce the behavior:
1. Run the "test.txt" code on the technic hug
2. remote control it with the keyboard
3. after a few commands it freezes
4. removing the print command solves the issue. 

**Expected behavior**
It would be nice to have the print command though.

**Screenshots**

[test.txt](https://github.com/pybricks/support/files/5506588/test.txt)

laurensvalk, 2020-11-08:
Thanks for the report.

By the way, you can include the code directly in your post instead of attaching it. It works like this:

[image]

kueden, 2020-11-08:
Thanks, I was wondering how you guys do this. My copy paste attempts failed
miserably.

I really appreciate what you guys are doing and I am happy to contribute.
However I do not want to overload your site with my stuff.

I have another project series working towards an autonomous robot with
spatial orientation written in your python code:
https://rebrickable.com/mocs/MOC-51737/kueden/hom3runn3r-ii/#details

If you want me to share it here let me know.

Markus

laurensvalk, 2020-11-08:
No problem. You can still edit your initial post to make this change if you like.

Don't worry about overloading - feel free to share! Especially if you want to discuss some Pybricks tips and tricks that others might find useful.

dlech, 2020-11-17:
possibly related print() bug: #36

dlech, 2021-04-02:
There have been significant changed made to Bluetooth in firmware v3.0.0b3. Can you please upgrade the firmware and test again?

kueden, 2021-04-03:
Thanks David

It works perfetly well! No crash after extensive testing.

Thanks for all your help

Markus

kueden, 2021-04-03:
thanks for this!

closed 2021-04-03 by kueden

kueden, 2021-04-03:
David
I got it to work, but there is another issue. There is a bug with the
v3.0.0b3. I can connect via blootooth. But the play bottun (F5 for download
and run) does not light up. I played around and got it to work. My original
issue is resolved. However I cannot reproducibly get software to run since
the play button does not show up. Eventually things run. I also learned
playing with it that the "getchar" that I use in all my software (issue
#154) is not available in this new version

So it looks like my minor issue is resolved, but I cannot run my software
any more due to the not active play buttion and the missing getchar
function. Now it is very probable that this is my lack of competence. I
tried on another computer. Same thing.

Thanks for all your help

Markus

laurensvalk, 2021-04-03:
See #287. It looks like this is affecting more people since a recent update.

kueden, 2021-04-03:
Thanks

Indeed exactly the same. No rush from my side to get this fixed. If you
could implement the getchar command in this version, this would be
appreciated.

Thanks

Markus

laurensvalk, 2021-04-03:
`getchar` has been replaced by more standard tools that can do the same thing, and more.

See [this commit message](https://github.com/pybricks/pybricks-micropython/commit/65b8ec05153d8f6c67782249d20db5f03969dbb8) for details. It would be nice to include this in an example at some point.