pybricks/support #165: [Question] BSOD after connecting to the EV3 via Bluetooth

KoenTimmermans, 2020-11-12:
**Question**
My laptop stops working as soon as it is connected to the EV3 via Bluetooth and my question is if there is someone here who has a (possible) solution for that?

**Context**
I am trying to connect my EV3 to my laptop, via Bluetooth. The connection has already been successful via the USB cable and I have been able to run programs.

Within half a minute of connecting the EV3 to the laptop, I get a blue screen of death (BSOD) with the stop code: "INVALID_MDL_RANGE".

When I look back in my Event Viewer, the message in the screenshot below continuously returns just before the laptop fails. So I suspect this has to do with it.

I write the code in VS Code, where I have the MicroPython and ev3dev browser extensions installed.

In the "ev3dev device browser" the EV3-brick is displayed, both with USB and Bluetooth connection. I also managed to send the program to the EV3 and run it via Bluetooth just before the laptop shuts down.

While trying to figure out the problem, I came across [this](https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth/) page, among other pages. In principle, this is also the step-by-step plan that I follow for connecting my EV3.

To rule out the fact that it is the EV3 I have already tried it on another EV3, which is experiencing the same problem.
My laptop runs Windows 10. If more specifications can be relevant, I can of course provide them.

The problem probably isn't in PyBricks itself, but I'm mainly curious if there is anyone here who has had the same or a similar problem and / or maybe has any suggestions.

**Screenshots**
This is the recurring message in Event Viewer:

[image]

dlech, 2020-11-20:
If you are getting a BSOD, it is most likely due to a hardware problem. Perhaps you could try a different Bluetooth dongle?

KoenTimmermans, 2020-11-21:
Thanks for your reply. As far as I know I don't have a separate Bluetooth dongle, but I can see if it works better on another laptop (with built-in Bluetooth adapter). Incidentally, other Bluetooth devices do work without problems with the laptop on which I now get the BSOD.

KoenTimmermans, 2020-11-25:
The problem is probably in the firmware of the Bluetooth adapter. I adjusted the firmware and it went well for a little longer, but eventually a BSOD again. It works fine with other devices, so the hardware is probably fine.

I have also tried it with another laptop (with built-in Bluetooth) and that works indeed. So for now I will continue with that one.

closed 2021-05-11 by dlech