pybricks/support #166: [Project] SteerBot - A 51515 Robot Inventor car that edge tracks a line

GusJansson, 2020-11-17:
**Description**
A robot with car like steering has been one of my favorite exercises in robot building.  This project, in fact, is largely a variation of what I have previously done with the LEGO RCX, the NXT and to a lesser extent the EV3.  But with the 51515 set I as able to take advantage of improved resolution of the color sensor as well as some cool features of Pybricks to make the main tracking code pretty simple.

The robot features Ackermann steering.  This means that the steering linkage is more trapezoidal than parallel.  The front wheels are only parallel when the steering is pointing straight ahead.  When it turns the inside wheel turns sharper than the outside wheel so the wheels track more accurately since the inside wheel has a tighter turning radius.  This is particularly important in order to achieve tight overall turning radius with a relatively wide robot.

The robot uses the Color Sensor mounted on an axle high and attached to the steering motor.  This means that in order to track a line, actually just the edge of the line, the tracking code tries to keep the sensor on the edge of the line.  If the sensor is on the edge, then as the robot drives it will follow the edge since the robot is now steering to where the edge is in front of the robot.  The sensor is mounted high in order to give a gradual reading over the edge so that while near the edge the program can determine where the edge is relative to the sensor.

The driving is done through the super cool 5 gear differential that is included with the set.  This is the first time I use this LEGO differential and I love it!  The drive motor is actually gearing up with the intention of being able to go really fast.  On my tight test track I was never able to make it go full speed without running off the line.  

**Video**
https://www.youtube.com/watch?v=i4WPa-rtFB8&feature=youtu.be

**Source Code**
```
# SteerBot for Pybricks
#
# To download on brick:
# in CMD cd to the directory for pybricksdev
# poetry run pybricksdev run usb 'Hub' ../pybricks/steerbot.py
#
from pybricks.hubs import InventorHub
from pybricks.pupdevices import Motor
from pybricks.parameters import Port, Button, Stop
from pybricks.tools import wait
from pybricks.iodevices import LUMPDevice

# Initialize the hub, motors, and sensor
hub = InventorHub()
steer_motor = Motor(Port.A)
drive_motor = Motor(Port.B)
lump = LUMPDevice(Port.C)

def WaitForButton(b):
    while b in hub.buttons.pressed():
        wait (10)
    while b not in hub.buttons.pressed():
        wait (10)
    while b in hub.buttons.pressed():
        wait (10)

# Get a light value from the ColorSensor
def GetLight():
    # Mode 6 for the color sensor returns HSV.
    # Testing with blue painters tape showed that the Saturation value
    # gave by far the best resolution and immunity to variation caused by
    # sensor height.
    return (lump.read(6)[1])

def Calibrate():
    global aRightLimit, aLeftLimit
    global lMin, lMax, signEdge
    
    # Reset steer motor to absolute angle
    steer_motor.reset_angle()

    # Find the Right and Left hard limits 
    steer_motor.run_until_stalled(400, then=Stop.BRAKE, duty_limit=100)
    aRightLimit = steer_motor.angle()
    steer_motor.run_until_stalled(-400, then=Stop.BRAKE, duty_limit=100)
    aLeftLimit = steer_motor.angle()

    # Scan from -30 to 30 to get min max of light sensor value
    steer_motor.run_target(1000,-30, then=Stop.BRAKE)
    lMin = 1024; lMax = 0
    lLeft = GetLight()
    steer_motor.run(100)
    c = 0
    while steer_motor.angle() < 30:
        c += 1
        l = GetLight()
        if l > lMax: lMax = l
        if l < lMin: lMin = l
        wait(5)
    steer_motor.stop()
    lRight = GetLight()
    # signEdge is positive 1 if left edge and -1 if right edge
    signEdge = 1 if lLeft < lRight else -1
        
    # Center the steering
    steer_motor.run_target(1000,0,then=Stop.BRAKE,wait=False)
    

SPEED_MAX = 1000
SPEED_INC = 3
SPEED_INIT = 300
SPEED_TURN = 500
SPEED_OFFLINE = 400
SPEED_MIN = 0
cur_speed = 0

def SetSpeed(s):
    global cur_speed
    if s > SPEED_MAX: s = SPEED_MAX
    if s < SPEED_MIN: s = SPEED_MIN
    if s != cur_speed:
        cur_speed = s
        drive_motor.run(cur_speed)
        
def SetSpeedTarget(s):
    global cur_speed
    if s > cur_speed:
        SetSpeed(min(cur_speed + SPEED_INC,s))
    elif s < cur_speed:
        SetSpeed(max(cur_speed - SPEED_INC,s))
        
def TrackSpeedControl():
    global lMin, lMax, signEdge
    global aLeftLimit, aRightLimit
    
    lMid = (lMax+lMin+1)//2
    aSteerMid = (aLeftLimit+aRightLimit+1)//2
    
    d = (lMax-lMid) // 20

    # Calculate a threshold to determine steering is not near the edge
    lOffEdgeThresh = (lMax-lMid) * 0.7

    SetSpeed(SPEED_INIT)
    while hub.buttons.pressed() == []:
        # Get a new light value and subtract mid to get signed error from edge
        l = signEdge * (GetLight()-lMid)

        # Create a new target for the steering motor to move toward the
        # approximate position of the edge
        a = steer_motor.angle()
        t = a - l//d
        if t > aRightLimit:
            t = aRightLimit
        elif t < aLeftLimit:
            t = aLeftLimit
        steer_motor.track_target(t)

        if abs(l) < lOffEdgeThresh:
            # On edge
            if abs(t-aSteerMid) < 25:
                # On edge and going straight
                SetSpeedTarget(SPEED_MAX)
            else:
                SetSpeedTarget(SPEED_TURN)
        else:
            SetSpeedTarget(SPEED_OFFLINE)
                
        wait(3)

    SetSpeed(0)
    steer_motor.track_target(aSteerMid)
    wait(200)
    steer_motor.stop()
    drive_motor.stop()

def Test():
    while True:
        WaitForButton(Button.RIGHT)
        Calibrate()
        TrackSpeedControl()

Test()
```

**Programming Notes**
Originally I was using the reflection() light value from the ColorSensor.  With the blue painters tape I had available and with the sensor mounted high in order to get a gradual value as the sensor passed over the edge, I got a rather unsatisfying resolution in the sensor value.  Something like 3 for on the tape and 29 when over the white surface off the tape.  Along with a lot of variation in value as the sensor bounced up and down on the axle.  After talking to Laurens about this I found out I could get the raw sensor values using the LUMPDevice class.  A little experimenting and I discovered that the raw Saturation value was perfect.  With the Saturation value you are not measuring how much light is coming back into the sensor, but how much color there is.  Since my tape is blue, I got a high saturation value over the tape (about 675) and a  low value, since there is no color, (about 100) over the white surface.

I also had some fun controlling the speed of the robot.  My test track is very tight and the width of the boards are only 2 feet (61cm) wide.  That means if the robot was driving fast it had no chance to make the turns and stay on the boards.  But a slow robot is no fun so I worked on a scheme to go fast when going straight but slow down in turns, especially when the sensor as no longer near the edge of the line.  And, as any good driver will tell you, it is best to control the speed smoothly.  Gradually speeding up and slowing down.

**Building instructions**
I have not created building instructions but I have modelled the robot using Studio.  If there is interest in building instructions then I will consider putting something together.
[image]

dlech, 2020-11-17:
This is really nice, thanks for sharing!

One thing I was curious about is if you could get away with not having a calibration of the steering since the motors have absolute encoders.

GusJansson, 2020-11-17:
Thanks, David!  And thanks for Pybricks.
I found that even though my motors are reasonably accurate, the absolute 0 position is within 3 degrees off the mark, that error was enough to have an impact on the steering.  My left and right limits are something like 6 degrees of each other.  One of my goals was to have very tight turning radius and my test track ended up being pretty much as tight as what my robot could handle.  In this case, every degree of additional steering counted!  
That said, the tracking code doesn't really care what the steering angle is.  It just sets the target for the steering relative to where it thinks the edge of the line is.  The hard limits only come into play in that the code prevents setting the target beyond the hard limits in order to reduce strain on the motor.

ZPhilo, 2020-11-18:
...and did you try to use LEGO Python to show benefit/drawbacks of Pybricks/LEGO implementations?

laurensvalk, 2020-11-18:
This is great, thanks for sharing both the code and the design :smile: 

Looks like you've already found quite a few fun Pybricks features. Here are a few more tricks you might be interested in for your next creation:
- For some slight extra accuracy in the reset procedure, you can read the return value of `run_until_stalled()` directly instead of reading `angle()` again.
- If you then do `steer_motor.reset_angle((aRightLimit - aLeftLimit)/2)`, the 0 point is now the middle and everything is symmetric, simplifying the remaining code.
- Are `SetSpeedTarget`/`SetSpeed` made to gradually change the speed? If so, perhaps setting [`steer_motor.control.limits`](https://docs.pybricks.com/en/latest/motors.html#pybricks._common.Control.limits) might be an interesting way to simplify this. This will set the speed and acceleration limits for all `run` commands.

laurensvalk, 2020-11-18:
I'm curious, is the extra resolution of `LUMPDevice` making a big difference or does this give similar results?

```python
from pybricks.pupdevices import ColorSensor
from pybricks.parameters import Port
from pybricks.tools import wait

sensor = ColorSensor(Port.B)

while True:
    # Gets the HSV value
    color = sensor.hsv()

    # Print saturation
    print(color.s)

    # Wait
    wait(500)
```

If it's a noticeable/big difference in performance in your application, then others might experience the same. If so, we can probably consider looking at increasing the resolution for `s`, and `v`, which are currently integers rounded to 0--100.

GusJansson, 2020-11-19:
Thanks for the feedback, Laurens!  I will incorporate your suggestions and then update the code here with these updates.  I will also investigate the use of reading the saturation from hsv.  I think the resolution should be adequate but will test to confirm.

Philo, there are definitely some features of the the Pybricks firmware that I took advantage of but I can't say conclusively that this can't be done as well with LEGO python.  While Pybricks still has some rough edges, crashed a few times, and is non trivial to install and setup so you can connect to the hub, it has one HUGE advantage over the LEGO Python and that is the documentation.  If it **fantastic**.   Take a look here: [https://docs.pybricks.com/en/latest/pupdevices.html](url) and make sure to click on Show Examples.

GusJansson, 2020-11-24:
I have created some preliminary building instructions for the SteerBot.  These do not show the cable routing but it was not difficult to route cables under and around the Hub.  
[steerbot.pdf](https://github.com/pybricks/support/files/5591587/steerbot.pdf)
Port A   Steering Motor
Port B   Drive Motor
Port C   Color Sensor

ZPhilo, 2020-11-25:
Nitpicker's dept.: differential is supposed to be grey, not white ;)

laurensvalk, 2020-11-25:
This is great @GusJansson --- thanks for sharing.

GusJansson, 2020-11-25:
Philo! You are right! I fixed it in the model.  It makes me happy that you went through the build.  Thanks.  Now for getting the wires done I need either go back to trusty old MLCad that I have not used it years or invest the time to learn LDCad.  What do you suggest?  There is no way to do the wires in Studio, right?

ZPhilo, 2020-11-25:
No, so far Studio can't create cables. I would definitely try to learn LDCad: creating flexibles parts with LDCad is almost easy... After that, remains the problem for compatibility between cables generated in LDraw domain and Studio building instructions. With direct import, Studio generates BIs for the cable itself ;)
- Documentation of LDCad is here: http://www.melkert.net/LDCad/docs
- There is an "official" video showing LDCad flexible part creation (pneumatic tubing) here: https://youtu.be/Peu5yqD2iAo
- I made a video showing creation of cables for Spike Prime: https://youtu.be/vDnkkPsyCT4
...and when you get more fluent with cables generation, I'll help you to reimport in Studio!

closed 2020-12-16 by dlech