pybricks/support #167: [Question] Getting Started Documentation for Robot Inventor?

johnscary-ev3, 2020-11-21:
**Question**
Is there documentation for how to get started using Pybricks with the Robot Inventor?

**Context**
Seems like it is being used to do projects by some, however the Docs don't show how to get going.

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

laurensvalk, 2020-11-21:
Thanks for asking John! You're right, we're quite far along with Pybricks for the Inventor Hub and the Prime Hub. The documentation currently says [coming soon](https://docs.pybricks.com/en/latest/start_inventor.html), because it is still a bit tricky for beginning users. This is because we haven't finished implementing Bluetooth. This means it can't be used with [Pybricks Code](https://www.code.pybricks.com/) just yet. The only way to use it is to install some Python libraries on your computer.

But since you are asking here, this is a good opportunity to show the more curious users how to get started already.

## Overview

The installation consists of three steps:
1. Running a script on the hub using the official app. This installs the Pybricks firmware alongside the original.
2. Starting Pybricks by holding the right button when you turn it on.
3. Installing a Python package on your computer, so you can run programs.

## 0. NOTE TO NEW USERS

I'm going to add one note of caution just in case:

*If you've never installed Python packages on your computer before, please wait until [this guide is ready](https://docs.pybricks.com/en/latest/start_inventor.html). If you're worried about how to recover your hub if something goes wrong, please wait as well. Proceed at your own risk.*

## 1: Installing the firmware

1. Make sure you are logged into GitHub.
2. Go to the latest [master firmware builds](https://github.com/pybricks/pybricks-micropython/actions?query=workflow%3ABuild+branch%3Amaster).
3. Click on the latest build with a green checkmark (:heavy_check_mark:)
4. Download the `primehub` firmware zip file. It contains the installers for both hubs: 
[image]
   - If you have the SPIKE Prime Hub (yellow), open the `install_pybricks.llsp` using the official app.
   - If you have the Robot Inventor Hub (teal green), open the `install_pybricks.lms` using the official app.
5. Connect using USB and run the script.
   - **NOTE**: This is a large script, so it takes a while to download. You will not see download progress. Just wait a few minutes and it will start.
   - Just wait for the script to complete. The hub lights indicate installation progress.
   - Then it reboots and you will see the Bluetooth light blink. Keep waiting, you're almost done.
   - Finally, the hub reboots and you will see the standard firmware as if nothing happened. Success!

## 2: Running Pybricks

1. Turn off the hub and unplug USB
2. Hold the right button and turn the hub on again. If the status light is blue, you are running Pybricks.
3. To run the regular firmware again, just reboot. 

**NOTE**: For the moment, the battery does not automatically charge while running Pybricks. So, make sure to reboot the hub when you are done using it, so it can charge the battery using the standard firmware.

## 3: Installing the Pybricksdev command line tool to download and run programs.

*Coming soon!*

BertLindeman, 2020-11-21:
@laurensvalk Maybe for the final getting started text on:

> 1: Installing the firmware
> ...
>        If you have the SPIKE Prime Hub (yellow), open the install_pybricks.llsp using the official app.
>        If you have the Robot Inventor Hub (teal green), open the install_pybricks.lms using the official app.

Maybe trivial, but as I have both hubs, I **started** using the "other" hub.
Add the name of the hubs:

`If you have the SPIKE Prime Hub (yellow), open the install_pybricks.llsp using the official` **Spike** `app.`
`If you have the Robot Inventor Hub (teal green), open the install_pybricks.lms using the official` **Robot Inventor** `app.`

laurensvalk, 2020-11-21:
Done, thanks @BertLindeman ! Looking forward to hear how it goes.

BertLindeman, 2020-11-21:
@laurensvalk Sure do too. 
Thanks

johnscary-ev3, 2020-11-22:
Thanks for all the detailed info on how to get going with Pybricks on the Robot Inventor.
I will give it a try.
Good to hear I can use Visual Studio Code also since I enjoyed using that with the EV3.
Can use Git and GitHub and the nice editor also.

johnscary-ev3, 2020-11-22:
Got stuck on install of pybricksdev.
Managed to get the pybrickdev package, find and install "poetry" ver 1.1.4,  and get python 3.8,
but getting this error below when doing the "poetry install" step.
The directory C:/Users/John/AppData/Local/pypoetry/ does not seem to exist.
Any ideas?

PS C:\Users\John\pybricksdev> poetry install
Installing dependencies from lock file

Package operations: 70 installs, 0 updates, 0 removals

  • Installing ipython-genutils (0.2.0)

  EnvCommandError

  Command C:\Users\John\pybricksdev\.venv\Scripts\pip.exe install --no-deps file:///C:/Users/John/AppData/Local/pypoetry/Cache/artifacts/e6/8e/a3/8f37e14310c0072b3fcc4240490bcb42630aa695d069aee89953ebd9f8/ipython_genutils-0.2.0-py2.py3-none-any.whl errored with the following return code 1, and output:
  Processing c:\users\john\appdata\local\pypoetry\cache\artifacts\e6\8e\a3\8f37e14310c0072b3fcc4240490bcb42630aa695d069aee89953ebd9f8\ipython_genutils-0.2.0-py2.py3-none-any.whl
  ERROR: Could not install packages due to an EnvironmentError: [Errno 2] No such file or directory: 'C:\\Users\\John\\AppData\\Local\\pypoetry\\Cache\\artifacts\\e6\\8e\\a3\\8f37e14310c0072b3fcc4240490bcb42630aa695d069aee89953ebd9f8\\ipython_genutils-0.2.0-py2.py3-none-any.whl'

  at ~\.poetry\lib\poetry\utils\env.py:1074 in _run
      1070│                 output = subprocess.check_output(
      1071│                     cmd, stderr=subprocess.STDOUT, **kwargs
      1072│                 )
      1073│         except CalledProcessError as e:
    → 1074│             raise EnvCommandError(e, input=input_)
      1075│
      1076│         return decode(output)
      1077│
      1078│     def execute(self, bin, *args, **kwargs):

PS C:\Users\John\pybricksdev> poetry --version
Poetry version 1.1.4
PS C:\Users\John\pybricksdev>

BertLindeman, 2020-11-22:
Just to report install on `Linux Mint  20 Ulyana` was successful.
As I do not want to dive in the poetry configuration and want to run programs in different directories made a small BASH script: `runpybricks.sh`. 

```bash
#!/bin/bash -u
#
# run primehub or inventorhub on USB
#

parm1="${1:-}"
parms="${*}"
# use default program if none is given:
if [ "${parms}" = "" ]; then
    parm1="/home/bert/py/pybricks/tools/show_firmware_version.py"
fi

# take the complete path also if the parm is a relative path
parm1="`realpath ${parm1}`"

# ignore the first parm
shift
parm_rest="$*"

set -x

# switch to the install location of the tools
cd /home/bert/Downloads/pybricksdev/pybricksdev
# poetry run pybricksdev run usb "Pybricks Hub" /home/bert/py/pybricks/${parms}
poetry run pybricksdev run usb "Pybricks Hub" ${parm1} ${parm_rest}

```

So I can "`runpybricks.sh`" in any directory.
 
@laurens If (when ;-) ) I find issues while running this way, should I create a separate issue for them or do you want them added here?
Assuming a new issue, now I re-read this.

Thanks for all the fun, It's great. 
Bert

laurensvalk, 2020-11-22:
Sounds good @BertLindeman! Yes, feel free to create new issues as you find them. 

Issues with the hub/motors/sensors are probably the most relevant, since Pybricksdev is only a temporary development tool.

We will probably keep using the llsp/lms project for the firmware installation, so that’s another one where bug reports are quite welcome. If we can, we’ll possibly run it from Pybricks Code directly so you don’t have to switch apps (and so you don’t need windows or Mac).

Alitai12, 2020-11-22:
> Got stuck on install of pybricksdev.
> Managed to get the pybrickdev package, find and install "poetry" ver 1.1.4, and get python 3.8,
> but getting this error below when doing the "poetry install" [...]

Do you have Python 3.8.2?

Open Powershell as Admin and install poetry "(Invoke-WebRequest -Uri https://raw.githubusercontent.com/python-poetry/poetry/master/get-poetry.py -UseBasicParsing).Content | python -"

Open a new Powershell and check version "poetry --version"

Alitai12, 2020-11-22:
> Got stuck on install of pybricksdev.
> Managed to get the pybrickdev package, find and install "poetry" ver 1.1.4, and get python 3.8,
> but getting this error below when doing the "poetry install" [...]

What does "poetry config --list" say?
For me: cache-dir = "C:\\Users\\Alitai\\AppData\\Local\\pypoetry\\Cache"

laurensvalk, 2020-11-22:
@johnscary-ev3 , could you try removing the `.venv` folder and the `poetry.lock` file and running `poetry install` again?

Alitai12, 2020-11-22:
I'm stuck with this message (on 2 clients): "Python 3.8.2 and poetry 1.1.4"

`C:\Users\Alitai\Downloads\pybricksdev-master\pybricksdev>poetry install
Creating virtualenv pybricksdev in C:\Users\Alitai\Downloads\pybricksdev-master\.venv
Installing dependencies from lock file

  FileNotFoundError

  [WinError 2] The system cannot find the specified file

  at c:\users\alitai\appdata\local\programs\python\python38\lib\subprocess.py:1307 in _execute_child
      1303│             sys.audit("subprocess.Popen", executable, args, cwd, env)
      1304│
      1305│             # Start the process
      1306│             try:
    → 1307│                 hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
      1308│                                          # no special security
      1309│                                          None, None,
      1310│                                          int(not close_fds),
      1311│                                          creationflags,`

laurensvalk, 2020-11-22:
That error can happen when Git is not installed. I will update the instructions above. Thanks for reporting!

Alitai12, 2020-11-22:
To run:
poetry run pybricksdev run usb "Pybricks Hub" C:\Users\YOURUSERNAME\Downloads\pybricksdev-master\demo\longdemo.py

and

Open Folder in VSC:
pybricksdev-master (not pybricksdev)

johnscary-ev3, 2020-11-23:
@laurensvalk , I removed those files and "poetry install" rebuilt them but then gave the same error. See below.
The directory C:/Users/John/AppData/Local/pypoetry/ which it is looking for does not exist.
Maybe something went wrong on installing poetry?
I have Git installed and using it successfully with Visual Studio Code.
Thanks.

----------------------------------------------------------------------------------------------------------------------------

PS C:\users\john\pybricksdev> rm .venv

Confirm
The item at C:\users\john\pybricksdev\.venv has children and the Recurse parameter was not specified. If you continue, all children will be
 removed with the item. Are you sure you want to continue?
[Y] Yes  [A] Yes to All  [N] No  [L] No to All  [S] Suspend  [?] Help (default is "Y"): A
PS C:\users\john\pybricksdev> rm poetry.lock
PS C:\users\john\pybricksdev> poetry install
The currently activated Python version 3.9.0 is not supported by the project (~3.8).
Trying to find and use a compatible version.
Using python3.8 (3.8.6)
Creating virtualenv pybricksdev in C:\users\john\pybricksdev\.venv
Updating dependencies
Resolving dependencies...

Writing lock file

Package operations: 70 installs, 0 updates, 0 removals

  • Installing ipython-genutils (0.2.0)

  EnvCommandError

  Command C:\users\john\pybricksdev\.venv\Scripts\pip.exe install --no-deps file:///C:/Users/John/AppData/Local/pypoetry/Cache/artifacts/e6/8e/a3/8f37e14310c0072b3fcc4240490bcb42630aa695d069aee89953ebd9f8/ipython_genutils-0.2.0-py2.py3-none-any.whl errored with the following return code 1, and output:
  Processing c:\users\john\appdata\local\pypoetry\cache\artifacts\e6\8e\a3\8f37e14310c0072b3fcc4240490bcb42630aa695d069aee89953ebd9f8\ipython_genutils-0.2.0-py2.py3-none-any.whl
  ERROR: Could not install packages due to an EnvironmentError: [Errno 2] No such file or directory: 'C:\\Users\\John\\AppData\\Local\\pypoetry\\Cache\\artifacts\\e6\\8e\\a3\\8f37e14310c0072b3fcc4240490bcb42630aa695d069aee89953ebd9f8\\ipython_genutils-0.2.0-py2.py3-none-any.whl'

  WARNING: You are using pip version 20.2.2; however, version 20.2.4 is available.
  You should consider upgrading via the 'C:\Users\John\pybricksdev\.venv\Scripts\python.exe -m pip install --upgrade pip' command.

  at ~\.poetry\lib\poetry\utils\env.py:1074 in _run
      1070│                 output = subprocess.check_output(
      1071│                     cmd, stderr=subprocess.STDOUT, **kwargs
      1072│                 )
      1073│         except CalledProcessError as e:
    → 1074│             raise EnvCommandError(e, input=input_)
      1075│
      1076│         return decode(output)
      1077│
      1078│     def execute(self, bin, *args, **kwargs):

dlech, 2020-11-23:
Did you do as the warning suggests and run `C:\Users\John\pybricksdev.venv\Scripts\python.exe -m pip install --upgrade pip`?

You can also skip `poetry` altogether if you just want to install the command line tool:

```powershell
py -3.8 -m pip install --upgrade pip
py -3.8 -m pip install pipx
py -3.8 -m pipx ensurepath
py -3.8 -m pipx install https://github.com/pybricks/pybricksdev/archive/master.zip
```

You may have to go as far as logging out and logging back in for the `ensurepath` to fully take effect - it seems some apps will pick it up after restarting the app while others don't.

Then just run `pybricksdev` in a terminal.

Alitai12, 2020-11-23:
@johnscary-ev3 it says you need python 3.8.2 and not 3.8.6. Maybe that's the issue?

dlech, 2020-11-23:
Any Python 3.8.x version should work.

johnscary-ev3, 2020-11-23:
Hi,

  Thanks for the help.
 I managed to get the Inventor Hub firmware installed and got the pybricksdev command line tool installed using pipx as above.
So I can use command line from inside VSC to run pybricks programs on the hub.
However if I try to use the VSC debug "Run USB" command it fails, looking for 'validators' module. See below.
I assume that this module was supposed to have been installed by poetry which did not work for me for some reason.

  Any way it seems that users (not developers) like me would benefit by a more simple VSC extension that just connected to the Inventor hub and downloaded programs to the hub. Probably a new "launch.json" file to do this using the command line tool?
 I am thinking of something like what we have for the EV3.  VSC is a nice environment and EV3 pybricks users would already be familiar with it. Don't really need to mess with the entire pybricksdev package and installation.

Any comments?

------------------------------------------------------------------------------------------------------------------------------------
"Command line"
PS C:\Users\John\pybricksdev> pybricksdev run usb "Pybricks Hub" demo/shortdemo.py
Connecting to COM6
2020-11-23 15:24:07,042:    INFO: Sending: 50%
2020-11-23 15:24:07,106:    INFO: Sending: 100%

Hello
World!
2020-11-23 15:24:08,340:    INFO: Disconnected by client.
PS C:\Users\John\pybricksdev> 

----------------------------------------------------------------------------------------------------------------------------------------
"Run USB"
PS C:\Users\John\pybricksdev>  cd 'c:\Users\John\pybricksdev'; & 'c:\Users\John\pybricksdev\.venv\scripts\python.exe' 'c:\Users\John\.vscode\extensions\ms-python.python-2020.11.371526539\pythonFiles\lib\python\debugpy\launcher' '59686' '--' '-m' 'pybricksdev' 'run' 'usb' 'Pybricks Hub' 'c:\Users\John\pybricksdev\demo\shortdemo.py' 
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.1776.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.1776.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\Users\John\.vscode\extensions\ms-python.python-2020.11.371526539\pythonFiles\lib\python\debugpy\__main__.py", line 45, in <module>
    cli.main()
  File "c:\Users\John\.vscode\extensions\ms-python.python-2020.11.371526539\pythonFiles\lib\python\debugpy/..\debugpy\server\cli.py", line 430, in main       
    run()
  File "c:\Users\John\.vscode\extensions\ms-python.python-2020.11.371526539\pythonFiles\lib\python\debugpy/..\debugpy\server\cli.py", line 317, in run_module 
    run_module_as_main(target, alter_argv=True)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.1776.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 194, in _run_module_as_main        
    return _run_code(code, main_globals, None,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.1776.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\John\pybricksdev\pybricksdev\__main__.py", line 6, in <module>
    from .cli import main
  File "C:\Users\John\pybricksdev\pybricksdev\cli\__init__.py", line 10, in <module>
    import validators
ModuleNotFoundError: No module named 'validators'

dlech, 2020-11-24:
`pybricksdev` is all we have for now. It will be quite some time before Pybricks is ready for "regular users" on SPIKE Prime/Robot Inventor. We have lots of work to do to get there first.

In VS Code, I would suggest making a build task (`tasks.json`) rather than a debug configuration (`launch.json`) to run `pybricksdev ...`. Otherwise it tries to debug `pybricksdev`, which is, of course, not what you are trying to do.

johnscary-ev3, 2020-11-24:
Ok, thanks for all the help and work you do on Pybricks. Really appreciate that.

I managed to create a tasks.json file for VSC as you suggested. See below.
It works great as a VSC "build" task or "regular" task to download and run the current file in the editor.
That should keep me going for now, although I am sure I will have more questions about using Pybricks on Inventor hub ;0)

```
{
    // See https://go.microsoft.com/fwlink/?LinkId=733558
    // for the documentation about the tasks.json format
    "version": "2.0.0",
    "tasks": [
        {
            "label": "Run Pybricks",
            "type": "shell",
            "command": "pybricksdev",
            "args": [
                "run",
                "usb",
                "Pybricks Hub",
                "${file}"
            ],
            "problemMatcher": [],
            "group": {
                "kind": "build",
                "isDefault": true
            }
        }
    ]
}
```

laurensvalk, 2020-11-24:
Glad to hear you got it working @johnscary-ev3, and thanks for sharing your task settings too.

We're looking forward to your feedback. Thanks a lot everyone for trying this super early version!

laurensvalk, 2020-11-24:
> Any way it seems that users (not developers) like me would benefit by a more simple VSC extension that just connected to the Inventor hub and downloaded programs to the hub.

We'll focus on making it work with [Pybricks Code](https://www.code.pybricks.com/) first, but having a proper Visual Studio Code extension would be very nice indeed. This sounds like a good future or [sponsored](https://github.com/sponsors/pybricks) project.

laurensvalk, 2020-11-24:
This one line reproduces it for that commit: https://github.com/pybricks/pybricks-micropython/commit/47e67ba8f94fc09ec58ca00e5671276c8cf7cee0#diff-36a9b4c1bd5a5882c0c831198fc00e605390a13eb204b3f55e716699ce52a3dfR77

So I wonder if we're just getting close to allocating ever so slightly too much static variables on Move Hub.

laurensvalk, 2020-12-07:
Hello everyone! Good news - we've just finished an [initial version with Bluetooth](https://github.com/pybricks/pybricks-micropython/suites/1620249528/artifacts/29762510). I've updated the installation instructions above.

This means that you can use it with [Pybricks Code](https://www.code.pybricks.com/) now. Other than the firmware install step, everything works just like the other hubs. Now you don't need complicated Python libraries anymore. All you need is Google Chrome (or Chromium).

USB support is dropped (for now), so this may break the workflow for some of you. Fortunately, you can still use Pybrickdev if you want to. Just replace `pybricksdev run usb` with `pybricksdev run ble`. This is updated in the steps above.

We're not actively talking about this early alpha version on social media just yet. We'd love to get your feedback first.

As stated above, battery charging is not yet implemented. Although the hub will automatically safely shut down on low battery, it's still recommended to reboot into the original firmware when you're not using it so it can charge.

And for @johnscary-ev3, this means you can get remote control now as per #154 .

johnscary-ev3, 2020-12-11:
Hi,
Have been trying to use the new Bluetooth version,
I think I got latest install version dated 12/7/20 and installed it.
Seemed to go ok but when hub comes up in Pybricks mode the blue status light flashes now, 2 blinks and a pause, repeating.
I tried to connect using Bluetooth via Pybricks Code in Chrome and also using the pybricksdev program but there is no connection found. USB does not work either, but I guess that is expected as it is supposed to be BLE now. Correct?
What am I missing?

I also tried to manually connect to Bluetooth device and my PC finds device called RaLinkLinuxClient which I guess is the RI Hub?
But it asks me for a PIN which I don't know.

Thanks for any help.

laurensvalk, 2020-12-11:
> Seemed to go ok but when hub comes up in Pybricks mode the blue status light flashes now, 2 blinks and a pause, repeating.

This is correct. It blinks, which means that is waiting for a connection.

Which operating system are you using on your computer?

> I tried to connect using Bluetooth via Pybricks Code in Chrome and also using the pybricksdev program but there is no connection found.

Does your computer have Bluetooth Low Energy (BLE)? If not, you will need a BLE USB dongle, which are fairly low cost.

If you have an Android phone or tablet, you can visit Pybricks Code to see if things are working as expected.

> USB does not work either, but I guess that is expected as it is supposed to be BLE now. 

Correct.

> my PC finds device called RaLinkLinuxClient which I guess is the RI Hub

No, that is most likely something else.

BertLindeman, 2020-12-11:
[not so important]
On WIN 10 Home current Mindstorms app 10.0.3
Have been busy for days (not 24 hours a day, not by far) to get the  official Mindstorms Robot Inventor app  and the hub connected by USB. The app would not open the connect dialog.
Just now connected via bluetooth and started the install program and left the machine do its trick.
It is so hard not to press any key in the first phase of the PyBricks install. 
The Spike-hub was not so difficult, that just worked...
[/not so important]
Now finally the PyBricks firmware is in place and I can do some fun stuff.
First connect and test seem ok.

Thanks

laurensvalk, 2020-12-11:
> It is so hard not to press any key in the first phase of the PyBricks install.

Could you elaborate on this? Do you mean the lack of download progress in the app?

> The Spike-hub was not so difficult, that just worked...

What was the difference between the SPIKE and Inventor hub/app?

BertLindeman, 2020-12-11:
> 
> 
> > It is so hard not to press any key in the first phase of the PyBricks install.
> 
> Could you elaborate on this? Do you mean the lack of download progress in the app?

The time needed to load the program I think. 
After pressing the start button (not sure what the correct name is, in Dutch it is "Spelen") it looks like **nothing** is happening. The "stop" button remains gray until the uploaded program is really started. If I remember correctly 
even the start button does no animation.
So my first reaction would be: I did not press start yet. So I leave the PC after pressing start.

> 
> > The Spike-hub was not so difficult, that just worked...
> 
> What was the difference between the SPIKE and Inventor hub/app?

Not sure what you mean with this, so I try an answer..
The Spike app is current at version 1.3.2.
The Spike hub connected to the Spike hub via USB and the PyBricks install did just fine.
I did not try to update the inventor hub via the Spike App. Was not sure that this would be safe to do.
Updated the firmware of the Inventor hub via the spike app. No problem there 
Main problem I think is connect to Mindstorms app on USB.

laurensvalk, 2020-12-11:
Right - unfortunately the official app shows no download progress. I imagine this might be added at some point, since we aren't the only ones with big programs.

In the long run, we may be able to add USB support to Pybricks Code so you wouldn't need the official app to install it. And then we could show installation progress.

One of these two probably needs to happen before we can say that Pybricks is ready for SPIKE Prime / MINDSTORMS Robot Inventor. I agree that the installation is a bit akward right now.

> Main problem I think is connect to Mindstorms app on USB.

I see, thanks for clarifying! Sounds like that is not related to our install script then.

BertLindeman, 2020-12-11:
> > Main problem I think is connect to Mindstorms app on USB.
> 
> I see, thanks for clarifying! Sounds like that is not related to our install script then.

Yes, the app does not open the connect dialog, so the PyBrick program is not in the picture yet.

USB support would sure enhance the "PyBricks firmware install"

johnscary-ev3, 2020-12-11:
Hi,
After some research I figured out that my PCs do not have BLE, aka Bluetooth 4.0   ;0(
Guess they are too old.
The RI Hub does connect ok to these PCs using Bluetooth, but I guess it is using the older Bluetooth protocols for that. 
Looks like I need to get that BLE USB dongle you mentioned. Any particular one needed or suggested?
Thanks.

laurensvalk, 2020-12-12:
It's hard to guarantee which one will or will not work, but pretty much any BLE should work.

> The RI Hub does connect ok to these PCs using Bluetooth, but I guess it is using the older Bluetooth protocols for that.

Indeed, SPIKE / RI supports both Bluetooth Low Energy and Bluetooth Classic. I think the default app uses classic. All the other hubs have only Bluetooth Low Energy.

We use Bluetooth Low Energy for that reason, and also because Bluetooth Classic cannot be used with the browser like Pybricks Code does.

johnscary-ev3, 2020-12-15:
Hi,
I got an EVEO Bluetooth 5.0 dongle and it is working great for Pybricks Code and Visual Studio Code using BLE ;0)
Now I can run my RI "Charlie" around the room while it sends status info back to my PC using a Pybricks program.
Thanks very much for the help.

laurensvalk, 2020-12-15:
Glad to hear it! Looking forward to seeing what you've made :)

laurensvalk, 2020-12-15:
Update: we've changed the dual boot button combination. I'll update the instructions above.

From now on, you just need the center button so it's easier to operate with one hand.

Short press = standard firmware
Long press / hold = Pybricks firmware

laurensvalk, 2020-12-24:
Update: since a recent update of the SPIKE app, the firmware installation procedure used here will no longer work. We will be working on an alternative solution.

damodaran-ai, 2020-12-31:
Happy new year folks! Was wondering if I could ask for some quick help. 

I followed the instructions to install the firmware on the Inventor hub (Teal). Install seemed to work as per instructions.  After the long press I get to the status light blinking blue as described above (2 short flashes, pause) and the pixels are lit up in a 3x3 grid in the middle of the unit. I am unable to get this to work with pybricks code for some reason as it just sits in scanning. The official app is able to connect to the official firmware on the same Mac mini from 2020. Any ideas on how I may debug this? BLE doesn't seem to be the issue given that the machine is Bluetooth 5.0 spec and supports BLE.

laurensvalk, 2021-01-01:
Happy new year to you too! Thanks for trying Pybricks.

Since it seems to work fine on Windows but not on OSX, I think I’ll move your comment to a separate issue so we can better help you and further investigate.

laurensvalk, 2021-01-04:
We have just updated the installation instructions. Due to changes in how the official apps work, there is now a separate `download_pybricks` project and a `install_pybricks` project.

Please let us know if you run into any issues. Confirmation that it works on both hubs would be highly appreciated as well.

laurensvalk, 2021-01-12:
Since recent updates in the official app, this doesn't seem to work. I've updated the post accordingly.

To make sure things become a bit more stable, we will probably be developing our own installation tool at some point, so please stay tuned. :)

BertLindeman, 2021-01-12:
**Small thing**

In the instructions above we "have" to unplug the USB cable. Did not see it getting plugged-in ;-) 

> 2: Starting Pybricks
    Turn off the hub and unplug USB

The first update of the official firmware needs USB, but after that I think USB is not **needed** anymore.

**even smaller thing**

Would it be a good idea to have this instruction on the code.pybricks doc web pages?
Each time I want to follow these instructions,  I have to search for then. (Sorry, my memory)  

I would accept the rationale: it is not stable enough to do this

laurensvalk, 2021-01-12:
> Would it be a good idea to have this instruction on the code.pybricks doc web pages?
> Each time I want to follow these instructions, I have to search for then. (Sorry, my memory)
> 
> I would accept the rationale: it is not stable enough to do this

That's indeed why they're here for now. For now, it's only for hackers who don't mind reading through Github discussions :smile:

We really need to make it a lot easier before we put it on Pybricks Code.

BertLindeman, 2021-01-12:
You **are** making steps. Thanks.
The split of the download and the install of the firmware **does** make it more clear.

Install upon the spikehub (minor so did not want a new issue for this):
Hub has current LEGO firmware (updated a few minutes ago)

- first phase shows progress on the display until all pixels are lit.
- the hub seems to be shutdown (no button/pixel lighted)
- pressed (and kept for a few seconds) the center button.
- bluetooth light varying in color
- the hub switched off.
- long-press starts pybricks firmware and shows firmware a13

 In the end it works oke,

laurensvalk, 2021-01-15:
I updated the post above  --- it should now work again on the Inventor Hub as well.

BertLindeman, 2021-01-15:
Re-did the download / install upon the Spikehub.
Similar as my previous install the primehub did not auto reboot after all pixels were lit at the end of the install.

 Pressed and hold the main button until bluetooth light got green and then blinking different colors.
After that also no auto reboot.
Short press starts LEGO firmware. Long press starts pybricks firmware.
Next the inventor hub.
The same procedure as the spike hub.
No auto reboots but fine dual boot after that.

But strangely the inventorhub reports it's firmware as
```
('primehub', '3.0.0a13', 'v3.0.0a13-5-g9a80bef6 on 2021-01-15')
```

laurensvalk, 2021-01-16:
They both use the same firmware, so that seems about right :)

BertLindeman, 2021-01-16:
> They both use the same firmware, so that seems about right :)

Oke. I used to look at the hub name in the firmware to determine what hub to instantiate.
Like so:
```
hw_type = version_tuple[0]

if str(hw_type) == "technichub":
    from pybricks.hubs import TechnicHub
    hub = TechnicHub()
elif str(hw_type) == "cplushub":
    from pybricks.hubs import CPlusHub
    hub = CPlusHub()
elif hw_type == "inventorhub":
    from pybricks.hubs import InventorHub
    hub = InventorHub()
elif hw_type == "primehub":
    from pybricks.hubs import PrimeHub
    hub = PrimeHub()
```
Sorry, side-stepping...

Just re-did the inventor hub now with certainty that the battery is full.
No change.

**Maybe add in the instructions to open the console window.**

I was a bit disappointed that I could not copy the text from the console. 
But I cannot blame you guys for that, pitty :-)  You would have fixed that in no-time.
So screenshot:
[image]

Two remarks about the environment: 
The Spikehub has been done full battery and without any connected devices.
The Inventor hub also full battery but I did that in a model, so with attached sensors and motors.

[image]

laurensvalk, 2021-01-16:
> I used to look at the hub name in the firmware to determine what hub to instantiate.

And that's still fine. In Pybricks, `InventorHub` is just an alias for `PrimeHub`. Both names are available on both hubs; they do the same.

You can use the color sensor to self-identify which one it is :smile:

BertLindeman, 2021-01-16:
>  In Pybricks, InventorHub is just an alias for PrimeHub. Both names are available on both hubs; they do the same.

Good to know the alias.
Maybe sometime in the future good for a line in a doc item 

> You can use the color sensor to self-identify which one it is 😄

HA ha ha hi hi . . . . Great idea Laurens, Scan the color of the hub to be sure and stop the program if its the wrong color.

laurensvalk, 2021-01-23:
The installation link above has been updated to `firmware.v3.0.0b1`.

laurensvalk, 2021-01-23:
> > In Pybricks, InventorHub is just an alias for PrimeHub. Both names are available on both hubs; they do the same.
> 
> Good to know the alias.
> Maybe sometime in the future good for a line in a doc item

There is now a [single page for both hubs](https://docs.pybricks.com/en/latest/hubs/primehub.html) to make things simpler.

BertLindeman, 2021-01-29:
Part of my comment from [some time ago in this issue](https://github.com/pybricks/support/issues/167#issuecomment-759007876)

> Install upon the spikehub (minor so did not want a new issue for this):
> Hub has current LEGO firmware (updated a few minutes ago)
>
> * first phase shows progress on the display until all pixels are [...]

Have re-installed the current pybricks firmware on the spike primehub again today.
If the hub is connected via USB to the spike app the auto reboot **does** work.
That did not work for me if bluetooth connected.

Maybe update the install instructions above from

> 4. Run the download_pybricks project in the app.

to
4. Connect the hub via USB to the app and run the download_pybricks project in the app.

That also validates the "unplug USB" later in the instructions.
USB might even be a bit faster? I did not time....

ws1088, 2021-01-31:
[image]
I got this error while pressing the middle button (after i successfully run the program from the IDE)

dlech, 2021-01-31:
> I got this error while pressing the middle button (after i successfully run the program from the IDE)

See https://github.com/pybricks/support/issues/234

BertLindeman, 2021-02-03:
Due to #250 

I think it would be good to add in the Getting started doc
**after this**
```
Just wait a few minutes until you hear the usual start sound.
The program stops immediately because it is just data that does nothing by itself.
```
**add this**
```
Open the console log to verify no error occurred while downloading the firmware data
```

That would ensure the console is open in the next step, so progress can be seen there.

laurensvalk, 2021-02-03:
Thanks --- the installation method is going to be simplified so that this should no longer be an issue.

BertLindeman, 2021-02-03:
> 
> 
> Thanks --- the installation method is going to be simplified so that this should no longer be an issue.

NOW I see why you want the console log **closed** on the download.

By the way now I did the download a few times: via USB is only a bit faster than via bluetooth
one minute 45 versus just over two minutes.

laurensvalk, 2021-02-03:
Thanks for the time data. It won't be much faster through Pybricks Code, but we will then be able to start lighting up the LEDs *during* the download already, so it's easier to be patient :)

[28 more replies]

closed 2022-01-14 by laurensvalk