pybricks/support #168: [Question] Questions and Observations using Pybricks on Robot Inventor Hub

johnscary-ev3, 2020-11-28:
I don't want to sound like I am complaining since I know you guys are working hard to get Pybricks working on Inventor Hub.
However I have a few questions and observations after using this for a few days with VSC. Hopefully some feedback is useful.

1. Since we just have USB for now, I noticed that my program hangs when I disconnect the USB, I suspect this is due to print statements in the program. Program restarts when I connect the USB but print outputs don't show up anymore. Is this normal?
2. Is there, or will there be, a mechanism to restart a program from the hub or have multiple programs in the hub?
   Seems like I need to download it to rerun a program.
3. Looks like we don't have a way to do remote control now since no IR control like EV3. Any plan for remote controls?
   I am using the color sensor now to give commands but that is limited to a few colors, and you need to chase the robot around.
4. No capability to make sounds now. Will that be added?
5. Color Sensor seems to detect all the colors of LEGO bricks I have except the new Cyan ones that come with Robot Inventor.
   We have the Cyan color defined, but the sensor calls those bricks "blue".

Thanks for any comments.

laurensvalk, 2020-11-28:
Thanks for trying this out. Those are some good questions.

> 1. Since we just have USB for now, I noticed that my program hangs when I disconnect the USB, I suspect this is due to print statements in the program. Program restarts when I connect the USB but print outputs don't show up anymore. Is this normal?

For the moment, please disregard USB/connectivity issues. We are working to make it work just like the other hubs, using BLE and [Pybricks Code](https://www.code.pybricks.com/). When that's ready, we're probably going to improve reliability for all hubs at once.

We're not actually sure what we'll do with USB yet. Easy and fast access to the file system maybe?

> 2. Is there, or will there be, a mechanism to restart a program from the hub or have multiple programs in the hub?

There will be. Currently, the user flash storage is not yet enabled, so that's why there's only Download & Run for now. See #139 for how it might work. Let us know if you have additional ideas on that one.

[image]

> 3. Looks like we don't have a way to do remote control now since no IR control like EV3. Any plan for remote controls?
I am using the color sensor now to give commands but that is limited to a few colors, and you need to chase the robot around.

Once Bluetooth is working for running programs, there are a lot of exciting possible next steps. There's the remote that comes with the trains, or even gamepads at some point.

See also #154, but you may have to wait for Pybricks Code for that.

> 4. No capability to make sounds now. Will that be added?

Yes, we plan add it, but low priority for the moment.

> 5. Color Sensor seems to detect all the colors of LEGO bricks I have except the new Cyan ones that come with Robot Inventor.
We have the Cyan color defined, but the sensor calls those bricks "blue".

This is fully configurable. See [the Color Sensor](https://docs.pybricks.com/en/latest/pupdevices.html#color-sensor) and check out the second example under "Advanced color sensing", called "Changing the detectable colors". Based on your questions about the EV3 sensor some time ago, I think you'll like this feature :smile:

johnscary-ev3, 2020-11-29:
Thanks for detailed reply to my feedback.

This all sounds great!
Bluetooth support and program storage will be very nice.
Remote using your phone would be good also.
Having a way to see what programs are in each "slot" would be good in case you forget.
Minimal sounds like just a "beep" function would go a long way.

I will work on the pybricks "advanced color sensing" with respect to the LEGO Cyan color bricks.

I have also been using the LEGO Inventor App Python, so some of these comments come from a comparison to that.
I think Pybricks motor controls have more capabilities, but LEGO Python App has some of these other features.
(Not the remote control in Python yet, as far as I can tell.)
I think others will probably also be comparing Pybricks to LEGO Python now to decide what to use.

johnscary-ev3, 2020-11-29:
Advanced color sensing with detectable colors works great!
I can detect all my brick colors now including cyan and also gray ;0)
Thanks for the suggestion.

I am look forward to pybricks BLE and program storage for Inventor hub.
Will the BLE also work with the pybricksdev downloader program so we can use that from VSC?

laurensvalk, 2020-11-30:
> Advanced color sensing with detectable colors works great!
I can detect all my brick colors now including cyan and also gray ;0)

Glad to hear it!

> Will the BLE also work with the pybricksdev downloader program so we can use that from VSC?

Yes, it's already in there for the other hubs :smile:

laurensvalk, 2020-12-18:
> No capability to make sounds now. Will that be added?

Very basic sound was added last night, and [documented too](https://docs.pybricks.com/en/latest/hubs/inventorhub.html)!

There's no support for volume control or files yet, but you can debug your code entirely with beeps now :robot: .

johnscary-ev3, 2020-12-19:
Hi,
Thanks for working on adding sounds.

I downloaded latest RI firmware I could find    "firmware-version": "v3.0.0a12-1-gaeac9a68" files dated 12/11/2020.
This one does not seem to recognize the speaker yet.  I get this error:
AttributeError: 'InventorHub' object has no attribute 'speaker'
on this statement:
hub.speaker.beep(freq, duration)

Is there a newer firmware?

BertLindeman, 2020-12-19:
@johnscary-ev3 
Have a look at: [Issue 179 ](https://github.com/pybricks/support/issues/179#issuecomment-748335999)

johnscary-ev3, 2020-12-19:
@BertLindeman 
Thanks for the guidance on this.
Now I found the latest builds location.
Downloaded latest one "v3.0.0a12-13-g46d1da0e" and now I have beeps ;0)

BertLindeman, 2020-12-19:
> Thanks for the guidance on this.
> Now I found the latest builds location.
> Downloaded latest one "v3.0.0a12-13-g46d1da0e" and now I have beeps ;0)

I cannot remember the link myself, so I refer to the link above to look for brand new firmware ;-)
I have the beeps too (on the inventorhub)

laurensvalk, 2020-12-19:
We’ll make sure to add it to the beta instructions, thanks! 😄

johnscary-ev3, 2020-12-20:
I have another question, this time related to the Bluetooth and 30 sec power down function.
Not sure if this is a new thing but I noticed now that if there is no Bluetooth connection for 30 sec, the hub will power down.
When using Pybricks Code environment, this is generally not a problem since you connect once and it holds the connection open even when you are not running a program.

However, when using the pybricksdev program under Visual Studio Code, it seems to open the connection to send and run a program but when the program terminates, it disconnects the Bluetooth (blue light starts blinking) and after 30sec the hub powers down. So I find myself constantly restarting it. 

Can this time out behavior be controlled by us users, or am I doing something wrong?
Would be better if it was like 10 min instead of 30sec.
Thanks.

laurensvalk, 2020-12-20:
I agree, 30 seconds is pretty short; I'd prefer to leave it on for a few minutes too, even when using Pybricks Code.

laurensvalk, 2020-12-28:
Timeout fixed via https://github.com/pybricks/pybricks-micropython/commit/d2f7e65ff2f72009f87e9e0a71c9829c55aa6b26

johnscary-ev3, 2020-12-29:
@laurensvalk   Thanks very much  ;0)

laurensvalk, 2021-07-13:
Most of the feature requests here are either supported by now, or covered by other issues like #139, so we can close this issue.

closed 2021-07-13 by laurensvalk