pybricks/support #170: [Bug] PyBricks Code and REPL has issue on Windows Chrome using Paste Mode

GusJansson, 2020-12-07:
**Describe the bug**
Using Pybricks Code on Chrome running on Windows.  Trying to use paste mode but when I hit Ctrl-D to finish the Paste Chrome intercepts the keystroke and opens a Edit Bookmarks page.

**To reproduce**
Steps to reproduce the behavior:
1. Using Chrome running on Windows start PybricksDev.
2. Select some code and copy to clipboard, I used:
```python
from pybricks.hubs import InventorHub
from pybricks.pupdevices import Motor, ColorSensor
from pybricks.parameters import Port, Button, Stop
from pybricks.tools import wait

# Initialize the hub, motors, and sensor
hub = InventorHub()
steer_motor = Motor(Port.A)
drive_motor = Motor(Port.B)
sensor = ColorSensor(Port.C)

def WaitForButton(b):
    # Wait for press
    while b not in hub.buttons.pressed():
        wait (10)
    # and release
    while b in hub.buttons.pressed():
        wait (10)

# Use the color saturation value to track line
def GetLight():
    return (sensor.hsv().s)
```
3. Go to REPL pane and get a command prompt with an Inventor Hub connected via BT.
4. Enter Paste mode by using Ctrl-E
5. Paste the code from the clipboard using Right Click and selecting Paste.
6. Hit Ctrl-D to finish Paste mode.  This opens Edit Bookmarks Page on Chrome.

My impressions is that the problem is that when I paste using Right Click the REPL pane looses focus and that is why it does not receive the Ctrl-D.  If I paste with Ctrl-Shift-V then the REPL pane maintains focus and Ctrl-D works.

**Expected behavior**
I expected Paste mode to finish and not REPL not loose focus.

Also, why does the REPL pane require Shift in Paste keyboard shortcut.  I would expect Ctrl-V to also paste.

**Screenshots**
[image]

closed 2020-12-10 by dlech

dlech, 2020-12-10:
Thanks for the detailed bug report. We were able to reproduce the issue and fix it.

dlech, 2020-12-10:
> Also, why does the REPL pane require Shift in Paste keyboard shortcut. I would expect Ctrl-V to also paste.

To match <kbd>CTRL</kbd>+<kbd>SHIFT</kbd>+<kbd>C</kbd> for copy.  <kbd>CTRL</kbd>+<kbd>C</kbd> has special meaning to interrupt the program in the REPL (and in terminals in general), so we can't use it to mean copy.