pybricks/support #171: Error running "make primehub" command

eYoung8475, 2020-12-08:
I run the "make primehub" command under Ubuntu16.04 and the following error occurs

metadata.py: error: argument <hub-type>: invalid choice: '-mno-unicode' (choose from 'technic_hub', 'prime_hub', 'city_hub', 'move_hub')
../stm32/stm32.mk:610: recipe for target 'build/firmware.metadata.json' failed
make[1]: *** [build/firmware.metadata.json] Error 2
make[1]: Leaving directory '/micropython/pybricks-micropython/bricks/primehub'
Makefile:104: recipe for target 'primehub' failed
make: *** [primehub] Error 2

dlech, 2020-12-08:
A couple of ideas:

1. Ubuntu 16.04 is a bit old. It could be that we are relying on features that require a newer version of `make`. We use Ubuntu 20.04 for development, so we know that version works.
2. You could try running make with `V=1` set to get more verbose output.

eYoung8475, 2020-12-09:
I recompile in ubuntu20.04，it is ok， thank you very much！

closed 2020-12-09 by eYoung8475