pybricks/support #173: [Feature] Next steps for Pybricks Code editor

laurensvalk, 2020-12-11:
We've been keeping [Pybricks Code](https://www.code.pybricks.com/) very simple so far. This post is to collect some ideas for next steps. This can be split up into dedicated issues as we go along.

- Progress indicator:
   - Firmware download progress
   - Program download progress (only needed for scripts greater than certain size)
   - Either as pop up, or built-in as we have today, just more visible.
- Connection status:
   - Connection type
   - Hub type
   - Hub name
   - Hub battery voltage
- Regroup buttons by function and add settings context menu.
- Code Tabs:
  - More than one tab will make it much easier to code and test out different snippets and experiments.
  - To keep things simple, there's no file manager; closing the tab discards the script.
  - Save downloads current tab. Open creates new tab.
  - Doc samples can have a button "open in tab" for easy experimentation
- Toggle to include current script when updating firmware
- Dark mode
- Python code completion engine
    - Syntax error highlighting
    - Ships with pybricks-api

[image]

BertLindeman, 2020-12-19:
@laurensvalk Might well be that I misread the features, but ...
Does the "save" of a tab rewrite the file that was opened into that tab?

Rationale:
At the moment I have a separate editor (notepad++)  with the python code I am using in pybricks.
Do numerous "select-all copy" followed by "select-all paste"
then run flake8 this-python.py
etc...
With the risk that I do the copy/paste the wrong way.
The sources now are bare small snippets, but later . . . . . ;-)

laurensvalk, 2020-12-19:
Thanks for the input. I think it’s important to keep in mind that we do not intend to make Pybricks Code a fully fledged editor. It’s designed to make programming very accessible for beginning users.

We might possibly reserve more advanced use cases for a future Visual Studio Code extension similar to the ev3dev extension. That way, we don’t have to reinvent the wheel as far as file management, linting, etc are concerned.

We’re still a bit more focused on the firmware side for now, but this may change towards editors and extensions once the firmwares are fairly complete.

BertLindeman, 2020-12-19:
>  
> We’re still a bit more focused on the firmware side for now, but this may change towards editors and extensions once the firmwares are fairly complete.

Understood @laurensvalk No intention to push. Just wanted to offer my input before it's too late ;-) .

laurensvalk, 2020-12-19:
No worries. The more you ask, the faster they get done, so please keep asking :smile:

BertLindeman, 2020-12-19:
> 
> 
> No worries. The more you ask, the faster they get done, so please keep asking 😄

Oke, here goes ;-)

1. built-in flake8
2. pre-compile verification for the movehub

Just mildly kidding
You guys are **so** helpful

BertLindeman, 2021-01-15:
Looks like dark mode is implemented [via](https://github.com/pybricks/support/issues/48#issuecomment-760564333) and works.

Added issue #213 (linked here as code completion is in this item)

dlech, 2021-04-02:
Suggested in https://github.com/pybricks/support/discussions/278, show compiled size of user program.

dlech, 2021-04-02:
Suggested improved firmware flash experience:

1. User clicks flash button in IDE or drops custom firmware on the button.
2. If this is custom firmware, checks are done first to ensure it is "good" firmware.
2. Opens Web Bluetooth selection dialog and start scanning. Currently, this only shows hubs in bootloader mode. Improved will show hubs not in bootloader mode as well running Pybicks firmware or official LEGO firmware. TODO: also need to consider USB for SPIKE Prime/MINDSTORMS Inventor.
3. User selects hub.
4. Connects to hub.
5. If hub is not supported (e.g. Duplo train), show appropriate "sorry" message.
6. If this is custom firmware, additional checks are done to ensure the hub is compatible.
7. If the firmware version is already the latest, ask if they really want to continue. The only reason to do this is to change the name or the user program. There is a common belief that flashing the same firmware again can fix bugs (probably comes from NXT days where this was actually true since critical files could accidentally be deleted from the NXT) but this is not true for Pybricks since it resides in read-only flash memory (unless a cosmic ray flipped a bit).
6. If hub is supported, show the firmware flash dialog.
  - Has text box to enter hub name (If the hub is not in bootloader mode, the current hub name will be filled in here, otherwise the default "Pybricks Hub" is used. We could potentially also keep a table of hub serial number to names to "remember" the name in bootloader mode in the browser local storage)
  - Has widget to select LED color. (Like the name, this should be the current color, if possible, otherwise default to blue)
  - Has switch to select including the current user program when flashing the firmware or not.
  - Explains that Pybricks is 3rd party firmware.
  - Has text expressing that by flashing firmware you are agreeing to license agreements with link/button to show full firmware license text.
7. User can cancel or continue
8. The user program is compiled, if needed. An error message is shown if it fails.
8. If the hub was not already in bootloader mode, it reboots into bootloader mode. (Due to technical restrictions in Web Bluetooth, this means showing the Bluetooth scan dialog again and the user has to select the "LEGO Bootloader" from the list.
9. Firmware is flashed to the hub.
10. Hub restarts.
11. User must manually click the Bluetooth button to connect.

Additional changes to Bluetooth connection:

1. User clicks Bluetooth button to connect to hub.
2. Scan dialog pops up and shows hubs running Pybricks firmware or official LEGO firmware.
3. User selects hub and clicks OK
4. Connect to Hub.
5. If hub is official LEGO firmware, explain that Pybricks is 3rd party firmware. Use the same flash firmware button in the dialog box and suggest to user to click it to flash Pybricks. This way they will know what button to look for later. This jumps to the step in the firmware flash sequence where the hub is already connected.
6. If hub is Pybricks firmware, check if it is out of date. If Pybricks protocol version is too old (or new), require firmware update. If Firmware version is old, suggest, but don't require update. If the user opts to update the firmware, jump to the step in the firmware update where the hub is already connected.

closed 2022-08-01 by dlech