pybricks/support #174: [Question] How to contribute to pybricks-code repo?

littldr, 2020-12-13:
Hi!

First of all: Huge thanks for creating such a great project!

**Question**
I was facing the same issue as described in [this issue](https://github.com/pybricks/pybricks-code/issues/161) and wanted to help fixing it.
But I cannot access all dependencies to run the code. It looks like that the `@pybricks` namespace on npmjs.com is private.

Is there any chance to get access to these dependencies?

**Context**
I want to help fixing issues :)

**Screenshots**
[image]

dlech, 2020-12-13:
Awesome! Super excited to get more people contributing here.

The issue you are running into is due to the fact that the packages are hosted on GitHub rather than NPM and GitHub requires authentication even though the packages are "public". 

Did you follow the instructions to generate a new GitHub access token and login in https://github.com/pybricks/pybricks-code/blob/master/CONTRIBUTING.md? If there aren't clear enough, we would gladly take a PR to improve that as well.

littldr, 2020-12-13:
ohh, my mistake!

I scanned the `CONTRIBUTING.md` too fast and never used Github Packages before. 🙈 

Maybe reordering the `CONTRIBUTING.md` sections and putting the Github package dependency on top would help a bit. 😅 
I will give it a try and propose a PR. :)

Thanks for the fast response!

closed 2020-12-13 by littldr