pybricks/support #175: [Bug] Problem installing beta firmware onto MoveHub

BertLindeman, 2020-12-13:
This issue was first entered in #43. 

Not sure how to identify the current beta firmware. The movehub files were stamped "2020-12-11 10:34"

**Describe the bug**
Various symptoms while installing firmware on MoveHub 

**To reproduce**
I would be very glad if there was something consistent in this.

**Expected behavior**
Hub flashing shows progress in progress-bar
Flash completes and the hub goes to flashing blue.

Copied from #43 and updated a bit:

Installed this firmware onto the movehub (as it is the most restricted hub)

Running Windows10 Home.
At (or after?) the start of firmware flash an error on Chrome (Version 87.0.4280.88 (Official Build) (64-bit)) console:
```
index.js:511 Uncaught TypeError: S.current.contains is not a function
    at HTMLDocument.ee (index.js:511)
ee @ index.js:511
```

The web page looks like this:
[image]

The hub keeps blinking red-green-blue **as if** downloading the firmware and I awaited 20 minutes to no avail.

This is not the first time I installed this same firmware on the same hub.
After an attempt to start the hub with the color sensor connected, the connect did not succeed.
The hub switched to steady blue but the web page stayed like this:
[image]

So had to report it...

trying to go back to regular firmware fails the same way now. In both cases NO progress in progress-bar. 
No error at the moment, the same as on flash of beta firmware.

Re-loaded batteries (even if I just did that) to be sure and re-try.
No success.

Reboot PC and re-try to flash the beta firmware NOW works.

So I can make no sense out of this.

@laurensvalk if you decide: "close this rubbish issue" then I will agree.

laurensvalk, 2020-12-13:
Were there any motors or sensors connected to the hub when you encountered this?

BertLindeman, 2020-12-13:
> 
> 
> Were there any motors or sensors connected to the hub when you encountered this?

The problems started after I tried to connect the movehub with on port-C 
That connect failed and after that  all attempts to re-flash were  with all ports **not** connected.

dlech, 2020-12-18:
This sounds like duplicate of #178

BertLindeman, 2020-12-18:
> 
> 
> > The hub keeps blinking red-green-blue as if downloading the firmware and I awaited 20 minutes to no avail.
> 
> This sounds like duplicate of #178

Let's say so ;-) 
This is a not so helpful problem report, either.

Thanks @dlech

closed 2020-12-18 by BertLindeman