pybricks/support #176: [Bug] PyBricks web page does not connect to movehub after memory problem

BertLindeman, 2020-12-14:
**Describe the bug**
PyBricks web page does not connect to movehub after memory problem running a too large program on the hub.

Related question:
How to reset the movehub after a python memory problem

**Environment**
windows 10 Home update 20H2
Google Chrome Version 87.0.4280.88 (Official Build) (64-bit)
PyBricks beta firmware dated 2020-12-11 10:34

No motors nor sensors connected to the movehub.

**To reproduce**
Sorry for so much text.

Steps to reproduce the behavior:
1. On win10 open on Google chrome code.pybricks.com
2.  connect to the movehub (at firmware v3.0.0a11-34-gd00427d5 on 2020-12-11 )
3. Load program 1 (below)
   Runs oke for a few times (did not count and remember).
 4. Output:
```
DEBUG = True
movehub
3.0.0a11
v3.0.0a11-34-gd00427d5 on 2020-12-11
('movehub', '3.0.0a11', 'v3.0.0a11-34-gd00427d5 on 2020-12-11')
Running on hw_type: movehub
No help(micropython)
.const()      9
 
no .opt_level()
 
stack: 3758080460
GC: total: 6016, used: 1824, free: 4192
 No. of 1-blocks: 38, 2-blocks: 5, max blk sz: 36, max free sz: 231
.mem_info()   None
 
qstr pool: n_pool=1, n_qstr=15, n_str_data_bytes=145, n_total_bytes=241
.qstr_info    None
.stack_use    -536886868
None
```

5. Add some text to the end of the program:
```
print("just some text to make the program longer, so the hub might get a memory problem")
print("2 just some text to make the program longer, so the hub might get a memory problem 2")
print("3 just some text to make the program longer, so the hub might get a memory problem 3")
print("4 just some text to make the program longer, so the hub might get a memory problem 4")

a[1] = "just some text to make the program longer, so the hub might get a memory problem"
a[2] = "2 just some text to make the program longer, so the hub might get a memory problem 2"
a[3] = "3 just some text to make the program longer, so the hub might get a memory problem 3"
a[4] = "4 just some text to make the program longer, so the hub might get a memory problem 4"

a[11] = "11 just some text to make the program longer, so the hub might get a memory problem 11"
a[12] = "12 just some text to make the program longer, so the hub might get a memory problem 12"
a[13] = "13 just some text to make the program longer, so the hub might get a memory problem 13"
a[14] = "14 just some text to make the program longer, so the hub might get a memory problem 14"
```
6. Run the now larger program.
7. Now the program does **not** end normally but the hub stays steady green until some time later as it goes into blinking blue.
8. output is terminated like this.
```
DEBUG = True
movehub
3.0.0a11
v3.0.0a11-34-gd00427d5 on 2020-12-11
('movehub', '3.0.0a11', 'v3.0.0a11-34-gd00427d5 on 2020-12-11')
Running on hw_type: movehub
No help(micropython)
.const()      9
 
no .opt_level()
 
stack: 3758080460
GC: total: 6016, used: 3312, free: 2704
 No. of 1-blocks: 50, 2-blocks: 5, max blk sz: 42, max free sz: 100
.mem_inf
```

9. The code webpage shows a normal blue not-connected bluetooth button.
**Looks** all ok now, but it is **not**

10. Try to reconnect:
   Hub goes to steady blue but the webpage does not show a complete BT connection. 
   No bluetooth icon on the top tab,
   only light-blue bluetooth button.
   Looks like the recovery in the hub is OK but the webpage is not.
[image]

11. After some time the hub goes to blink-blue and switches immediately to steady-blue 
    code.pybricks.com shows a blue BT button (state: not connected) 
    Pressing the BT-connect button does not show the movehub to be paired.

12. Press hub button until off, wait and then re-press - goes to steady blue without blinking.
    While hub is steady blue, a short press on the hub button makes the hub steady green for 2 or 3 seconds.
  
13. Going in circles now, so reboot pc to get things sorted ;-(

**Expected behavior**
Connected movehub  after reset of hub and [ctrl]-F5 on web page

**Python source code 1** (not yet too large)
```
# source at:
# Z:\home\bert\py\pybricks\debug_movehub\debug_memory.py

# noqa  E722 do not use bare 'except'

DEBUG = True
print("DEBUG =", DEBUG)

try:
    import uerrno
    help(uerrno)
except:
    pass

from pybricks import version
version_tuple = version
if DEBUG:
    print(version_tuple[0])
    print(version_tuple[1])
    print(version_tuple[2])

print(version_tuple)
hw_type = version_tuple[0]

if (DEBUG):
    print("Running on hw_type:", hw_type)

if str(hw_type) == "cplushub":
    from pybricks.hubs import CPlusHub
    hub = CPlusHub()

if hw_type == "movehub":
    from pybricks.hubs import MoveHub
    hub = MoveHub()

try:
    help('modules')
except:
    pass

try:
    import micropython
except:
    print("No micropython import")

try:
    help(micropython)
except:
    print("No help(micropython)")

print(".const()     ", micropython.const(9))

print(" ")
try:
    print(".opt_level() ", micropython.opt_level())
except:
    print("no .opt_level()")

print(" ")
try:
    print(".mem_info()  ", micropython.mem_info())
except:
    print("No micropython.mem_info()")

print(" ")
try:
    print(".qstr_info   ", micropython.qstr_info())
except:
    # print("No micropython.qstr_info()")
    print("q")

try:
    print(".stack_use   ", micropython.stack_use())
except:
    print("s")

#  heap_lock -- <function>
#  heap_unlock -- <function>

try:
    print(micropython.kbd_intr(0))
except:
    print("k")
```

dlech, 2020-12-14:
> PyBricks

We are not fancy. It is "Pybricks" rather than "PyBricks".

> How to reset the movehub after a python memory problem

If pressing and holding the button does not turn off the hub, pull the battery to reset the hub.

>     2\. Now the program does **not** end normally but the hub stays steady green until some time later as it goes into blinking blue.

It sounds like we have more work to do to protect against stack overflow.

BertLindeman, 2020-12-14:
> > PyBricks
> 
> We are not fancy. It is "Pybricks" rather than "PyBricks".

;-) Sorry The Bricks are important to us, right....

> If pressing and holding the button does not turn off the hub, pull the battery to reset the hub.

Will try your method, thanks.

This item has not a very high priority to me, I just started with the movehub as it is the most restricted one.

BertLindeman, 2020-12-14:
Small update:

These two browsers also show the same problem on my win10 pc:

Microsoft Edge is bijgewerkt.
Versie 87.0.664.60 (Officiële build) (64-bits)

Chrome development: 
Google Chrome is up to date
Version 89.0.4350.6 (Official Build) dev (64-bit)

After a memory problem on the movehub, even the inventorhub does not connect.

As far as I know the only to get out of the situation is PC reboot.
After reboot, the inventorhub and the movehub connect fine.

Does the web code keep a status around that is refreshed / deleted on PC reboot?
Just thinking alound without the backing knowledge ;-)

dlech, 2020-12-14:
Closing and opening the browser should be enough to fix Pybricks Code without rebooting the entire computer.

<kbd>CTRL</kbd>+<kbd>F5</kbd> should be enough to fix Pybricks Code without even restarting the browser.

BertLindeman, 2020-12-14:
> 
> 
> Closing and opening the browser should be enough to fix Pybricks Code without rebooting the entire computer.
> 
> CTRL+F5 should be enough to fix Pybricks Code without even restarting the browser.

Thanks David. **should** is the important word here.

That is not what I see here. 
Neither close/open nor CRTL_F5 does the trick here.

dlech, 2020-12-14:
Hmm... if that is the case, then there is probably a bug in Windows.

BertLindeman, 2020-12-16:
Maybe this adds a clue?

In the problem state as describrd above, trying to get the inventorhub connected (firmware a12 from yesterday) shows:

[image]

while the same hub neatly connects to linux code.pybricks.com

BertLindeman, 2021-01-09:
After update to firmware a13 still a problem.

My feeling is that the hub has a bit more storage available.
The first time I ran my test program it finished OK and I could run it a second time in which it choked.

Had to reboot win10 to get a normal situation again.

BertLindeman, 2021-01-14:
Maybe a circumvention on win10.

Keep in mind that I do not have any Bluetooth device on my PC connected, only the hub(s) we play with!
So I can easily switch OFF bluetooth.

1. In the situation where

- the movehub runs and gets a memory problem and after some time disconnects,
- code.pybricks.com "connect button" seems to work but the buttons remain light blue:

[image]

- The hub shows a steady blue light: connected. 

2. try to repair by switch-off bluetooth and ON again after a few seconds:
[image]

**Result**
code.pybricks reports an unexpected error and the hub shows a blink-blink blue light.
"connect" on code/pybricks after a few seconds allows connect to the movehub
and all looks OKE:
[image]

No need to re-boot the PC.
"Just" recycle bluetooth on the PC. A lot faster than reboot ;-) 

Can I close this issue? does not look like a pybricks bug.

dlech, 2021-01-14:
Yes, no sense in keeping it open if there is nothing we can do about it.

BertLindeman, 2021-01-14:
> 
> 
> Yes, no sense in keeping it open if there is nothing we can do about it.

Right, so closed.
Thanks David

closed 2021-01-14 by BertLindeman