pybricks/support #177: [Question] Communications demo - a few issues

Nelvarion, 2020-12-15:
I'm playing around with the Communications demo to share input/output between hubs and ran into a few issues.

1. Running `poetry install` fails with the following error:
```
[SolverProblemError]
Because pybricksdev (0.1.0) depends on bleak (^0.9.1) which doesn't match any versions, pybricksdev is forbidden.
So, because pybricksdev-demo depends on pybricksdev (0.1.0 git), version solving failed.
```
I resolved this by manually  installing bleak 0.9.1 `poetry add bleak="0.9.1"` and then running `poetry install` a second time.

2. In each of the sections that will run on the hubs, the first line creates the file in the build folder
`%%file build/remote.py`
However, the build folder doesn't exist so the cell fails
I suggest to add an empty `build` folder in the repo so it gets cloned as well.

3. The last cell that runs both hubs and the computer scripts fails for me

```
NotImplementedError                       Traceback (most recent call last)
<ipython-input-11-ea080da4f02c> in <module>
----> 1 await gather(
      2     standalone.run('build/standalone.py'),
      3     train.run('build/train.py'),
      4     forwarder(standalone, train)
      5 )

c:\users\user\documents\code\pybricksdev-demo\.venv\src\pybricksdev\pybricksdev\connections.py in run(self, py_path, print_output)
    285 
    286         # Compile the script to mpy format
--> 287         mpy = await compile_file(py_path)
    288 
    289         # Get length of file and send it as bytes to hub

c:\users\user\documents\code\pybricksdev-demo\.venv\src\pybricksdev\pybricksdev\compile.py in compile_file(path, compile_args, mpy_version)
     72 
     73     # Get version info
---> 74     out = await run_mpy_cross(["--version"])
     75     installed_mpy_version = int(out.strip()[-1])
     76     if mpy_version is not None and installed_mpy_version != mpy_version:

c:\users\user\documents\code\pybricksdev-demo\.venv\src\pybricksdev\pybricksdev\compile.py in run_mpy_cross(args)
     38 
     39     # Run the process asynchronously
---> 40     proc = await asyncio.create_subprocess_exec(
     41         mpy_cross.mpy_cross, *args,
     42         stdout=asyncio.subprocess.PIPE,

c:\program files (x86)\python38-32\lib\asyncio\subprocess.py in create_subprocess_exec(program, stdin, stdout, stderr, loop, limit, *args, **kwds)
    234     protocol_factory = lambda: SubprocessStreamProtocol(limit=limit,
    235                                                         loop=loop)
--> 236     transport, protocol = await loop.subprocess_exec(
    237         protocol_factory,
    238         program, *args,

c:\program files (x86)\python38-32\lib\asyncio\base_events.py in subprocess_exec(self, protocol_factory, program, stdin, stdout, stderr, universal_newlines, shell, bufsize, encoding, errors, text, *args, **kwargs)
   1628             debug_log = f'execute program {program!r}'
   1629             self._log_subprocess(debug_log, stdin, stdout, stderr)
-> 1630         transport = await self._make_subprocess_transport(
   1631             protocol, popen_args, False, stdin, stdout, stderr,
   1632             bufsize, **kwargs)

c:\program files (x86)\python38-32\lib\asyncio\base_events.py in _make_subprocess_transport(self, protocol, args, shell, stdin, stdout, stderr, bufsize, extra, **kwargs)
    489                                          extra=None, **kwargs):
    490         """Create subprocess transport."""
--> 491         raise NotImplementedError
    492 
    493     def _write_to_self(self):

NotImplementedError: 
```

laurensvalk, 2020-12-16:
Could you try installing [pybricksdev](https://github.com/pybricks/pybricksdev) separately first, to see if that is working?

Nelvarion, 2020-12-16:
shortdemo running without an issue
```
PS> poetry run pybricksdev run ble 'Pybricks Hub' demo/shortdemo.py
Searching for Pybricks Hub
Connecting to 90:84:2B:04:C3:81: Pybricks Hub
Connected successfully!
2020-12-16 09:30:30,062:    INFO: Sending: 50%
2020-12-16 09:30:30,418:    INFO: Sending: 100%

Hello
World!

Nelvarion, 2020-12-16:
Forgot to mention - this is on Windows, python 3.8.3

laurensvalk, 2020-12-16:
> shortdemo running without an issue

Thanks for confirming --- makes it easier for me to debug.

laurensvalk, 2020-12-16:
Do any of these steps fail?

[image]

laurensvalk, 2020-12-16:
I think I found it. It looks like the preconfigured lock file is outdated. To fix it do:

`poetry update`
`poetry install`

I will push the latest lock file (which is automatically created by the update step) so a clean clone will also just work again.

laurensvalk, 2020-12-16:
This is probably a good time to update the demo as well - thanks for the reminder! :smile: 

Things should work a lot better since `getchar` was introduced some time ago, so we can update the demo to use this.

Before I do, what did you have in mind for your experiment? Maybe I can add something for it in the demo.

Nelvarion, 2020-12-16:
I'll wait for your lock file, running `poetry update` triggers a bunch of other dependencies, `cryptography` needs Microsoft Visual C++ 14.0...

When you push, add the `build` folder 😉 

I'm going to use it to manage a train layout - there's going to be a Control+ hub connected to at least one switch (motor and color/distance sensor) and at least 2 trains with their own color sensors.
So for example, you could make sure that when a certain train is approaching, the switch will be pointing the correct way.

laurensvalk, 2020-12-16:
That is good to know, so if I understand correctly it is more about sending occasional messages instead of a fast data for a remote control.

You may already want to try #154 to see if you can capture a keyboard key to trigger the switch in the way you like.

The pybricksdev library / notebooks essentially offer a way to send bytes ("keyboard presses") from code instead, in your case triggered by what one of the other hubs is printing out.

laurensvalk, 2020-12-16:
I've pushed the update, by the way, so hopefully this will allow the install to proceed.

laurensvalk, 2020-12-16:
Updated car demo coming in just a few minutes.

Nelvarion, 2020-12-16:
Cloned and tried to run `poetry install` - it failed on the `cryptography (3.3.1)` part since it requires Microsoft Visual C++ 14.0 or greater is required.
I haven't gone through the dependency chain, but is this package required for `pybricks`?

laurensvalk, 2020-12-16:
I wonder if that's because `pybrickdev-demo` additionally also includes `numpy` and `matplotlib`. What happens if you delete those from `pyproject.toml`, then run `poetry update` and `poetry install`.

laurensvalk, 2020-12-16:
Build folder and communication demo has been updated.

Nelvarion, 2020-12-16:
> I wonder if that's because `pybrickdev-demo` additionally also includes `numpy` and `matplotlib`. What happens if you delete those from `pyproject.toml`, then run `poetry update` and `poetry install`.

Still failing at the same place

Nelvarion, 2020-12-16:
Checked each dependency with `poetry show --tree` and doesn't seem like `cryptography` is in any of them 🤔

laurensvalk, 2020-12-16:
> Checked each dependency with `poetry show --tree`

It returns this for me:

```
pybricksdev 0.1.0 Pybricks developer tools
|-- aioserial >=1.3.0,<2.0.0
|   -- pyserial * 
|-- argcomplete >=1.11.1,<2.0.0
|-- asyncssh >=2.2.1,<3.0.0
|   `-- cryptography >=2.8 
|       |-- cffi >=1.12 
|       |   `-- pycparser * 
|       `-- six >=1.4.1 
```

So I'm a bit puzzled why your regular install of `pybricksdev` did work.

laurensvalk, 2020-12-16:
By the way, what happens if you just do

```poetry run jupyter-notebook```

from within the `pybricksdev` project which does work? It contains a demo notebook too.

`pybricksdev-demo` was just a convenience repository with more examples, but maybe this gives more issues than convenience.

Nelvarion, 2020-12-16:
Ah, I wasn't sure how to resolve your packages since they originate from github.
Seems like it's coming from the SSH option for EV3 connections, right?

Tested from `pybricksdev` a simple notebook
```
%%file build/main.py

from pybricks.hubs import CityHub
from pybricks.parameters import Color

hub = CityHub()

for i in range(3):
    hub.light.on(Color.WHITE)
    wait(1000)
    print("Hello!", i)
    hub.light.on(Color.RED)
    wait(1000)
```
Got this error - seems like it's the same one from my original post:
```
---------------------------------------------------------------------------
NotImplementedError                       Traceback (most recent call last)
<ipython-input-6-3d9173a9c6e3> in <module>
----> 1 await hub.run('build/main.py')

~\Documents\code\pybricksdev\pybricksdev\connections.py in run(self, py_path, print_output)
    285 
    286         # Compile the script to mpy format
--> 287         mpy = await compile_file(py_path)
    288 
    289         # Get length of file and send it as bytes to hub

~\Documents\code\pybricksdev\pybricksdev\compile.py in compile_file(path, compile_args, mpy_version)
     72 
     73     # Get version info
---> 74     out = await run_mpy_cross(["--version"])
     75     installed_mpy_version = int(out.strip()[-1])
     76     if mpy_version is not None and installed_mpy_version != mpy_version:

~\Documents\code\pybricksdev\pybricksdev\compile.py in run_mpy_cross(args)
     38 
     39     # Run the process asynchronously
---> 40     proc = await asyncio.create_subprocess_exec(
     41         mpy_cross.mpy_cross, *args,
     42         stdout=asyncio.subprocess.PIPE,

c:\program files (x86)\python38-32\lib\asyncio\subprocess.py in create_subprocess_exec(program, stdin, stdout, stderr, loop, limit, *args, **kwds)
    234     protocol_factory = lambda: SubprocessStreamProtocol(limit=limit,
    235                                                         loop=loop)
--> 236     transport, protocol = await loop.subprocess_exec(
    237         protocol_factory,
    238         program, *args,

c:\program files (x86)\python38-32\lib\asyncio\base_events.py in subprocess_exec(self, protocol_factory, program, stdin, stdout, stderr, universal_newlines, shell, bufsize, encoding, errors, text, *args, **kwargs)
   1628             debug_log = f'execute program {program!r}'
   1629             self._log_subprocess(debug_log, stdin, stdout, stderr)
-> 1630         transport = await self._make_subprocess_transport(
   1631             protocol, popen_args, False, stdin, stdout, stderr,
   1632             bufsize, **kwargs)

c:\program files (x86)\python38-32\lib\asyncio\base_events.py in _make_subprocess_transport(self, protocol, args, shell, stdin, stdout, stderr, bufsize, extra, **kwargs)
    489                                          extra=None, **kwargs):
    490         """Create subprocess transport."""
--> 491         raise NotImplementedError
    492 
    493     def _write_to_self(self):

NotImplementedError:
```

laurensvalk, 2020-12-16:
> Seems like it's coming from the SSH option for EV3 connections, right?

Yes.

> Tested from pybricksdev a simple notebook

I don't quite understand why the command line tool would work but not the notebook, since it would ideally use the same environment.

How are you starting the Jypyter-Notebook (from which folder, and using which command)? And did you modify the demo or create a new notebook?

Nelvarion, 2020-12-16:
I cheated a little and copied the packages folders over from `pybricksdev`
There's a difference in versions:
`Updating cryptography (3.1.1 -> 3.3.1)`
The `demo` version is higher, could be more dependencies for that behind the scenes?

Each repo is cloned to a separate folder
CD into the relevant folder, run `poetry run jupyter-notebook`
I modified the provided notebooks since I don't have a standard motor yet.

laurensvalk, 2020-12-16:
> The demo version is higher

Presumably because we ran `poetry update` today.

> Each repo is cloned to a separate folder
> CD into the relevant folder, run poetry run jupyter-notebook
> I modified the provided notebooks since I don't have a standard motor yet.

:+1: Thanks for confirming.

In that case I'm not sure why a notebook wouldn't work inside the `pybricksdev` repo folder, since I think you mentioned that from here the command line option does work. 

What happens if you do:

```
poetry run python
```
And then run ```import os; print(os.environ['VIRTUAL_ENV']);```

And run it in a notebook cell as well.

Does it give the same output?

Nelvarion, 2020-12-16:
Sorry, not sure what you mean by `And run it in a notebook cell as well.` after the `import` command.

laurensvalk, 2020-12-16:
I was wondering what the output of this is, if you run it in your Jupyter Notebook:
```
import os
print(os.environ['VIRTUAL_ENV'])
```

Nelvarion, 2020-12-16:
[image]

laurensvalk, 2020-12-16:
If that's the same as the command line output of this, then we're good.

```
poetry run python -c "import os; print(os.environ['VIRTUAL_ENV']);"
```

But at that point I'm not sure why it is giving different results with the same Python environment.

Nelvarion, 2020-12-16:
This can very well be a client side issue on my end, I'll try to destroy everything and start again from scratch.
Thank you very much for taking all this time to troubleshoot!

Nelvarion, 2020-12-16:
> This can very well be a client side issue on my end, I'll try to destroy everything and start again from scratch.
> Thank you very much for taking all this time to troubleshoot!

So yeah, it was something in my python/pip/poetry cache.
Removed everything, reinstalled and tested a simple DCMotor.DC(speed) variant of the `basics` notebook - working without a problem.

I'm glad that at least we got an updated demo and a `build` folder out of this issue 😅

closed 2020-12-16 by Nelvarion

laurensvalk, 2020-12-16:
Glad to hear it!

Looking forward to see what you'll make. One last idea that may be interesting in your case:

The demo is set up to send something between hubs indirectly.

But another approach that may be useful is to make the PC *start* one script based on the output of another. It all depends on if that's quick enough for your scenario. Then it could be something like:

- run hub 1 forever in parallel with PC code (as we already have now):
- wait for hub 1 to report/print something interesting (detect something)
- Download and run a script on hub 2 (switch the switch)

In some applications, this may be a lot simpler than trying to mimic direct communication.

Nelvarion, 2020-12-16:
The problem with this approach is the time it takes to make the BLE connection - that's a train crash waiting to happen 😁

laurensvalk, 2020-12-16:
Sure, although there should be no need to make the BLE connection again, just calling `hub.run` again would do. It should be possible to try this in the basic demo.

I agree though, it's not fast enough for all applications.

Nelvarion, 2020-12-16:
I see what you mean, will give it a try later on.
Many many thanks for all the help!

johnscary-ev3, 2020-12-22:
Hi guys,

Trying to install the pybricksdev-demo but I keep getting this poetry error below.
It is looking in the /AppData/Local/pypoetry/Cache   for a file but /pypoetry/ directory does not even exist on my PC.
What is supposed to make that directory?
Have tried removing .venv directory and poetry.lock files as was suggested before.
It rebuilds those but still get same error.
Any clues?

PS C:\users\john\pybricksdev-demo> poetry install
Installing dependencies from lock file

Package operations: 84 installs, 0 updates, 0 removals

  • Installing ipython-genutils (0.2.0)

  EnvCommandError

  Command C:\users\john\pybricksdev-demo\.venv\Scripts\pip.exe install --no-deps file:///C:/Users/John/AppData/Local/pypoetry/Cache/artifacts/e6/8e/a3/8f37e14310c0072b3fcc4240490bcb42630aa695d069aee89953ebd9f8/ipython_genutils-0.2.0-py2.py3-none-any.whl errored with the following return code 1, and output:
  Processing c:\users\john\appdata\local\pypoetry\cache\artifacts\e6\8e\a3\8f37e14310c0072b3fcc4240490bcb42630aa695d069aee89953ebd9f8\ipython_genutils-0.2.0-py2.py3-none-any.whl
  ERROR: Could not install packages due to an EnvironmentError: [Errno 2] No such file or directory: 'C:\\Users\\John\\AppData\\Local\\pypoetry\\Cache\\artifacts\\e6\\8e\\a3\\8f37e14310c0072b3fcc4240490bcb42630aa695d069aee89953ebd9f8\\ipython_genutils-0.2.0-py2.py3-none-any.whl'

  at ~\.poetry\lib\poetry\utils\env.py:1074 in _run
      1070│                 output = subprocess.check_output(
      1071│                     cmd, stderr=subprocess.STDOUT, **kwargs
      1072│                 )
      1073│         except CalledProcessError as e:
    → 1074│             raise EnvCommandError(e, input=input_)
      1075│
      1076│         return decode(output)
      1077│
      1078│     def execute(self, bin, *args, **kwargs):

PS C:\users\john\pybricksdev-demo>

Nelvarion, 2020-12-22:
@johnscary-ev3 did you install `poetry`?
`pip install poetry`

dlech, 2020-12-22:
Which version of Python are you using? If you have more than one version of Python installed, select 3.8.x using `poetry env use python3.8` first. (If `python3.8` isn't in your path on Windows, try `py --list-paths` to get the full path and use that instead of `python3.8`). You can run `poetry env info` to see the current state.

Also make sure `pip` and `setuptools` are up to date in the virtual environment. (Run `poetry shell` to activate the virtual environment, then run `python -m pip install --upgrade pip` and `python -m pip install --upgrade setuptools`, then try `poetry install` again.)

johnscary-ev3, 2020-12-22:
Hi guys,

Thanks for the help on this.
I had actually installed poetry several times but I did it once again as above and noticed this time that it was putting things in the location of Python 3.9 , but when I ran poetry install for pybricksdev-demo it would flip over to use Python 3.8.
Somewhere along the line I had installed both Python 3.8 and 3.9 .
So I uninstalled Python 3.9 and tried it again and this time everything worked as it should and pybricksdev-demo installed.
I guess having two Pythons versions creates problems in the land of Python ;0)

dlech, 2021-05-16:
Addressing the `NotImplementedError` from the original post...

This happens on Windows because `tornado` (used by the ipython kernel) requires `WindowsSelectorEventLoopPolicy()` and selects it automatically but `asyncio.create_subprocess_exec()` used in `pybricksdev` requires `WindowsProactorEventLoopPolicy()`.