pybricks/support #178: [Bug]  Flashing firmware (from build #760) on movehub does not complete

BertLindeman, 2020-12-18:
**Describe the bug**
Flashing this firmware onto the movehub does not complete

**To reproduce**
Steps to reproduce the behavior:
1. Download movehub firmware from [build 760](https://github.com/pybricks/pybricks-micropython/actions?query=workflow%3ABuild+branch%3Amaster)
2. Drop the zip file onto the "Flash the Firmware" button
3. Press and hold the green button on the movehub
4. pair that hub
5. wait until the progress bar is full
6. wait five minutes more and drop your patience
7. press the green button and hold until the hub switches off.
8. pair the hub and look at the current firmware:

```
from pybricks import version
print(version)
```

```
('movehub', '3.0.0a12', 'v3.0.0a12-11-g11a7ecbc on 2020-12-18')
```

So the firmware **is** flashed. And reports the same as the technichub:

```
('movehub', '3.0.0a12', 'v3.0.0a12-11-g11a7ecbc on 2020-12-18')
----------
('technichub', '3.0.0a12', 'v3.0.0a12-11-g11a7ecbc on 2020-12-18')
```

**Expected behavior**
Would expect the hub to switch to pairing mode after the firmware flash is complete

**Screenshots**
None sofar

BertLindeman, 2020-12-18:
Firmware in  Build #756 does complete normaly.

Reports:
```
('movehub', '3.0.0a12', 'v3.0.0a12-8-gd590a50b on 2020-12-17')
```

dlech, 2020-12-18:
I think sometimes the move hub just doesn't reboot correctly after flashing. I don't think there is anything we can do about this since the bootloader can't be changed.

BertLindeman, 2020-12-18:
> 
> 
> I think sometimes the move hub just doesn't reboot correctly after flashing. I don't think there is anything we can do about this since the bootloader can't be changed.

Thanks @dlech Can we tell users that after (say) 30 seconds after the  "flashing firmware  progress-bar" is at 100%  it would be safe to hardstop the movehub and re-boot?

dlech, 2020-12-18:
Yes, we plan on improving the firmware update process.

laurensvalk, 2020-12-18:
> I think sometimes the move hub just doesn't reboot correctly after flashing. I don't think there is anything we can do about this since the bootloader can't be changed.

I wonder what the statistics are.

I always flash with pybricksdev during development, and it never fails at 100% for me. It does fail occasionally, but only at a point somewhere halfway, but this is detectable at least.

So maybe we can find the best of both worlds somehow.

BertLindeman, 2020-12-18:
On the other hand **before** this build 760 I never had this behaviour and with this firmware it consistently is"going-on" after 100% here. And the previous I have does reboot at 100% firmware flash.

dlech, 2020-12-18:
Maybe it was just a bad CI build. There is a newer one to try now.

BertLindeman, 2020-12-18:
> 
> 
> Maybe it was just a bad CI build. There is a newer one to try now.

Thanks, will do.

BertLindeman, 2020-12-18:
Took movehub firmware from  Build #763 
No success, to be sure: twice.

Then flashed previous one and went back to 
```'movehub', '3.0.0a12', 'v3.0.0a12-8-gd590a50b on 2020-12-17')```
without the problem.

dlech, 2020-12-18:
OK, I'm seeing the same thing.

Console log shows: 

```
io-6de156f3.js:111 Error: failed to get final response: ,,true
    at flash-firmware.ts:321
    at u (runtime.js:63)
    at Generator._invoke (runtime.js:293)
    at Generator.next (runtime.js:118)
    at p (redux-saga-core.esm.js:1157)
    at c (redux-saga-core.esm.js:1251)
    at t (redux-saga-core.esm.js:698)
    at c (redux-saga-core.esm.js:1251)
b @ io-6de156f3.js:111
b @ redux-saga-core.esm.js:1003
s @ redux-saga-core.esm.js:813
t.cont @ redux-saga-core.esm.js:828
p @ redux-saga-core.esm.js:1178
c @ redux-saga-core.esm.js:1251
n @ io-6de156f3.js:177
c @ redux-saga-core.esm.js:1251
b @ redux-saga-core.esm.js:1012
s @ redux-saga-core.esm.js:813
t.cont @ redux-saga-core.esm.js:828
b @ redux-saga-core.esm.js:1012
s @ redux-saga-core.esm.js:813
t.cont @ redux-saga-core.esm.js:828
b @ redux-saga-core.esm.js:1012
s @ redux-saga-core.esm.js:813
t.cont @ redux-saga-core.esm.js:828
p @ redux-saga-core.esm.js:1178
c @ redux-saga-core.esm.js:1251
t @ redux-saga-core.esm.js:698
c @ redux-saga-core.esm.js:1251
Promise.then (async)
T @ redux-saga-core.esm.js:395
O.<computed> @ redux-saga-core.esm.js:527
(anonymous) @ redux-saga-core.esm.js:1203
m @ redux-saga-core.esm.js:1271
(anonymous) @ redux-saga-core.esm.js:721
O.<computed> @ redux-saga-core.esm.js:716
(anonymous) @ redux-saga-core.esm.js:1203
m @ redux-saga-core.esm.js:1271
p @ redux-saga-core.esm.js:1161
c @ redux-saga-core.esm.js:1251
n @ io-6de156f3.js:165
c @ redux-saga-core.esm.js:1251
t @ redux-saga-core.esm.js:698
c @ redux-saga-core.esm.js:1251
l @ redux-saga-core.esm.js:503
e.put @ redux-saga-core.esm.js:339
L.e.put @ redux-saga-core.esm.js:371
(anonymous) @ redux-saga-core.esm.js:1412
dispatch @ redux.js:638
(anonymous) @ io-6de156f3.js:133
(anonymous) @ redux-saga-core.esm.js:472
d @ redux-saga-core.esm.js:31
b @ redux-saga-core.esm.js:87
m @ redux-saga-core.esm.js:58
O.<computed> @ redux-saga-core.esm.js:584
(anonymous) @ redux-saga-core.esm.js:1203
m @ redux-saga-core.esm.js:1271
p @ redux-saga-core.esm.js:1161
c @ redux-saga-core.esm.js:1251
l @ redux-saga-core.esm.js:503
put @ redux-saga-core.esm.js:173
(anonymous) @ redux-saga-core.esm.js:264
t @ lwp3-bootloader-ble.ts:133
Show 24 more frames
io-6de156f3.js:112 The above error occurred in task vr
    created by takeEvery(flashFirmware.action.flashFirmware, vr)
    created by bo
    created by bo
Tasks cancelled due to error:
takeEvery(flashFirmware.action.flashFirmware, vr)
```

So apparently it is timing out waiting for the final response.

closed 2020-12-18 by dlech

yipjustin, 2020-12-28:
Hello, I am not sure if I should raise my question on this thread or open a new issue.

## Describe the bug

Flashing firmware onto movehub does not complete (it says on a reb-green-blue flash loop).

## To repro

1. Open code.pybricks.com
2. Press and hold the green button on the movehub until the led flashes
3. Pair the hub
4. See red-green-blue flash cycles, release the green button
5. Movehub keeps showing the red-green-blue loop for minutes

(Give up. Long press green-button for 10 seconds to shut off)
After restart, the led flashes purple light.