pybricks/support #179: [Question] Where to find the latest firmware ZIP

Walt-White, 2020-12-18:
I successfully found version v3.0.0a12 firmware zip file, and followed the steps outlined in the documentation for modifying the main.py file and downloading the modified main.py to the City hub so my program would run automatically when the hub is powered on.

This is a great feature!  Great Ball Contraption modules have to run for hours at a public show, and I don't want to have to leave a laptop running the whole time.  And the documentation outline is still valid!

But I just accidently stumbled on the GitHub page where I found the Artifacts with the City hub firmware:

[image]

Now that a new version is available, where can I find the page with the new firmware zip file?

dlech, 2020-12-18:
Regular releases are at https://github.com/pybricks/pybricks-micropython/releases

CI builds are available at https://github.com/pybricks/pybricks-micropython/actions?query=is%3Asuccess+branch%3Amaster+workflow%3ABuild

GitHub recently changed the page layout. After clicking on a build, you have to scroll down the page a bit to see the artifacts.

We only recommend CI builds for people that like finding regressions and reporting bugs and for people that just can't wait for the latest bug fix to make it into a release.

laurensvalk, 2021-01-04:
The link in the doc has been updated. Thank you @Walt-White!

Fixed via https://github.com/pybricks/pybricks-api/commit/4c6c149d4ad0f920a3c90bd7b577eafb60399b77

closed 2021-01-04 by laurensvalk