pybricks/support #180: Use similar function of "run_until_stalled" but with two paired motors

RobertoZaffa, 2020-12-22:
Sorry for asking for help but I'm a beginner. Is there a similar way of the "run_until_stalled" command when I use two paired motors by "DriveBase"?  I need to reset the position of the two motors while them are running together.

laurensvalk, 2020-12-22:
Thanks for asking, this is a good question. There are a few ways you can do it.

First, you might find it helpful to know that the `run_until_stalled` function is a convenience function that does essentially the same as this:

```python
# Start the motor at a given speed
my_motor.run(speed)
# Wait until it is stalled
while not my_motor.control.stalled():
    wait(10)
# Stop the motor (you can also choose brake or hold here)
my_motor.stop()
```

So you can adapt it to your needs by using these components as you like. You can learn more about the `control` and `stalled` attributes on the ["more about motors" page](https://docs.pybricks.com/en/latest/motors.html).

So one way to do it is by changing the while loop to `while not my_motor.control.stalled() and not my_other_motor.control.stalled()`.

But you can also use the DriveBase if you wish. As you can see on the documentation page, the drivebase has a `distance_control` and `heading_control`. In your application, I think you could do something like `while not my_drive_base.distance_control.stalled()` in the loop above.

Let us know if this helps!

I'm looking forward to seeing the end result. Feel free to create a discussion here if you want to share what you have made.

RobertoZaffa, 2020-12-22:
Hi Laurensvalk, thanks a lot for your feedback. I will try tonight and will reply.

RobertoZaffa, 2020-12-22:
Hi Laurensvalk, 
I tried both the "drivebase" version and the one with separate motors.
The code seems to work for both cases but the two large motors are too powerful and it is difficult to block.
I would need something to reduce the power like the "duty" parameter does but I couldn't find a similar parameter.
Maybe there is something wrong in my code. It is also first time coding in Python :)
Thanks for any suggestion.
Roberto 

**********************

# Initialize DriveBase for the two large motors
y_motor = DriveBase(left_motor, right_motor, wheel_diameter=40, axle_track=100)

# settings(straight_speed, straight_acceleration, turn_rate, turn_acceleration)
y_motor.settings(50,10,0,0)

# Start the motor at a given speed
y_motor.straight(500)

# Wait until it is stalled
while not y_motor.distance_control.stalled():
    wait(10)
# Stop the motor (you can also choose brake or hold here)
y_motor.stop()

*********************************************************

# Initialize DriveBase for the two large motors
y_motor = DriveBase(left_motor, right_motor, wheel_diameter=40, axle_track=100)

# run_time(speed, time, then=Stop.HOLD, wait=True)
left_motor.run_time(100,5000, then=Stop.HOLD, wait=False)
right_motor.run_time(100,5000, then=Stop.HOLD, wait=False)

# Wait until it is stalled
while not left_motor.control.stalled() and not right_motor.control.stalled():
    wait(10)
# Stop the motor (you can also choose brake or hold here)
left_motor.stop()
right_motor.stop()

dlech, 2020-12-23:
> I replaced # with "REM" to avoid big characters in the post

Even better... just put a code fence around your code:

    ```python
    # this works
    ```

laurensvalk, 2020-12-23:
> I tried both the "drivebase" version and the one with separate motors.
> The code seems to work for both cases but the two large motors are too powerful and it is difficult to block.
> I would need something to reduce the power like the "duty" parameter does but I couldn't find a similar parameter.

Glad to hear that most of it is working already. You can change the duty limit using the "limits" method documented on the motor page:

```
my_motor.control.limits(actuation=30)
```
And set it back to `100` when you are done, so that all your other moves happen at full power again.

RobertoZaffa, 2021-01-09:
Hi Laurens,
at the end I used the following code:

`left_motor.control.limits(actuation=20)
  right_motor.control.limits(actuation=20)
  left_motor.run_time(300,5000, then=Stop.COAST, wait=False)
  right_motor.run_time(300,5000, then=Stop.COAST, wait=False)
  # Wait until it is stalled
  while not left_motor.control.stalled() and not right_motor.control.stalled():
    wait(10)
  # Stop the motor (you can also choose brake or hold here)
  left_motor.stop()
  right_motor.stop()
`

The two large-motors are very powerful and to block them at the end of the track I had to build a kind of cage (see photo).
However everything now seems to be working.
Now I should start developing the game and I should make extensive use of arrays. Is there something like "numpy" that I can use?
Thanks for your help

[image]

RobertoZaffa, 2021-01-10:
Hi David, I saw your post on:
https://www.eurobricks.com/forum/index.php?/forums/topic/182181-ev3-can-it-run-numpy/ where do you recommend installing a numpy-like library from:
https://github.com/v923z/micropython-ulab
I tried to follow the instructions for compiling the library but my poor skills in this matter do not allow me to move forward.
I will greatly appreciate any guidance / suggestion for installing ULAB so that I can move forward with my project that otherwise I fear I will have to give up. Being able to manage arrays is essential. Thanks also for the great work in making Pybrick.

dlech, 2021-01-10:
What math do you need that Pybricks doesn't have already? Are you using EV3 or Powered Up?

RobertoZaffa, 2021-01-10:
I'm using EV3. 
Basically I need arrays that I can use in loop structures. This is an implementation of an old game, Mastermind.
The loop below generates all possible combinations used during the game.

`i=1
cp = np.zeros((1297,12), int)
    for d in range(1,7):
        for e in range(1,7):
            for b in range(1,7):
                for c in range(1,7):
                    #INIZ_CODICI
                    cp [i,0]=d
                    cp [i,1]=e
                    cp [i,2]=b
                    cp [i,3]=c
                    i=i+1		`

I had already made this software in basic / assembler and now I'd like to see EV3 play.
Here the software and documentation:
https://sourceforge.net/projects/zx-master-mind/

dlech, 2021-01-11:
How about using a list of lists?

```python
cp = [[1, 2, 3, 4] + [0] * 8] * 1297
```

```
>>> print(cp[0])
[1, 2, 3, 4, 0, 0, 0, 0, 0, 0, 0, 0]
>>> print(cp[0][1])
2
```

dlech, 2021-01-11:
Oops, I see now the initialization isn't quite the same.

```python
cp = [[0] * 12] * 1297
````

will make it all zeros to start with

dlech, 2021-01-11:
Or 

```python
cp = []
for d in range(1,7):
   for e in range(1,7):
        for b in range(1,7):
            for c in range(1,7):
                #INIZ_CODICI
                cp.append([d, e, b, c] + [0] * 8)
```

dlech, 2021-01-11:
Or a dictionary with tuples as keys might be more appropriate if you want to be able to quickly look things up.

```
cp = {(0, 0, 0, 0): [0] * 8}
for d in range(1, 7):
    for e in range(1, 7):
        for b in range(1, 7):
            for c in range(1, 7):
                # INIZ_CODICI
                cp[(d, e, b, c)] = [0] * 8
```

RobertoZaffa, 2021-01-12:
Thanks David for your suggestions.
Testing  your code and looking at the documentation of the lists, I discovered that them can be used as one-dimension array and the elements can be pointed by index.
When I assign an element to a variable, I obtain a sub-list and I can then point to the "sub-elements"

Here is the code:
(the two function: def cp(row,col) and def cpw(row,col,value) allow me to use the list as if it were a two-dimensional array)

`def init_mm_db():
    i=1
    for d in range(1, 7):
        for e in range(1, 7):
            for b in range(1, 7):
                for c in range(1, 7):
                    #INIZ_CODICI
                    mm_db.append([0, d, e, b, c, 0, 0, 0, 0, 0, 0, 0, 0])
                    #INIZ_NUMCOL
                    temp=mm_db[i]
                    temp[4+d]=temp[4+d]+1
                    temp[4+e]=temp[4+e]+1
                    temp[4+b]=temp[4+b]+1
                    temp[4+c]=temp[4+c]+1
                    #INIZ_PESI
                    temp[11]=255
                    if temp[5]<3 and temp[6]<3 and temp[7]<3 and temp[8]<3 and temp[9]<3 and temp[10]<3:
                        temp [12]=188
                    else:
                        temp [12]=238
                    mm_db[i]=temp
                    i=i+1
    #Print for debugging
    for i in range(1, 1297):
      print (mm_db[i])

#def to read the matrix
def cp(row,col): #get a value from mm_db matrix
    temp=mm_db[row]
    return temp[col]

#def to write the matrix
def cpw(row,col,value): #write a value at the passed row/column
    temp=mm_db[row]
    temp[col]=value
    mm_db[row]=temp

mm_db =[0]
init_mm_db()

#test function to read and write the matrix
print (cp(1295,11)) #will print 255
cpw(1295,11,170) #write in mm_db[1295,11] the value 170
print (cp(1295,11)) #print the changed value
print (mm_db[1295])`

Please let me know what do you think about this solution.

Thanks again for your help.
When I'm done (the journey is still long) I will share the project.
Roberto.

dlech, 2021-01-12:
Here are some optimizations you can try. You don't actually need to define functions to do the read and write (unless you really don't like typing `[` and `]` :smile: )

```python
mm_db = [0]

def init_mm_db():
    for d in range(1, 7):
        for e in range(1, 7):
            for b in range(1, 7):
                for c in range(1, 7):
                    # INIZ_CODICI
                    temp = [0, d, e, b, c, 0, 0, 0, 0, 0, 0, 255, 0]
                    # INIZ_NUMCOL
                    temp[4 + d] += 1
                    temp[4 + e] += 1
                    temp[4 + b] += 1
                    temp[4 + c] += 1
                    # INIZ_PESI
                    if temp[5]<3 and temp[6]<3 and temp[7]<3 and temp[8]<3 and temp[9]<3 and temp[10]<3:
                        temp[12] = 188
                    else:
                        temp[12] = 238
                    mm_db.append(temp)
    # Print for debugging
    for temp in mm_db:
        print(temp)

init_mm_db()

# test function to read and write the matrix
print(mm_db[1295][11]) # will print 255
mm_db[1295][11] = 170  # write in mm_db[1295,11] the value 170
print(mm_db[1295][11]) # print the changed value
print(mm_db[1295])
```

RobertoZaffa, 2021-01-12:
Great!! Much better your solution.
I haven't been programming since school and my mind has remained tied to the Basic language of the 80s :).
I am gradually discovering that Python is very powerful when used well.

dlech, 2021-01-12:
I started with BASIC too in the early 90s. :smile: It took me quite a while to get used to Python enough that it didn't look like C code translated to Python any more.

laurensvalk, 2021-06-04:
This has been answered, so we can close this issue.

closed 2021-06-04 by laurensvalk