pybricks/support #182: Duplo train

anthonyangel, 2020-12-27:
**Question**
Do you support the duplo train hub?

**Context**
Does this project support the Duplo train which comes in set 10874 / 10875?

dlech, 2020-12-27:
No, we don't support it. But we aren't opposed to supporting it eventually. The more popular hubs have higher priority though.

laurensvalk, 2020-12-27:
@anthonyangel - do you have such a Duplo train hub already? If so:

1. Did you ever see the app update the firmware on the hub? If so, it would have been most likely when you used the app for the first time.
2. What happens if you follow the first 5 seconds of the steps in [this video](https://pybricks.com/wp-content/uploads/2020/06/install.mp4) from [this page](https://docs.pybricks.com/en/latest/start_pup.html)?
   - In other words, does a light start blinking when you press and hold the button?
   - And if so, do you see "LEGO Bootloader" as in the video? That would be a good sign. Do **not** click pair, as we don't have a firmware for that hub yet. Nothing bad should happen though, but just in case...

anthonyangel, 2020-12-27:
I can't remember ever seeing a firmware update from the app.

I tried to pair, as per the video - the hub only turns on (lights come on & it makes a noise) when the green power button is released, and nothing was showing in the pairing menu

laurensvalk, 2020-12-27:
Thanks, that is good to know. It is quite possible that this hub is not upgradeable.

dlech, 2023-02-04:
This hub doesn't have upgradeable firmware but can be remote controlled using the `LWP3Device` class.

closed 2023-02-04 by dlech

laurensvalk, 2023-02-04:
I've made an example project for this. Still needs to be published though. We might keep the issue open for that.

FideliusFalcon, 2023-12-25:
I tried the [example code](https://pybricks.com/projects/tutorials/wireless/hub-to-hub/pybricks-to-official-fw/), but it's printing "Connected!" right away, and doens't really connect. Is there some method to debug this? 
There's also a type error regarding the color object:
```
TypeError: unhashable type: 'Color'
```