pybricks/support #183: [Bug] Flashing Move Hub does not succeed

laurensvalk, 2020-12-28:
_Originally posted by @yipjustin in https://github.com/pybricks/support/issues/178#issuecomment-751588117_

> Hello, I am not sure if I should raise my question on this thread or open a new issue.
> 
> ## Describe the bug
>
> Flashing firmware onto movehub does not complete (it says on a reb-green-blue flash loop).
>
> ## To repro
> 
> 1. Open code.pybricks.com
> 2. Press and hold the green button on the movehub until the led flashes
> 3. Pair the hub
> 4. See red-green-blue flash cycles, release the green button
> 5. Movehub keeps showing the red-green-blue loop for minutes
> 
> (Give up. Long press green-button for 10 seconds to shut off)
> After restart, the led flashes purple light.
> 
> ## Expected behavior
> 
> Movehub flashes blue lights, and is able to pair and run code.
> 
> (I am able to flash my technic hub with no problem)
> 
> 
> Thanks!

laurensvalk, 2020-12-28:
Thanks for reporting @yipjustin. I've moved it to a new issue. In #178, flashing succeeds but it just doesn't reboot automatically when done. In your case, the purple blinking light suggests that flashing did not succeed.

Did you see any progress in the progress bar in the bottom left? If so, approximately when did it stop?

I've found that Move Hub can be very picky about low battery and being too far from the BLE dongle. If you haven't already tried this, could you try it with new batteries and close to your BLE dongle / Laptop?

yipjustin, 2020-12-28:
Hi Lauren, thanks for the quick reply.

My battery is brand new. I placed my Move Hub right on the top of my macbook.

The javascript console shows that there are some check-sum failure, see [screenshot](https://photos.app.goo.gl/N1mZh5Hh2ZzoVSeK8)

Btw, I am very impressed by this project. My experience with the technic hub was extremely smooth. Thank you!

laurensvalk, 2020-12-28:
Thanks for getting back and sharing the screenshot, that really helps.

I'm glad the Technic Hub is working for you in the mean time.

laurensvalk, 2020-12-28:
If you happen to have an Android device nearby, it could be worth trying that as well. Once passed the flashing hurdle, chances are programs will run fine on your Mac Book.

yipjustin, 2020-12-28:
@laurensvalk 

Great. I am able to flash the Move Hub from my Pixel 4a's Chrome.

And, as expected, the rest of the system work. Thank you.

By any chance there are command line support for copying the main.py to hub? Even unpolished one is ok, I can figure out the details. I am not very used to write code on web browsers  :)

laurensvalk, 2020-12-28:
Yes, if you don't mind unpolished there is this: https://github.com/pybricks/pybricksdev

Once installed, you can run

``
poetry run pybricksdev run ble 'Pybricks Hub' demo/shortdemo.py
``

Or integrate it in your editor of choice.

yipjustin, 2020-12-28:
Nice. I will try that out. Thanks!

dlech, 2020-12-28:
> The javascript console shows that there are some check-sum failure, see [screenshot](https://photos.app.goo.gl/N1mZh5Hh2ZzoVSeK8)

That is actually a timeout waiting to receive the checksum. What OS was this running on?

yipjustin, 2020-12-29:
```
% system_profiler SPSoftwareDataType
Software:

    System Software Overview:

      System Version: macOS 10.15.7 (19H114)
      Kernel Version: Darwin 19.6.0
      Boot Volume: Macintosh HD
      Boot Mode: Normal
      Computer Name: ****
      User Name: ****
      Secure Virtual Memory: Enabled
      System Integrity Protection: Enabled
      Time since boot: 10 days 7:10
 ```

dlech, 2021-01-26:
Hi @yipjustin, we recently made some update to Pybricks Code. I'm not sure that they will fix this issue, but could you please try again anyway?

dlech, 2021-01-26:
> By any chance there are command line support for copying the main.py to hub?

Also this is implemented now too. :smile:  There is a switch under *Settings*.

laurensvalk, 2022-08-23:
I'm going to close this due to inactivity. There have been a lot of Bluetooth fixes in the mean time. Let's re-open if we see this issue again.

Thanks!

closed 2022-08-23 by laurensvalk