pybricks/support #184: [Question]

NoMoreCakee, 2020-12-28:
**Question**
Every time when I using "pip install " command PyCharm and the IDLE says there is no module named like the installed.
I tried to restart my PC but the problem is still alive. Please help!

P.S.: sry for bad English!

NoMoreCakee, 2020-12-28:
`Traceback (most recent call last):
  File "C:\Users\sanyi\AppData\Local\Programs\Python\Python38\lib\site-packages\numpy\__init__.py", line 305, in <module>
    _win_os_check()
  File "C:\Users\sanyi\AppData\Local\Programs\Python\Python38\lib\site-packages\numpy\__init__.py", line 302, in _win_os_check
    raise RuntimeError(msg.format(__file__)) from None
RuntimeError: The current Numpy installation ('C:\\Users\\sanyi\\AppData\\Local\\Programs\\Python\\Python38\\lib\\site-packages\\numpy\\__init__.py') fails to pass a sanity check due to a bug in the windows runtime. See this issue for more information: https://tinyurl.com/y3dm3h86`

This is the error message in IDLE (imported module: pyautogui)

laurensvalk, 2020-12-28:
Could you explain what you are trying to do? What would you like to achieve?

NoMoreCakee, 2020-12-28:
I just wanted to follow a video.

NoMoreCakee, 2020-12-28:
https://www.youtube.com/watch?v=YRAIUA-Oc1Y

This was the video

laurensvalk, 2020-12-28:
You may want to ask on a different forum. It looks like you are trying to install a Python package on your PC. There's nothing wrong with that of course, but I'm not sure it's related to Pybricks.

Pybricks uses MicroPython, which does not have `pip` or `numpy` and `upip` is not supported.

NoMoreCakee, 2020-12-28:
oh, I'm sorry. I'm new to github.

closed 2020-12-28 by NoMoreCakee