pybricks/support #185: [Question] network module not found

XilefTech, 2020-12-28:
I wanted to use the micropython module 'network' in my project to detect wether or not WIFI is actevated/connected and to turn it on/off accordingly. The [Micropython Docs](https://docs.micropython.org/en/latest/library/network.WLAN.html) are clearly talking about an inbuilt module but I only get the ImportError `no module named 'network'`. So why is a standard library not available on the Micropython EV3 installation? (Same goes for ubluetooth too btw.)

My Project CharlieOSX has got some remote functionality over wifi. I also want to be able to use it at FLL without having to stop the program, turn WIFI and bluetooth of, then restart the program

dlech, 2020-12-28:
This sort of low-level functionality isn't included in Pybricks. However, you can probably get what you want by using `uos.system()` to call [connmanctl](https://manpages.debian.org/stretch/connman/connmanctl.1.en.html). 

```python
import uos

def wireless(enable):
    cmd = 'enable' if enable else 'disable'
    uos.system('connmanctl {} bluetooth'.format(cmd))
    uos.system('connmanctl {} wifi'.format(cmd))
```

XilefTech, 2020-12-29:
will definately try that right away, but why are you using uos? from my experience normal os works just fine with pybricks

dlech, 2020-12-29:
I try to use `u<name>` instead of `<name>` for imports whenever possible. Otherwise it could import from `micropython-lib` instead of using the built-in modules, which is bad for startup performance. And it also makes it explicit that we are using the MicroPython version of the library.

XilefTech, 2020-12-29:
oh okay, makes sense ^^

closed 2020-12-30 by XilefTech