pybricks/support #186: [Feature] Powered Up Remote Control

laurensvalk, 2020-12-29:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

laurensvalk, 2020-12-29:
Tagging @Nelvarion, @ZPhilo, @Patlepirate01, @jorgepe, @kueden for ideas. Thank you!

JorgePe, 2020-12-29:
"Bluetooth address of the remote. If no address is given, it connects to the first available remote that it finds"
For us, knowing the address is easy, but for neophytes it would help to have something that shows all devices nearby (maybe a button in Pybricks IDE is enough but having a "scan" on the API would help on the long run).

Also is there a way to "unpair"? So I can pass my robot from "remote1" to "remote2" if I want to make a game with a team, each player controlling until completing his/her part then passing the "token" to the next player?

laurensvalk, 2020-12-30:
Thanks for your response!

Regarding scanning, how about this?

```python
remote = Remote()
print(remote.address)
```

Now you know the address. So if you have more than one you could then do:

```python
remote = Remote("01:23:45:67:89:AB")
my_other_remote =  Remote("00:11:22:33:44:55")
```

> Also is there a way to "unpair"? So I can pass my robot from "remote1" to "remote2" if I want to make a game with a team, each player controlling until completing his/her part then passing the "token" to the next player?

Thanks for sharing, this is a nice use case. Yes, a `disconnect()` method would be good.

Nelvarion, 2020-12-30:
> Also is there a way to "unpair"? So I can pass my robot from "remote1" to "remote2" if I want to make a game with a team, each player controlling until completing his/her part then passing the "token" to the next player?

Wouldn't it be better to handle the turns in the software to avoid the time it takes to reconnect?

ZPhilo, 2020-12-30:
What happens with several remotes connected to the same hub? Does it work at all? is it possible for the remotes to be detected in the order they are powered, without messing with the addresses?

JorgePe, 2020-12-30:
> Wouldn't it be better to handle the turns in the software to avoid the time it takes to reconnect?

Probably better in this quick example but I'm sure there will be cases  where one might need to make the switch, Also instead of "one robot many remotes" it could also be useful for "many robots one remote" (need to think a good particular case but right now can only think on a robot controlling other robots without a proper messaging protocol... yes too clumsy)

laurensvalk, 2020-12-30:
> is it possible for the remotes to be detected in the order they are powered, without messing with the addresses?

Sure, if you turn on one remote at a time, you could do it like this. You could set the light as a way to tell you when to turn on the next one.

```python
from pybricks.pupdevices import Remote
from pybricks.parameters import Color

philo1 = Remote()
philo1.light.on(Color.GREEN)

philo2 = Remote()
philo2.light.on(Color.GREEN)

philo3 = Remote()
philo3.light.on(Color.GREEN)
```

> What happens with several remotes connected to the same hub? Does it work at all? 

No promises yet, but technically that should be possible. Usage could look like the script above.

laurensvalk, 2020-12-30:
> many robots one remote

This could be interesting for the Liebherr, but we don't know yet if this can technically be done.

But indeed as you say, this could work: remote <---> hub1 <---> hub2

laurensvalk, 2020-12-30:
Good news. I was doing some experiments, and all buttons can be independently read.

Also "many robots one remote" may actually be possible. It does not seem to refuse a new connection when a connection already exists.

I suppose @Nelvarion can already use it --- I'm connecting to the remote with a Jupyter Notebook for now to see what messages it sends :smile: Should I push it?

JorgePe, 2020-12-30:
>  Should I push it?

Any doubts? :D

A little out of scope: if you can connect to the Remote... you can also connect to pretty anything BT right? Like the EV3 or even the NXT... or even our own gadget as long as we offer the proper BT service. Not for now... but can a GeneralBTDevice be considered in the roadmap?

laurensvalk, 2020-12-30:
Right now I'm still connecting from my PC to the remote, to figure out what the remote sends when you push a button.

Next step is to make the hub do it, but @dlech said that part was easy (:sweat_smile:).

> if you can connect to the Remote... you can also connect to pretty anything BT right? 

Yes, though most hubs speak only BLE. If we ever enabled classic on SPIKE you can use it as a bridge to talk to EV3. :thinking:

laurensvalk, 2020-12-30:
>> Should I push it?
>
> Any doubts? :D

[Here you go](https://github.com/pybricks/pybricksdev-demo/blob/master/pup_remote.ipynb).

JorgePe, 2020-12-30:
don't forget that ev3dev can talk pure BLE - on EV3 with a USB dongle, on a BrickPi directly with the onboard BT chipset... until now I only used ev3dev as a GATT client but a couple of years ago I read something about creating our own BLE GATT services on a Raspberry Pi (doing it on the EV3 will probably be pushing it too much)

littldr, 2020-12-30:
Huge upvote for this from my side! I'm super excited to see this implemented!

It would allow me to not only build models together with my son (4), but also to allow him to easily interact with these models. :)

After scanning through the API docs I have the following questions:

* Has `Button.LEFT_UP` the same value as `Button. LEFT_PLUS`? The remote has a plus and minus button on each side, and if one is up or down can be changed.
* What about the red button in the middle of each side? I would assume it would be called `Button.LEFT_CENTER` or something like that? Or I am missing here something?
* Can the green button in the middle also be read? Or is changing the color (like in the original firmware) something the remote does on it's own and not via the hub?
* What would happen if the `timeout` is exceeded? Is the whole script stopped or a catchable exception raised? Or is the `remote.address` simply empty?

You do not need to answer these questions, but maybe it helps to find blindspots in the api docs or design. :)

laurensvalk, 2020-12-30:
> Huge upvote for this from my side! I'm super excited to see this implemented!

That's good to hear --- makes it more interesting for us to implement it too :)

> Has Button.LEFT_UP the same value as Button. LEFT_PLUS?

The Button constants do not really have a value; `Button.LEFT_PLUS` is just the "entity" that *is* that button. There's a few numbers in the fake API, but they are not shown in the docs since they don't really mean anything. We should probably remove these.

> What about the red button in the middle of each side? I would assume it would be called Button.LEFT_CENTER or something like that? Or I am missing here something?

I was thinking to just keep those as `Button.LEFT` and `Button.RIGHT` so we didn't have to introduce more. In any case, I'm planning to add a diagram [like this](https://docs.pybricks.com/en/latest/_images/primehub_buttons_label.png) to remove all doubt.

> Can the green button in the middle also be read?

Yes :tada: The only catch is that if you press it too long (> 3--4 seconds) it turns off.

> Or is changing the color (like in the original firmware) something the remote does on it's own and not via the hub?

The color can also be set, see the code snippet in the original post for the suggested API.

> What would happen if the timeout is exceeded? Is the whole script stopped or a catchable exception raised? Or is the remote.address simply empty?

I think we will raise a `TimeoutException`, which we can add to the docs indeed, thanks!

kueden, 2020-12-31:
I fully support this. Most users are into remote control. It will give them easy access to pybricks if you provide a simple way of remotely control. Personally I prefer physical buttons to virtual ones on a mobile phone. So this tool will allow to combine any model with physical tools that are even on a Lego controller (rather than ps4) keeping the purists on board.

I would really appreciate to see this implemented.

joosbuijsNL, 2021-01-07:
I would also really appreciate to see this implemented. This would allow builders to get from behind their laptops and control their builds physically and interact with it more. I can already see several applications! 

Please let me/us know how the community can help, other than cheering on the side of course :D

laurensvalk, 2021-01-07:
> Please let me/us know how the community can help, other than cheering on the side of course :D

Cheering is good, [sponsoring](https://github.com/sponsors/pybricks) is even better 😄.

mwinkler, 2021-01-08:
> [sponsoring](https://github.com/sponsors/pybricks) is even better 😄.

Done :)

Really keen to see this feature running on the hub!

laurensvalk, 2021-01-09:
Thank you, much appreciated!

laurensvalk, 2021-01-09:
And @joosbuijsNL and others, another great way to help is to test the latest releases to help us find (and maybe even fix) bugs. We just released a new firmware yesterday, which you can get via [Pybricks Code](https://www.code.pybricks.com/). If you still have something built, running your existing code with new firmware is a nice way to verify that we didn't break anything.

Another great help is to share your example code for official sets (e.g. Technic) at [Pybricks Projects](https://github.com/pybricks/pybricks-projects) so others can try it too.

TheVinhLuong102, 2021-01-10:
@laurensvalk, do you mean the support for this particular Remote Control is in the latest firmware build already? If so, Antoni and I can test its working with the SPIKE/Inventor Hub. :)

joosbuijsNL, 2021-01-10:
@laurensvalk Of course, would be more than happy to help. Although currently I'm in the state where I can mainly test the documentation by trying to follow it ;)

Another point, as indicated by @TheVinhLuong102 's question: how can we know when a new firmware version has been released, what has been changed or added, and how can we signal that everything seems ok (because no bug reports could also mean we just didn't test).

laurensvalk, 2021-01-10:
@joosbuijsNL 

> how can we know when a new firmware version has been released, what has been changed or added

There is a pinned issue #48 with release announcements. I like your idea of keeping tracking that you’ve tested it. Not sure what the best format is to do that. Maybe give the post in #48 a 🚀 ? Or a new issue?

@TheVinhLuong102, no, not yet. This post was to gather ideas from users for features, and to get an idea of how important this feature is to all of you. We might be doing more posts like these to help us prioritize new features.

But it is probably fair to say that this one is very popular.

Nelvarion, 2021-01-10:
> There is a pinned issue #48 with release announcements. I like your idea of keeping tracking that you’ve tested it. Not sure what the best format is to do that. Maybe give the post in #48 a 🚀 ? Or a new issue?

Maybe a discussions thread?

mrstux, 2021-01-13:
I was asked by PyBrick's on FB to add some comments to this thread regarding usinng PyBricks with Trains,

Some things worth considering from someone who uses PoweredUp City Hubs with Trains...

It'd be great to be able to use PyBricks with Trains and the handset, there are some different scenarios though, and many of them would mean wanting to use the same handset with multiple hubs.

For example, you might have two trains, each with a hub, and any of these hubs might be running two train motors... and you would want one side of the handset to control one hub, and the other side to control a different hub (in a different train)

Also, you may actually want multiple handsets, so that two children could independently and simultaneously control the same trains (teaches cooperation).

Another scenario is that you might have a train with two hubs (one at each end, each controlling a different train motor), but you would want a single handset side to control both hubs, this is used when you combine multiple trains (ie https://www.lego.com/en-au/product/passenger-train-60197)

And in that scenario, you may also want the same handset to be used to control other hubs, ie you might have a freight train, and a long passenger train.

So, I guess, my point is whatever design you choose, it would be ideal, if multiple hubs could be listening to multiple handsets :)

All of the above can be done using the PoweredUp app, but in that case, I'm guessing its actually the app listening to multiple handsets... so I'm not sure if its even possible with the distributed pybrick's model

Nelvarion, 2021-01-13:
@mrstux great scenarios!
I'm using `pybricks` to control and automate my train layout for our LUG's show and some of these scenarios are on my to-do list - mainly to allow the layout to be interactive to the visitors.
I will gladly share my work once it's ready.

Vinz1911, 2021-02-10:
Great to see, I did some testing and implemented an example how to connect a spike prime hub to a lego remote by using micro pythons ubluetooth api. This can be found here: [https://github.com/Vinz1911/PrimePowerUP/blob/master/examples/powerup_remote/remote_dots/remote_dots.py](Spike Prime + Remote)

The cool thing is, bluetooth bleiben has a notification service, so its possible to trigger a callback with button  byte at the moment the button is pressed. this works super fast and you don't need to read out button press in a loop.

laurensvalk, 2021-02-17:
Thanks for sharing and becoming a sponsor @Vinz1911! That's a great demo.

For the reasons you mention in your README (RAM & freezes) we might do it in a slightly different way in our firmware, probably without `ubluetooth`. We're actively revising Bluetooth at this very moment :smile:.

But if you think certain MicroPython features such as `ubluetooth` should be enabled, feel free to share your thoughts in #140.

Vinz1911, 2021-02-17:
Thank you :) 

Great to hear, I can't wait to see a Bluetooth implementation from you for PyBricks *_*

Brecked, 2021-02-19:
Hi Could it be possible to use this same API to use it with xbox or other bluetoth remote?

Vinz1911, 2021-02-19:
No this is not Possible. Because the Bluetooth Control Codes and Button Mapping, etc. are completely different

laurensvalk, 2021-02-19:
It may not be impossible, but it would have to use a different class indeed. For a dedicated discussion for other remotes, see #191.

mwinkler, 2021-02-19:
@Brecked I may start working on this (connecting to a BLE gamepad -> HID over GATT) base on @Vinz1911 implementation for the Powered Up Remote Controller for Spike/Inventor hub.

Brecked, 2021-02-19:
Thanks all for fast comments on my question. Overall this thing about programming lego hubs directly is awesome

mwinkler, 2021-02-19:
Proof of concept is working (connecting to BLE HID device) -> https://www.youtube.com/watch?v=RIcseIcdAm0

stonie08, 2021-04-07:
I'd love to see this being implemented.

It'd be great if the implemenation of the handset could be used to overcome some nasty limitations of the vanilla implementation regarding trains:
If the controller goes out of range of the train the vanilla implementation of the hub stops the train immediately, will wait briefly for a reconnect and then simply shutdown. While there are certain situations where this could be theoretically useful, it is aweful if you have sturdy walls and want build train routes between rooms.
So ideally it'd be great if one could implement that the hub is paired with one or many controllers. If any controller goes out of range the hub will keep executing the main loop. If a controller comes into range again it should re-pair with the hub and be able to provide inputs again.
This should also mimic the way it's done with the power functions system, which didn't need a constant connection between the hub and the controller.

dlech, 2021-05-29:
Do we want/need an indication that the remote has gone out of range? For example, it could be nice to stop your vehicle if the remote goes out of range so that it doesn't run off without you.

dlech, 2021-05-29:
Another question: how annoying will it be if the remote disconnects each time the program stops? The remote turns itself off after about 10 seconds if it is not connected. Maybe it is not big deal to press the button on the remote to turn it on since you will be holding it anyway?

JorgePe, 2021-05-29:
not annoying if the program never stops (most of mine). But is it possible to check for connectivity status and reconnect if for any reason connection drops (bug, distance, radio inferference, changing batteries...)

dlech, 2021-05-29:
From the hub side, starting your program should be the only action required.

laurensvalk, 2021-05-29:
> Maybe it is not big deal to press the button on the remote to turn it on each time you modify your program since you will be holding it anyway?

👍

kueden, 2021-05-30:
No big deal to reconnect. For debugging we can use keyboard commands.
Indication of loss of connection would be a nico, but not urgent feature.
Thanks for all your efforts!

dlech, 2021-06-02:
One hub working, 2 to go... plus lots of polishing...

[image]

Vinz1911, 2021-06-02:
Awesome Work!!! When can I test it on my Spike Prime ? 😜

dlech, 2021-06-02:
Right now. 😄  Example program and link to firmware: https://gist.github.com/dlech/dddeab5bc970a4691df1ca6ef1ce981c

Vinz1911, 2021-06-02:
Awesome!! I will have a lot of fun Tomorrow 😍

AntoniLuongPham, 2021-06-03:
Dear @dlech,

This is fantastic and I and my sister would love to try but my dad doesn't know  how to install the firmware.
Could you tell us how?

Thank you very much!

Antoni

laurensvalk, 2021-06-03:
We've just [updated the beta installer to include it](https://beta.pybricks.com/) so it works as usual.

TheVinhLuong102, 2021-06-03:
Hi @laurensvalk,

I've just connected the Mindstorms Inventor Hub to the laptop through USB, but https://beta.pybricks.com/ shows "No compatible devices found. My screenshot is attached.

Should I have done some step differently?

Thank you @laurensvalk.

Vinh

[image]

Vinz1911, 2021-06-03:
Short feedback from my side, big thumps up 👍 You guy's did a great work. It works really nice!! That the remote is disconnecting after program stops is also a great solution.

TheVinhLuong102, 2021-06-03:
Hi @Vinz1911, may I ask if you installed the new firmware through beta.pybricks.com as usual?

I've been trying to install on my end but not been successful.

Thank you.

TheVinhLuong102, 2021-06-04:
@dlech @laurensvalk the Remote works very well -- this is brilliant!! <3

laurensvalk, 2021-06-22:
Has anyone paid attention to the commit messages lately? :thinking: 

:rocket: :rocket:

TheVinhLuong102, 2021-06-22:
@laurensvalk which repo are you referring to? `pybricks-micropython`?

laurensvalk, 2021-06-22:
That's the one. I added a link for clarity. Thanks!

laurensvalk, 2021-06-22:
That's the one. I added a link for clarity. Thanks!

This should do:

[image]

laurensvalk, 2021-06-22:
To clarify, it is experimental still. Eventually it will make its way into the next release. But it means we're close!

It's [already useable](https://pybricks.com/install/technic-boost-city/#installing-the-latest-build) if you don't mind a bit of extra work to install the latest build.

JorgePe, 2021-06-22:
missed this for move hub... need to charge some batteries (but been following reduce "flash and RAM usage by uartdev")

laurensvalk, 2021-06-22:
> missed this for move hub

You missed it by 9 whole hours! :smile:

[23 more replies]

closed 2021-11-17 by dlech