pybricks/support #187: [Question] Multi hub notebooks - receiving hub doesn't receive input

Nelvarion, 2020-12-31:
**Question**
Playing around with a modified version of [comminucation notebook](https://github.com/pybricks/pybricksdev-demo/blob/master/communication.ipynb)
Modified version is [switch notebook](https://github.com/Nelvarion/pybricksdev-demo/blob/master/switch.ipynb).
First hub (train) has a color sensor and outputs a variable based on the color detected.
Second hub (switch) _should_ receive this variable and run some code based on the value.

1. When I run everything the train hub seems to be outputting its value properly.
2. Something generating 2 empty lines (added delay to the first to make sure it's not the sensor initialization)
3. The PC cell doesn't seem to be getting any data, so switch hub doesn't get it either.

**Screenshots**
[image]
Forgive the MSPaint skillz...

laurensvalk, 2020-12-31:
It seems that the switch notebook is based on an older version of the communication example.

I’m not entirely sure that is the cause, but it could help to have a look at the newer version you linked to above. The BLEPUPConnectionExtra class is no longer required.

Nelvarion, 2020-12-31:
I think I failed to mention that this is all running on CityHubs, having looked around the firmware for it, seems like it doesn't have the `pybricks.experimental` class - so I can't use `getchar`.

I'll try and switch back to the previous method you used for passing the data.
Are you planning to add this class to the other hubs?

laurensvalk, 2021-01-01:
Ah, yes, we can enable it on City Hub. Good point.

laurensvalk, 2021-01-02:
Just enabled it. You can grab the latest master build to try it. I didn't try it myself, but it should work the same as on other hubs.

Nelvarion, 2021-01-02:
Last commit on `master` is from 5 days ago, found it on `work` 😉 
Will you merge it later on?

And it's working perfectly, thank you!

laurensvalk, 2021-01-02:
Cool! Sounds like we can close this one then. Looking forward to see what you've made. :rocket:

closed 2021-01-02 by laurensvalk