pybricks/support #189: [Feature] Multi-File Project on PrimeHub/InventorHub

TheVinhLuong102, 2021-01-02:
Hi @dlech and @laurensvalk,

Happy New Year to you! Antoni and I are looking forward to exciting PyBricks releases this year.

Antoni and I are trying out Pybricks on the InventorHub, and we would like to ask if it is possible yet/soon to download multiple files of one same project onto the InventorHub and import code classes/functions among such files?

Each robot in the Inventor set has many variants and plays many different games, so it would be great if we can organize a base class and many subclasses for each robot.

Thank you very much!

Vinh

laurensvalk, 2021-01-02:
Wow, you guys are quick! :rocket: 

> we would like to ask if it is possible yet/soon to download multiple files of one same project onto the InventorHub and import code classes/functions among such files?

In the long run, probably yes. However, knowing how fast you are, probably not before you have built all the models :smile: 

> Each robot in the Inventor set has many variants and plays many different games, so it would be great if we can organize a base class and many subclasses for each robot.

However, a base class for each robot is probably still a good idea. You'd have to have a copy in each program for now.  Perhaps we could split it out to a library (or something like it) later when we have file system support.

Since this system is a bit different from EV3, I do have a few suggestions for minor style changes for these projects. Perhaps we can discuss those when you make the first pull request.

laurensvalk, 2021-01-02:
Since you've shared a snippet in #190, perhaps I can share what I mean here. These are some performance considerations as well as suggestions to make them easier to read for beginning users.

```python
#!/usr/bin/env pybricks-micropython

from pybricks.hubs import InventorHub
from pybricks.pupdevices import Motor, ColorSensor, UltrasonicSensor
from pybricks.parameters import Direction, Port
from pybricks.robotics import DriveBase

class Tricky:
    WHEEL_DIAMETER = 44
    AXLE_TRACK = 88

    def __init__(
            self,
            left_wheel_motor_port: Port = Port.A,
            right_wheel_motor_port: Port = Port.B,
            sport_motor_port: Port = Port.C,
            distance_sensor_port: Port = Port.D,
            color_sensor_port: Port = Port.E):
        self.hub = InventorHub()

        self.driving_base = \
            DriveBase(
                left_motor=Motor(port=left_wheel_motor_port,
                                 positive_direction=Direction.CLOCKWISE),
                right_motor=Motor(port=right_wheel_motor_port,
                                  positive_direction=Direction.CLOCKWISE),
                wheel_diameter=self.WHEEL_DIAMETER,
                axle_track=self.AXLE_TRACK)

        self.sport_motor = Motor(port=sport_motor_port,
                                 positive_direction=Direction.CLOCKWISE)

        self.color_sensor = ColorSensor(port=color_sensor_port)

        self.distance_sensor = UltrasonicSensor(port=distance_sensor_port)

if __name__ == '__main__':
    TRICKY = Tricky()

    TRICKY.distance_sensor.lights.on(100)

    TRICKY.distance_sensor.lights.off()
```

- You may omit `#!/usr/bin/env pybricks-micropython`. This was specific to ev3dev
- I think it would be good to omit this as well:
```python
            left_wheel_motor_port: Port = Port.A,
            right_wheel_motor_port: Port = Port.B,
            sport_motor_port: Port = Port.C,
            distance_sensor_port: Port = Port.D,
            color_sensor_port: Port = Port.E):
```
Basically I think we can rely on the building instructions for cables/ports, just like we assume users have built the rest of the robot according to the instructions.
- Within the class methods, it may be better to use regular arguments for required arguments, while using keyword arguments only for optional arguments. Sometimes, this makes the script a bit easier to read, and on these hubs it is also better for performance.
- Omitting `if __name__ == '__main__':` would be fine as well.

What do you think?

laurensvalk, 2022-08-23:
We've got a proof of concept where this works! If you want to give it a go:

- Try experimental firmware:  https://github.com/pybricks/pybricks-micropython/tree/fix-526
- Experimental Pybricksdev branch to match: https://github.com/pybricks/pybricksdev/pull/37

laurensvalk, 2022-08-23:
> `if __name__ == '__main__':`

It would be a good to have a closer look at this kind of functionality before we merge it. This is a part I haven't looked at extensively so far. I believe this works though.

dlech, 2022-10-21:
This is now possible in both Pybricks Code Beta and pybricksdev. Both "magically" search for imports and include imports in the same directory as the main script when downloading and running.

closed 2022-10-21 by dlech