pybricks/support #190: [Bug] InventorHub hangs when running simple program with UltrasonicSensor

TheVinhLuong102, 2021-01-02:
I created the below simple program with the Tricky bot turning on and off the lights of its Ultrasonic Sensor.

I ran the program with pybricksdev through BLE, and every time the program would hang and fail to terminate. The only way to terminate is to take out the battery from the InventorHub.

```
#!/usr/bin/env pybricks-micropython

from pybricks.hubs import InventorHub
from pybricks.pupdevices import Motor, ColorSensor, UltrasonicSensor
from pybricks.parameters import Direction, Port
from pybricks.robotics import DriveBase

class Tricky:
    WHEEL_DIAMETER = 44
    AXLE_TRACK = 88

    def __init__(
            self,
            left_wheel_motor_port: Port = Port.A,
            right_wheel_motor_port: Port = Port.B,
            sport_motor_port: Port = Port.C,
            distance_sensor_port: Port = Port.D,
            color_sensor_port: Port = Port.E):
        self.hub = InventorHub()

        self.driving_base = \
            DriveBase(
                left_motor=Motor(port=left_wheel_motor_port,
                                 positive_direction=Direction.CLOCKWISE),
                right_motor=Motor(port=right_wheel_motor_port,
                                  positive_direction=Direction.CLOCKWISE),
                wheel_diameter=self.WHEEL_DIAMETER,
                axle_track=self.AXLE_TRACK)

        self.sport_motor = Motor(port=sport_motor_port,
                                 positive_direction=Direction.CLOCKWISE)

        self.color_sensor = ColorSensor(port=color_sensor_port)

        self.distance_sensor = UltrasonicSensor(port=distance_sensor_port)

if __name__ == '__main__':
    TRICKY = Tricky()

    TRICKY.distance_sensor.lights.on(100)

    TRICKY.distance_sensor.lights.off()
```

laurensvalk, 2021-01-04:
The following program works for me:

```python
from pybricks.pupdevices import UltrasonicSensor
from pybricks.parameters import Port
from pybricks.tools import wait

sensor = UltrasonicSensor(Port.A)

sensor.lights.on(100)

wait(3000)

sensor.lights.off()
```

Could you test your example in Pybricks Code? And potentially make it smaller to identify what causes it to hang?

TheVinhLuong102, 2021-01-04:
@laurensvalk I can confirm that the program works with any wait time of at least 1 milisec (i.e. wait(1) works), while any wait time less than 1 milisec (e.g. wait(0.999999)) would cause the Inventor Hub to hang.

(Actually, wait(0.999999999) works sporadically -- but still causes hanging from time to time).

So the magic threshold seems to be very, very precisely 1 milisec!

Question is: why? :) Is there a magic number behind the implementation that causes this behavior?

Vinh

dlech, 2021-01-04:
> Question is: why? :) Is there a magic number behind the implementation that causes this behavior?

Waiting has a 1 millisecond resolution so everything after the decimal is ignored. `wait(0.999)` is the same as `wait(0)`.

TheVinhLuong102, 2021-01-04:
@dlech, I see. So this issues reduces to the `hub.light.on()` function needing a `wait()` then. :)

laurensvalk, 2021-01-05:
Thanks @TheVinhLuong102 for investigating further; that's very helpful.

Bug can be reproduced with 

```python
from pybricks.pupdevices import UltrasonicSensor
from pybricks.parameters import Port
from pybricks.tools import wait

sensor = UltrasonicSensor(Port.A)

sensor.lights.on(100)

sensor.lights.off()
```

laurensvalk, 2021-01-05:
> This one is interesting because the mode switch already triggers a 50ms delay, and setting it to 60ms does not resolve it. So it's strange that 1ms in the user program would make the difference.

It's unrelated to mode changing. Rather, it's related to setting values on the sensor. I've added a delay as a workaround. We may or may not be able to fix this at the driver level when we revisit the drivers for the LUMP devices.

laurensvalk, 2021-01-05:
Fixed via https://github.com/pybricks/pybricks-micropython/commit/f94ab4e3117fcce2815e0b5e9460c2e0a3dfea47

Thanks for reporting!

closed 2021-01-05 by laurensvalk

TheVinhLuong102, 2021-01-05:
Fantastic @laurensvalk, thank you!

So when should we redownload and reinstall the firmware for this change to take effect? Are your firmware builds created nightly?

P.S.: Now I can tell Antoni this not Daddy's fault. He was like "Daddy you are playing so roughly with my Hub -- you are breaking it!!" :))

laurensvalk, 2021-01-05:
> So when should we redownload and reinstall the firmware for this change to take effect? Are your firmware builds created nightly?

Any time you like. The download link and getting started steps in #167 still apply; just grab the latest build via that link.

We never managed to truly break a hub so far, and believe me we've tried :smile:

TheVinhLuong102, 2021-01-05:
@laurensvalk @dlech I've installed the latest firmware and can confirm that the problem has gone away in our Tricky programs.

Thank you for the quick fix.

Vinh

P.S.: Great to learn of the Hub's proven durability. Can't afford to fund any replacements -- Santa in December seemed to have a good budget but Daddy in January is fundamentally broke. :D