pybricks/support #198: Change mindsensors read speed

ksj537465, 2021-01-05:
now i using AbsoluteIMU and PPS58 sensor

this sensors read speed is 5times per second(millisecond)

i want change read speed as 1time per second

how can i do that

ksj537465, 2021-01-06:
this is main script

#!/usr/bin/env pybricks-micropython
#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.parameters import Port, Stop, Direction, Button, Color
from pybricks.tools import wait, StopWatch, DataLog
from pybricks.ev3devices import Motor
from pybricks.media.ev3dev import Font
from  mindsensorsPYB import PPS58
from  mindsensorsPYB import ABSIMU

import os
import sys
import time 

from time import sleep

def PPS58Test() :
# Create your objects here.
    ev3 = EV3Brick()
    pps58 = PPS58(Port.S2,0x18)

    # Write your program here.
    print(pps58.readAbsolute(),pps58.readGauge(),pps58.readRef())
    return pps58.readAbsolute()

def DepthTest() :
# Create your objects here.
    ev3 = EV3Brick()
    pps58 = PPS58(Port.S2,0x18)

    # Write your program here.
    print(pps58.readAbsolute(),pps58.readGauge(),pps58.readRef())
    return pps58.readGauge()

def ABSIMUTest() :
# Create your objects here.
    ev3 = EV3Brick()
    imu = ABSIMU(Port.S1,0x22)

    # Write your program here.
    print(imu.get_accelall(),imu. get_heading(),imu.get_gyroall())
    return imu.get_accelall(),imu.get_gyroall(),imu.get_heading()

# Base_Press = PPS58Test() 	
data = DataLog('Pressure','ABS','time') 
watch = StopWatch()

for x in range(99999):
    # Motor(Port.A).run_time(1000, 18000, Stop.BRAKE, False)
    # Motor(Port.B).run_time(1000, 18000, Stop.BRAKE, False)
    Pressure = PPS58Test() 
    ABS = ABSIMUTest() 
    time = watch.time()/1000 
    Diff = DepthTest() 
    data.log(Pressure,ABS,time) 
    if Diff < 5: continue 
    # Motor(Port.A).run_time(-1000, 18000, Stop.BRAKE, False)
    # Motor(Port.B).run_time(-1000, 18000, Stop.BRAKE, False)
    if Diff == 1: 
        sleep(10)

and this is sensor class part 

## PPS58 : this class provides EV3Dev specific interface for the   PPS58
#  for read and write operations.
class PPS58(mindsensors_i2c):
    def __init__(self, port,i2c_address=0x18):
        mindsensors_i2c.__init__(self,port, i2c_address)
[... 199 lines cut ...]
        self.command(99)
       

    ## Sets accelerometer sensitivity to 2G
    #  @param self The object pointer.
    def accel_2G(self):
        self.command(49)
        

    ## Sets accelerometer sensitivity to 4G
    #  @param self The object pointer.
    def accel_4G(self):
        self.command(50)
        

    ## Sets accelerometer sensitivity to 8G
    #  @param self The object pointer.
    def accel_8G(self):
        self.command(51)
        

    ## Sets accelerometer sensitivity to 16G
    #  @param self The object pointer.
    def accel_16G(self):
        self.command(52)

laurensvalk, 2021-01-06:
Where is `mindsensors_i2c` defined? Is this a module obtained from somewhere? I can't seem to find it on the mindsensors website.

Back to your original question:

> this sensors read speed is 5times per second(millisecond)

> i want change read speed as 1time per second

If you want to decrease the speed as you say, I suppose you could just read the sensor less often in your script?

But if you are asking to increase the speed, I believe you can change the poll time of the sensor.

Have a look at the second example snippet in the [Ev3devSensor](https://pybricks.github.io/ev3-micropython/iodevices.html#ev3dev-sensors) class. It shows you how to access sensor settings by reading and writing to files. Here is an [overview of those files](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensors.html#sysfs). The example shows how to read the `modes` file. In your case, I think you want to read (and write) the `poll_ms` file.

ksj537465, 2021-01-06:
the attached file is the full ver
 
i email to mindsensor to use this sensors at pybricks
 
back to answer, what i want is 
 
decrease sensors read speed(pps58, absoluteimu) 1time per seccnd
 
now the unit seems as milli-second but i want change it to second
 
at exel file you can see the time 0.372 0.562 etc it because on my script  time = watch.time()/1000 
 
i want that time measrue like 1, 2 ,3 second 
 
did i explain well enough for u to understand??

ksj537465, 2021-01-06:
> I see, in that case let's have a look at this portion of your code.
> It looks like you are measuring data in a loop. You could add a wait(1000) to the loop to wait a second.
> I've also added a suggestion about initializing the sensor. See the comments I added below.
> # Base_Press = PPS58Test() data = DataLog('Pressure','ABS','time') watch = StopWatch() for x in range(99999): # Motor(Port.A).run_time(1000, 18000, Stop.BRAKE, False) # Motor(Port.B).run_time(1000, 18000, Stop.BRAKE, False) Pressure = PPS58Test() # <------ It would be better to do this *before* the for loop ABS = ABSIMUTest() # <------ It would be better to do this *before* the for loop time = watch.time()/1000 Diff = DepthTest() data.log(Pressure,ABS,time) # <----- Here you could wait for a while. For example, do `wait(1000)` if Diff < 5: continue # Motor(Port.A).run_time(-1000, 18000, Stop.BRAKE, False) # Motor(Port.B).run_time(-1000, 18000, Stop.BRAKE, False) if Diff == 1: sleep(10)

can i ask why are u comment comment wait(1000)??

and the motor command is correct?

first motor go down while measure data

while motor go down if depth >5 motor go up

and while motor go up depth ==1 stop 1sec

this is the loop and during the loop every measure data will datalog is it right?
​

laurensvalk, 2021-01-06:
> can i ask why are u comment comment wait(1000)??

I only made a few suggestions as comments, so it was easier to see where to place it. 

But in your code, it should not be a comment:

```python
wait(1000)
```

ksj537465, 2021-01-07:
can i get some example for use poll_ms ????

 

wait(1000)

laurensvalk, 2021-01-07:
That's only needed, if you want to *increase* the *polling frequency*. I believe you said you wanted to *decrease* the *logging frequency*, so you're going to need the wait.

ksj537465, 2021-01-07:
thank u

closed 2021-01-07 by ksj537465