pybricks/support #200: [Bug] Inventor Set motor movement goes on forever despite instructed otherwise

TheVinhLuong102, 2021-01-05:
Hi @dlech, @laurensvalk,

In the below program, when the commented-out block is included, the robot movement would go on until the program is forcefully terminated.

```
from pybricks.hubs import InventorHub
from pybricks.pupdevices import Motor, ColorSensor, UltrasonicSensor
from pybricks.parameters import Color, Direction, Port, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait

class TrickyPlayingSoccer:
    WHEEL_DIAMETER = 44
    AXLE_TRACK = 88

    def __init__(self):
        self.hub = InventorHub()

        left_motor = Motor(port=Port.B,
                           positive_direction=Direction.COUNTERCLOCKWISE)

        right_motor = Motor(port=Port.A,
                            positive_direction=Direction.CLOCKWISE)

        self.drive_base = DriveBase(left_motor=left_motor,
                                    right_motor=right_motor,
                                    wheel_diameter=self.WHEEL_DIAMETER,
                                    axle_track=self.AXLE_TRACK)

        self.kicker_motor = Motor(port=Port.C,
                                  positive_direction=Direction.CLOCKWISE)

        self.distance_sensor = UltrasonicSensor(port=Port.D)

        self.color_sensor = ColorSensor(port=Port.E)

    def celebrate(self):
        self.hub.display.image(
            image=[[00, 11, 33, 11, 00],
                   [11, 33, 66, 33, 11],
                   [33, 66, 99, 66, 33],
                   [11, 33, 66, 33, 11],
                   [00, 11, 33, 11, 00]])

        self.hub.speaker.beep(
            frequency=1000,
            duration=1000)

        self.hub.light.animate(
            colors=[Color.CYAN, Color.GREEN, Color.MAGENTA],
            interval=100)

        self.drive_base.drive(
            speed=0,
            turn_rate=360)

        # *** THIS BLOCK CAUSES THE MOVEMENT TO GO ON FOREVER ***
        # self.kicker_motor.run_angle(
        #     speed=1000,
        #     rotation_angle=5 * 360,
        #     then=Stop.COAST,
        #     wait=True)

        wait(1000)

        self.hub.display.off()

        self.hub.light.on(color=Color.BLACK)

        self.drive_base.stop()

TRICKY = TrickyPlayingSoccer()
TRICKY.celebrate()
```

laurensvalk, 2021-01-05:
Thanks and good timing! I’m planning to look at motors tomorrow.

Thanks also for the example to reproduce.

If you find more issues, it would be great if you could try to make the example as small as possible next time. Preferably with only one motor or sensor if that is enough to reproduce it.

TheVinhLuong102, 2021-01-05:
@laurensvalk yes we'll try to make the bug examples small to isolate the problem more quickly. In fact we discovered this bug a few days back and have been trying to isolate the issue, but without success. The individual code blocks do run ok, but when we run the whole thing together the combo doesn't work. :)

laurensvalk, 2021-01-05:
That’s good to know, thanks!

laurensvalk, 2021-01-06:
I can't seem to reproduce it just yet.

> In the below program, when the commented-out block is included, the robot movement would go on until the program is forcefully terminated.

To be specific, what do you mean with "robot" movement? Does the drivebase not stop or does the `kicker_motor` not stop?

Also, does the program freeze? A good way to tell would be to omit displaying the `image` in your program so you can see if the run animation freezes or not.

laurensvalk, 2021-01-06:
> program is forcefully terminated.

Was pressing the hub button enough, or did you have to reset with a long press?

laurensvalk, 2021-01-06:
I have just fixed another motor control bug. Depending on your answers to the questions above, this may have been the same issue.

Here is a [build](https://github.com/pybricks/pybricks-micropython/actions/runs/465883843) with that potential fix. It would be great if you want to retry the issue *before* you test the new build, just to be more sure that the new build fixes it.

Thanks for the help!

TheVinhLuong102, 2021-01-06:
@laurensvalk, with the existing build, when I uncomment the concerned block then the code lines after that do not seem to have any effect:
- `self.hub.display.off()`: the display remains on
- `self.hub.light.on(color=Color.BLACK)`: the button keeps animating Cyan, Green, Magenta, Cyan, Green, Magenta,...
- `self.drive_base.stop()`: the drivebase keeps turning forever

The program doesn't freeze, because I can still see the button colors animating continuously, and the drivebase still keeps turning. I can terminate the program with a usual short press.

TheVinhLuong102, 2021-01-06:
@laurensvalk I can confirm that the latest firmware build has resolved this issue successfully. :)

Could you clarify the nature of the bug that was fixed?

laurensvalk, 2021-01-06:
Thanks for your responses. Glad to hear it got fixed.

I think it’s what I mentioned above; the kicker motor motion just did not complete because the controller thought it was stalled even though it wasn’t.

This could happen when the motor was ahead of its target position, which can happen briefly in some mechanisms. The fix was to consider it stalled if there was a large position error only if it’s behind the target.

laurensvalk, 2021-01-06:
So I guess we can close this. Thanks a lot! 🎉

closed 2021-01-06 by laurensvalk

TheVinhLuong102, 2021-01-06:
@laurensvalk thank you. Yes, I think the simultaneous turning and kicking may have complicated the overall movement and measurements and caused this. It's certainly an interesting case. :)

Thanks again for the quick fix!