pybricks/support #201: [Bug] Close resolved SyntaxError notifications

laurensvalk, 2021-01-06:
**Describe the bug**
When a SyntaxError is raised, the notification stays around even after you fix the syntax error.

**To reproduce**
Steps to reproduce the behavior:
1. Go to Pybricks Code
2. Write a program with a syntax erorr and click run
3. Fix the error and click run again
4. Notification stays around

**Expected behavior**
Old syntax errors should go away when you click run.

**Screenshots**
[image]

dlech, 2021-01-22:
This should be fixed now.

BertLindeman, 2021-01-23:
Can confirm this is fixed.

I wanted to show the related web version, but the `about box` does not state that precise enough, I think:

[image]

dlech, 2021-01-23:
The web app has been unversioned up to this point. So the fact that it shows a version at all says that you did in fact get the update.

closed 2021-01-26 by dlech