pybricks/support #205: Experience with updated firmware a13

BertLindeman, 2021-01-09:
**Question**
Not really a Q, but just reporting "No problem" updating / using the a13 firmware.

**Context**
Used "MS Edge Versie 87.0.664.75 (Officiële build) (64-bits)" this time to update two Technichubs and a MoveHub.

**Screenshots**
none now

**MoveHub**
First firmware upgrade failed:
prev firmware on movehub was :v3.0.0a12-14-g2c0e5f45 on 2020-12-19

update firmware states:
```
	Unexpected error while trying to connect. Check console log and report the error.
	error-log.ts:32 DOMException: Connection failed for unknown reason.
```	

movehub shows the blinking colors of "updating"
No progress bar movement (Still at 100% of prev update of TechnicHub).
	
Waited 5 minutes before I stopped the movehub
Paired and see that the update did not succeed:
```
('movehub', '3.0.0a12', 'v3.0.0a12-14-g2c0e5f45 on 2020-12-19')
```

Did "ctrl-F5" to start fresh:
Re-try flashing:
Now I see progressbar moving ;-)

And retry succeeds..

laurensvalk, 2021-01-09:
Thank you. Maybe we can enforce refresh somehow after we release an update.

In my case the whole screen is blank after the update, so I'm kind of forced to press `ctrl+F5` :)

dlech, 2021-01-09:
I had the blank screen issue too. I found that there were two separate web worker registrations in Chromium (maybe related to local development?). I deleted one and it seems to be OK since. 

Here is how to get to the service worker registrations:

[image]

I had two for `code.pybricks.com`. One was not activated, so I couldn't change it, but the other looked like this and I unregistered it:

[image]

BertLindeman, 2021-01-24:
As we are further in the firmware updates: closing this positive issue

closed 2021-01-24 by BertLindeman