pybricks/support #206: [Bug] idna module missing in pybricksdev environment (Poetry install file?)

joosbuijsNL, 2021-01-09:
**Describe the bug**
idna module is missing (in Poetry install file?) in pybricksdev environment

**To reproduce**
Steps to reproduce the behavior:
1. Follow the instructions at https://github.com/pybricks/pybricksdev to setup a pybricksdev environment (or navigate to this environment in a command prompt)
2. Run the command `poetry run pybricksdev run --help` as described under https://github.com/pybricks/pybricksdev#running-pybricks-micropython-programs

**Expected behavior**
Output with help and/or example information.

**Screenshots**
[image]

joosbuijsNL, 2021-01-11:
The problem seems to be a bit more subtle, because idna is of course already present in the poetry lock file.

However, for me it still fails (tried yesterday from a fresh pybricksdev and pybricksdev-demos). What works is do a `pip install idna` and then run `poetry install` again... I assume the versions are not as intended in the poetry file but at least the errors 'no module named idna' disappear.

dlech, 2021-04-08:
Is this still a problem? I haven't been able to reproduce the issue.

closed 2021-05-19 by dlech