pybricks/support #207: [Bug] pybricksdev-demo basics.ipynb BleakError

joosbuijsNL, 2021-01-09:
**Describe the bug**
Running the last cell in basics.ipynb raises a `BleakError: Characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e was not found!` error

**To reproduce**
Steps to reproduce the behavior:
1. Activate/open the pybricsdev-demo project
2. Run the basics.ipynb
3. Scroll down to the third cell where the main.py file is send to the hub
4. See error

**Expected behavior**
The motor connected to the hub turning happiliy, and 'Hello!'-s on the console.

**Screenshots**
[image]
[image]

joosbuijsNL, 2021-01-11:
To add, I was able to flash new firmware to the same TechnicHub, so the code and connection should be working. Also, code.pybricks.com happily works on the same hub and laptop. I also checked out pybricksdev and pybricksdev-demo after this report but same result.

Any hints/directions towards the rootcause of this error would be very welcome to help me start debugging.

Nelvarion, 2021-01-12:
How are you running the notebook?
I've had these kind of errors when running in an environment where poetry and the required packages weren't installed properly.

joosbuijsNL, 2021-01-18:
Hi @Nelvarion ,

thanks for replying, and apologies for the late feedback.

Based on your tip I cleaned my Conda environment, and re-installed poetry and ran the installer. Unfortunately I still get the same error. My list of installed packages via `pip list` (from the Jupyter notebook):

````
 Package               VersionNote: you may need to restart the kernel to use updated packages.
--------------------- -------------------
aioserial             1.3.0
appdirs               1.4.4
argcomplete           1.12.2
argon2-cffi           20.1.0
async-generator       1.10
asyncssh              2.5.0
attrs                 20.3.0
backcall              0.2.0
bleach                3.2.1
bleak                 0.9.1
CacheControl          0.12.6
cachy                 0.3.0
certifi               2020.12.5
cffi                  1.14.4
chardet               4.0.0
cleo                  0.8.1
clikit                0.6.2
colorama              0.4.4
crashtest             0.3.1
cryptography          3.3.1
cycler                0.10.0
decorator             4.4.2
defusedxml            0.6.0
distlib               0.3.1
doc8                  0.8.1
docutils              0.16
entrypoints           0.3
filelock              3.0.12
flake8                3.8.4
html5lib              1.1

idna                  3.1
ipykernel             5.4.2
ipython               7.19.0
ipython-genutils      0.2.0
ipywidgets            7.6.2
jedi                  0.18.0
Jinja2                2.11.2
jsonschema            3.2.0
jupyter               1.0.0
jupyter-client        6.1.7
jupyter-console       6.2.0
jupyter-core          4.7.0
jupyterlab-pygments   0.1.2
jupyterlab-widgets    1.0.0
keyring               21.8.0
kiwisolver            1.3.1
lockfile              0.12.2
MarkupSafe            1.1.1
matplotlib            3.3.3
mccabe                0.6.1
mistune               0.8.4
mpy-cross             1.12
msgpack               1.0.2
nbclient              0.5.1
nbconvert             6.0.7
nbformat              5.0.8
nbstripout            0.3.9
nest-asyncio          1.4.3
notebook              6.1.6
numpy                 1.19.4
packaging             20.8
pandocfilters         1.4.3
parso                 0.8.1
pastel                0.2.1
pbr                   5.5.1
pickleshare           0.7.5
Pillow                8.0.1
pip                   20.3.3
pkginfo               1.6.1
poetry                1.1.4
poetry-core           1.0.0
prometheus-client     0.9.0
prompt-toolkit        3.0.8
pybricks              3.0.0a3
pybricksdev           0.1.0
pycodestyle           2.6.0
pycparser             2.20
pyflakes              2.2.0
Pygments              2.7.3
pylev                 1.3.0
pyparsing             2.4.7
pyrsistent            0.17.3
pyserial              3.5
python-dateutil       2.8.1
pythonnet             2.5.1
pyusb                 1.1.0
pywin32               300
pywin32-ctypes        0.2.0
pywinpty              0.5.7
pyzmq                 20.0.0
qtconsole             5.0.1
QtPy                  1.9.0
requests              2.25.1
requests-toolbelt     0.9.1
restructuredtext-lint 1.3.2
Send2Trash            1.5.0
setuptools            51.0.0.post20201207
shellingham           1.3.2
six                   1.15.0
stevedore             3.3.0
terminado             0.9.1
testpath              0.4.4
toml                  0.10.2
tomlkit               0.7.0
tornado               6.1
tqdm                  4.55.0
traitlets             5.0.5
urllib3               1.26.2
validators            0.18.2
virtualenv            20.3.1
wcwidth               0.2.5
webencodings          0.5.1
wheel                 0.36.2
widgetsnbextension    3.5.1
wincertstore          0.2
```

Not sure how to proceed or debug further.

code.pybricks.com works just fine as I just flashed a new firmware before trying again.

Nelvarion, 2021-01-23:
Sorry, that's the limit of my troubleshooting knowledge currently.
Maybe @laurensvalk can help here

dlech, 2021-04-08:
We've made some updates to `pybricksdev` recently. Does this problem still exist?

closed 2021-07-13 by laurensvalk