pybricks/support #208: [Bug] Short Press did not boot stock firmware

HeyLlama, 2021-01-11:
I have "successfully" flashed ('primehub', '3.0.0a13', 'v3.0.0a13-1-g2aa1cfef on 2021-01-08') into my inventor hub. At least the Mindstorms app was saying it was successful.

It did reboot right after flashing but it was booting up pybricks instead of standard firmware as stated in #167 . Tested https://code.pybricks.com/ on Chrome, that works.

I tried both short press and long press. It was all pybricks for some reason. I'm not sure how to access standard firmware to charge the battery...Please help : (

HeyLlama, 2021-01-11:
https://github.com/pybricks/pybricksdev/blob/master/README_dfu.rst
This saved me. Thank you 🙏

laurensvalk, 2021-01-11:
Thanks for sharing this.

What happened when you tried a short press (~half a second)? Did Pybricks come on straight away? Or did nothing happen at all?

If you ever feel like trying this again, it would be great to know what happens if you power on by plugging in USB, instead of pressing any buttons. The expected result is the same as a short press, so stock firmware.

For your information, we intend to implement the DFU recovery step in our online app so users won't have to install anything. That's why the Prime Hub / Inventor Hub is not in beta just yet for now.

HeyLlama, 2021-01-11:
I'll try again tonight now I know how to restore the stock OS :)

Half a second short press didn't boot up anything. Nor does plugging in USB.

One red flag was after flashing Pybricks, my hub rebooted Pybricks instead of stock firmware. I'll let you know if I could reproduce this behaviour.

Cheers

laurensvalk, 2021-01-11:
I can confirm this. Thanks a lot for highlighting this issue.

laurensvalk, 2021-01-11:
I think we will update that installation guide with a link to a fixed version that is known to work, instead of the latest build, just in case.

I'm rolling back a few changes to make one that works.

laurensvalk, 2021-01-11:
Can you try it with [this builld](https://github.com/pybricks/pybricks-micropython/suites/1808538282/artifacts/34734192)?

That would be super helpful - thanks!

laurensvalk, 2021-01-11:
In the top comment you should see the version. Just to make sure, you need
```python
# Version: v3.0.0a13-7-g97e3745a
```

HeyLlama, 2021-01-11:
Will do! I'll let you know

Cheers

HeyLlama, 2021-01-11:
Unfortunately, the build you provided did not work either. Same behaviour, no reboot at all this time after the bluetooth light blinked pink-green-blue. No reaction from plugging in USB cable nor short press. Long press does boot Pybricks tho. I'm starting to suspect if this has anything to do with the newest firmware update from MINDSTORMS. Perhaps they changed the bootloader?

HeyLlama, 2021-01-11:
[image]
This was the console output from the app.

laurensvalk, 2021-01-11:
Thanks for looking into this. This is interesting.

Good reason to expedite the work on the easy installer and recovery from within our online app 😄

HeyLlama, 2021-01-11:
Can't wait for the beta release! Thank you for the hard work 😄

HeyLlama, 2021-01-11:
[image]
I ran the download program again and this showed up in the console. Looks like there was an error before the installation program.

laurensvalk, 2021-01-11:
This is the expected output for the download program. It’s a lot of data that doesn’t do anything by itself.

The installer then reads it as a text file. Line by line instead of all at once to avoid that memory error.

HeyLlama, 2021-01-11:
Gotcha. Thank you!

laurensvalk, 2021-01-12:
To get a taste of the easier recovery, check out [this page](https://github.com/devanlai/webdfu).

For vendor ID, choose `0x0694`. After connecting, choose this from the given options:

[image]

You can use the default for everything else.

The only caveat is that some sort of USB driver must already be installed or access to the USB device must be given. For Windows [this one](https://zadig.akeo.ie/) works.

danielgaertner, 2021-01-12:
Hello, my name is Daniel. I am a beginner and I flash the pybricks firmware on my lego inventor hub with the original lego software for use in Linux. The dual boot dosn't work any more with the new firmware from pybricks. Can you tell me easy and exact steps to flash a new firmware on the inventor hub with dual boot, or restore the original lego firmware. Thank you!

laurensvalk, 2021-01-12:
Thanks Daniel. Yes, we’ll give you step by step instructions. Specifically which OS do you have?

I’m also going to temporarily close the beta on the dual boot, until we document the restore step a bit better. Thank you both for highlighting this.

HeyLlama, 2021-01-12:
> Hello, my name is Daniel. I am a beginner and I flash the pybricks firmware on my lego inventor hub with the original lego software for use in Linux. The dual boot dosn't work any more with the new [...]

Hi Daniel, I'm running MacOS so if you have other unix based system this method should work too.

Follow the steps to install pybricksdev
https://github.com/pybricks/pybricksdev

Make sure poetry and pyenv are installed. And you'd need dfu-util too. I’d recommend homebrew for installing these packages: https://brew.sh/

Pick a newer firmware (1.0ish) from the firmware folder and download (.bin), rename to a simpler name if you will
https://github.com/gpdaniels/spike-prime

This shows you how to get into DFU mode
https://github.com/gpdaniels/spike-prime/blob/master/firmware/README.md

To check if your hub is connected, use this command:
sudo dfu-util --list

Finally, to restore the stock firmware:
https://github.com/pybricks/pybricksdev/blob/master/README_dfu.rst

Remark: remember to cd into pybricksdev before calling the restore command

laurensvalk, 2021-01-12:
Thanks for helping out @HeyLlama --- much appreciated.

If you want to try it, I'm curious to hear if the web page above works on Mac. I've confirmed it on Windows and Linux. And it seems a bit easier than installing `pybricksdev` for most people.

laurensvalk, 2021-01-12:
Daniel, feel free to ask if you need any additional help.

HeyLlama, 2021-01-12:
Hmm...I can't seem to find 0x0694 in the Vendor ID list on MacOS. Hub was indeed in DFU mode since I see it in the dfu-util --list
OS version is Catalina 10.15.7
Chrome version 87.0.4280.141

laurensvalk, 2021-01-12:
You can just type it in. The listed devices (dapboot, mbed) are only (apparently frequently used) suggestions.

danielgaertner, 2021-01-12:
Yes I need more help! I tried to install poetry and a runtime error appares!
daniel@PC-Stift:~$ poetry install

  RuntimeError

  Poetry could not find a pyproject.toml file in /home/daniel or its parents

  at .local/lib/python3.8/site-packages/poetry/core/factory.py:367 in locate
      363│             if poetry_file.exists():
      364│                 return poetry_file
      365│ 
      366│         else:
    → 367│             raise RuntimeError(
      368│                 "Poetry could not find a pyproject.toml file in {} or its parents".format(
      369│                     cwd
      370│                 )
      371│             )

danielgaertner, 2021-01-12:
Poetry is now installed correctly! The error above is when I type
poetry run pybricksdev udev | sudo tee /etc/udev/rules.d/99-pybricksdev.rules

laurensvalk, 2021-01-12:
> Poetry could not find a pyproject.toml file in /home/daniel or its parents

Did you download (or clone) pybricksdev? You need to be in that folder before calling `poetry install`. Then it can find the required files.

HeyLlama, 2021-01-12:
Hey Daniel! Try this one instead: https://devanlai.github.io/webdfu/dfu-util/ You'd just need the .bin firmware for this one.
Much easier. Thank you! @laurensvalk 
[image]
[image]

laurensvalk, 2021-01-12:
I guess that confirms it works on Mac too :smile: Thanks!

danielgaertner, 2021-01-12:
whitch vendor id can I use!

HeyLlama, 2021-01-12:
> whitch vendor id can I use!

Type in 0x0694 and select the Lego option
PS make sure your hub is in DFU mode

laurensvalk, 2021-01-12:
Daniel, check out [this post above](https://github.com/pybricks/support/issues/208#issuecomment-758524669) which has some screenshots for guidance.

danielgaertner, 2021-01-12:
I shows SecurityError: Access denied.

laurensvalk, 2021-01-12:
At which step does it show that?

danielgaertner, 2021-01-12:
After I tried to connect!

laurensvalk, 2021-01-12:
Would you like to move over to this discussion forum [here](https://gitter.im/pybricks/community)?

laurensvalk, 2021-01-15:
@HeyLlama, I think I was able to reproduce it. It was due to two distinct issues so it took a while to find it. I would be greatful if you want to test [this build](https://github.com/pybricks/pybricks-micropython/suites/1836443916/artifacts/35500786). (v3.0.0a13-5-g9a80bef6).

HeyLlama, 2021-01-15:
> @HeyLlama, I think I was able to reproduce it. It was due to two distinct issues so it took a while to find it. I would be greatful if you want to test [this build](https://github.com/pybricks/pybricks-micropython/suites/1836443916/artifacts/35500786). (v3.0.0a13-5-g9a80bef6).

Excellent! I'll let you know :D

HeyLlama, 2021-01-15:
v3.0.0a13-5-g9a80bef6 works! Thank you so much!!

closed 2021-01-15 by HeyLlama

laurensvalk, 2021-01-15:
Yes! :rocket: Thanks for the help.

I've updated #167 to use this version instead of the latest build, just to be safe. Latest builds are only for those who know how to use recovery mode, which now includes you :wink:

laurensvalk, 2021-01-16:
@HeyLlama, @danielgaertner: https://beta.pybricks.com/dfu/

HeyLlama, 2021-01-16:
> @HeyLlama, @danielgaertner: https://beta.pybricks.com/dfu/

Can confirm it works :D

laurensvalk, 2021-01-16:
The firmware zip file also contains the .bin file for Pybricks, by the way. That will give you the regular boot instead of dual boot.

danielgaertner, 2021-01-17:
I installed build # 838! No dual boot possible! Only Pybricks! In future I will use a older build!

danielgaertner, 2021-01-17:
I tried to restore the original lego firmware with

 "poetry run pybricksdev dfu restore ~/Downloads/96112deb24a934bfc19a13c7ea620f54.bin" 

and the error 
"dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available" 

appiers!

danielgaertner, 2021-01-17:
Now it works! I have not put the hub in DFU Mode!

laurensvalk, 2021-01-17:
You can also restore the original firmware with that website. It's just a convenient alternative to `pybricksdev`.

Thanks for sharing that it is working for you.