pybricks/support #209: pandas or csv.reader compatible function

faizalfada, 2021-01-11:
hi would like to read an csv files and as transform it into an array or list.  i try to use csv module, but no reader or dictreader available. Can we install pandas  (pip install pandas) into ev3 bricks or what is the compatible function to read a structured csv   ?

# READ DATA
import csv
filename="shoulder_elbow_1.csv"
with open(filename,'r') as data:
    for line in csv.reader(data):
        print(line)

AttributeError: 'module' object has no attribute 'reader'

dlech, 2021-01-11:
Hi, Pybricks is based on [MicroPython](https://micropython.org/) instead of regular Python, so most Python libraries are not available.

For simple csv files (e.g. no quoted strings, no extra spaces) you could read it like this:

```python
with open(filename, 'r') as data:
    for line in file.readlines():
        line = line.strip()
        row = line.split(',')
        print(row)
```

faizalfada, 2021-01-12:
hi, thanks alot for your support. it runs very well

closed 2021-01-12 by faizalfada