pybricks/support #212: [Bug] Bluetooth disconnect does not stop a running program on the hub

BertLindeman, 2021-01-13:
**Describe the bug**
Bluetooth disconnect while running REPL in code.pybricks does not stop the program running on the hub.
Trying to re-connect does not show the hub.
Stop and reboot the hub to be able to re-connect.
No data is lost, just some of my time. 
So **not** very important.

**To reproduce**
Steps to reproduce the behavior:
1. Go to code.pybricks.com
2. long press the center button on the (in this case) primehub to boot the pybricks firmware (button gets blue)
3. Press the "connect using bluetooth" button, select the "pybricks hub" and press "pair"
4. Press the "Start REPL in terminal" button
5. Press the "Disconnect Bluetooth" button
6. See the hub still running (moving square on the display)
7. Press the "connect using bluetooth" button and see **no** hub to select

**To recover**
1. Press and keep the blue button on the hub to shutdown
2. Redo steps 2. through 4. from "to reproduce"

**Expected behavior**
On "Disconnect Bluetooth" stop any program running on the hub

**prevent the problem**
Do not disconnect while in REPL or running a program ;-) 

**Screenshots**
Running hub: 
[image]

**Environment**
win10 
Google Chrome Version 87.0.4280.141 (Official Build) (64-bit)

primehub with firmware:
MicroPython v3.0.0a13-7-g97e3745a on 2021-01-11; SPIKE Prime Hub with STM32F413VG

laurensvalk, 2021-01-13:
Thanks - whether or not we should allow reconnect while a program is running is indeed an interesting design choice.

We used to have it, but disabled it for technical reasons. It could be worth bringing this back. It would also allow you to disconnect, leave the program running, and reconnect using another device, which is sometimes useful.

BertLindeman, 2021-01-13:
> 
> 
> Thanks - whether or not we should allow reconnect while a program is running is indeed an interesting design choice.

Sure it is.

> We used to have it, but disabled it for technical reasons. It could be worth bringing this back. It would also allow you to disconnect, leave the program running, and reconnect using another device, which is sometimes useful.
> 
> I would probably argue to keep the program running when disconnecting though. In addition to the reason mentioned above, this way your robot won't suddenly stop once out of reach of your computer.

Just for the sake of discussion:
Would you want your robot to run on (out of control), if out of reach?
It has pro and con arguments.

However... When using the LEGO firmware:
You can choose a program to run, no control from a PC or such needed.
That is tempting.
Maybe I do have a preference ;-)

Nelvarion, 2021-01-15:
> We used to have it, but disabled it for technical reasons. It could be worth bringing this back. It would also allow you to disconnect, leave the program running, and reconnect using another device, which is sometimes useful.

How about a default behavior that is configurable?

laurensvalk, 2021-01-15:
> How about a default behavior that is configurable?

That would be much harder. We'll be looking at the way we handle Bluetooth more broadly in the near future and may be able to look at this when we do that.

dlech, 2021-02-17:
One thing I have noticed recently is that if you are using the REPL and then disconnect Bluetooth, there is no way to reconnect and no way to stop the REPL, so you just have to reboot the hub.

dlech, 2021-05-17:
>  if you are using the REPL and then disconnect Bluetooth, there is no way to reconnect and no way to stop the REPL, so you just have to reboot the hub.

This was fixed a while ago.

dlech, 2021-05-17:
I think this issue (program keeps running after disconnect) is working as designed and can be close. We can start a new issue for "can't connect to Bluetooth while user program is running" if there is interest/need for such a feature.

closed 2021-06-17 by dlech