pybricks/support #214: [Bug] Inventor hub does not boot in original os

StephanP1979, 2021-01-16:
I flashed an version of the pybricks firmware (~6th Jan 2021) on my inventor hub. Since then, the hub does not charge and or boot in lego os.  With an earlier version everthing was fine.
I now read that there is already a bugfix, but how can I remove the current (not running) version of the firmware without being able to boot the lego firmware?

dlech, 2021-01-16:
Please see the discussion in https://github.com/pybricks/support/issues/208 and let us know if you run into any problems.

laurensvalk, 2021-01-16:
There's no need to read all of #208; the bug has been found and fixed.

In principle, you just need to follow the steps on [this page](https://beta.pybricks.com/dfu/).

StephanP1979, 2021-01-16:
Puh! Thank's a lot! First I couldn't connect, but after installing the Zadig tool, i could. => Maybe you should add a hint to your instructions on https://beta.pybricks.com/dfu/ ...?

laurensvalk, 2021-01-16:
It's already there, was working on it as we speak :rocket: . May need to refresh the page :)

laurensvalk, 2021-01-16:
Glad to hear you got your hub back :)

closed 2021-01-18 by laurensvalk