pybricks/support #215: [Question] speaker.play_notes() on Inventor Hub

johnscary-ev3, 2021-01-17:
**Question**
I have been using speaker.play_notes() to play music on Inventor Hub.
It works great, but I have a couple questions:

1. Is there a way to use it in a non-blocking way so I can do other things while playing a long tune?
2. Will there be a way to control the volume?

**Context**
Would like to play tunes while moving around or taking some other actions.

Thanks.

dlech, 2021-01-17:
For now you have to write your own async code [like this](https://gist.github.com/dlech/066cc9593eecf88d044963a0a374f65e). Having a built-in way to do this is farther down our roadmap (i.e. Pybricks v4).

Volume control should be possible in the shorter term though.

johnscary-ev3, 2021-01-18:
Thanks for the 'background music', music.py,  program.
That works great integrated into my program.
I was thinking that timers and yield statements would probably be needed.
Now I just need to do the same thing to other parts of my program running motors at the same time ;0)
I think I know how to do that using motor.control.done() and yield statements also.

Volume control would be nice when working on the robot late at night ;0)