pybricks/support #217: [Bug] slice of list issues error: 'int'  object isn't subscriptable

BertLindeman, 2021-01-18:
**Describe the bug**
list object cannot be sliced

**To reproduce**
Steps to reproduce the behavior:
1. code.pybricks.com and paste this program into it
```
try:
    from pybricks import version
    print(version)
except:
    pass
e = ['[Errno', '19]', 'ENODEV:', 'A', 'sensor', 'or']
print("type(e)", type(e))
print(e[:3])
```

2. Connect to a hub 
3. Make the console part visible
4. Run the program
5. See the error: 
```
----------
('technichub', '3.0.0a13', 'v3.0.0a13 on 2021-01-08')
type(e) <class 'list'>
Traceback (most recent call last):
  File "main.py", line 5, in <module>
TypeError: 'int' object isn't subscriptable
----------
('primehub', '3.0.0a13', 'v3.0.0a13-5-g9a80bef6 on 2021-01-15')
type(e) <class 'list'>
Traceback (most recent call last):
  File "main.py", line 5, in <module>
TypeError: 'int' object isn't subscriptable```

**Expected behavior**
Running the same program on EV3 micropython like this:
` micropython issue_int_not_subscriptable.py` 
shows the result like this:
```
type(e) <class 'list'>
['[Errno', '19]', 'ENODEV:']

```

**Screenshots**
None at the moment

laurensvalk, 2021-01-18:
See also #140. We can probably enable this on most hubs, but we'll need to do some soul searching for how to deal with Move Hub in the future.

BertLindeman, 2021-01-18:
Ah, thanks @laurensvalk 

**the** problem I have with this issue is the reported error is misleading me.
So I go debug my code in the wrong direction.

I understand that making the error message fitting the problem **does** increase the program-memory used.

Wrong error messages like this one might throw off other python newbies later.

**what was I trying to do**
I was trying to get only the start part of `Errno 19` to drop the following lines  for easy of debugging:
```
OSError: [Errno 19] ENODEV: 

A sensor or motor is not connected to the specified port:
--> Check the cables to each motor and sensor.
--> Check the port settings in your script.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.
```

laurensvalk, 2021-01-18:
> the reported error is misleading

I see what you mean now. It looks like this might be a MicroPython issue. I'm going to have a look at it.

laurensvalk, 2021-01-18:
```python
l = ["a", "b", "c", "d", "e"]
l[:]
```
Technic Hub v3.0.0a13 on 2021-01-08:
```
TypeError: 'int' object isn't subscriptable
```

Minimal MicroPython Unix port (both on latest build and v1.13) I get:
```
SyntaxError: invalid syntax
```
Neither are great and the discrepancy is a bit odd, which may depend on other features being enabled or not. It seems like a minor issue though.

BertLindeman, 2021-01-18:
>  It seems like a minor issue though.

I agree Laurens, especially in the current phase.

Added my up vote for slicing in #140 
Closing this issue.

Thank you for the fast support, @laurensvalk

closed 2021-01-18 by BertLindeman

laurensvalk, 2021-01-18:
Thanks for reporting. Would be great to have your opinion about Move Hub in #218.

BertLindeman, 2021-01-18:
> Thanks for reporting. Would be great to have your opinion about Move Hub in #218.

Will do.