pybricks/support #219: [Bug]  Not all ports on Primehub see sensor or motor devices

BertLindeman, 2021-01-19:
**Describe the bug**
(As an example) The SPike Color sensor is not recognized/initiated on all Primehub ports similar on inventorhub 

**To reproduce**
Steps to reproduce the behavior:
1. Go to code.pybricks.com and connect the primehub
2. Insert the color sensor into a port and wait 5 seconds to allow the hub to recognize the sensor
3. See if the lights on the sensor light up
4. If not lit ifs obviously not recognized
5. Run the program (see below) and see if the sensor is reported

**Expected behavior**
The sensor should be reported on each of the ports
Did the same test with a Control+ motor with similar result.

**Condensed result Color sensor:**
| port | Primehub | Inventorhub |
| ---- | -------- | ----------- | 
| A | OK | OK |
| B | not lit | not lit |
| C | OK | not lit |
| D | lit but not reported | not lit |
| E | OK | not lit |
| F | not lit | not lit |

**Screenshots**

**Spike app**

[image]

**Mindstorms app**

[image]

**source**
``` python
#!/usr/bin/env pybricks-micropython

# source at Z:\home\bert\py\pybricks\debug_invertorhub\debug_controlPLUSmotor.py

from pybricks.tools import wait

from pybricks.iodevices import PUPDevice

try:
    # from pybricks.pupdevices import ColorDistanceSensor
    from pybricks.pupdevices import Motor
    # from pybricks.pupdevices import ColorSensor
    # from pybricks.pupdevices import UltrasonicSensor
    # from pybricks.pupdevices import ForceSensor
except Wxcept as e:  # NOQA:  E722 do not use bare 'except'
    print("could not import all pupdevices", e)

from pybricks.parameters import Port
# from pybricks.tools import wait

# create dictionary from device.info()['id'] =-> description
__devID = {}

__devID[46] = "Technic Control+ Large Motor "
__devID[47] = "Technic Control+ XL Motor    "
__devID[48] = "SPIKE Prime Medium Motor     "
__devID[49] = "SPIKE Prime Large Motor      "
__devID[61] = "SPIKE Prime Color Sensor     "
__devID[62] = "SPIKE Prime Ultrasonic/Distance Sensor"
__devID[63] = "SPIKE Prime Force Sensor     "

def who_am_i(xport):  # noqa C901 'who_am_i' is too complex (24)
    #  try to initialize generic device:
    try:
        # print("try Port:", xport)
        device = PUPDevice(xport)
    except OSError as e:
        print("40 OSError on:", xport, ":", e)
        return
    except ValueError as e:
        print("43 ValueError Exception", e, "occurred on", xport)
        return
    except Exception as e:
        print("46 General Exception", e, "occurred on", xport)
        return
    else:
        try:
            print("%-7s device.info()['id']=%2d " % (xport, device.info()['id']), end="")
        except ValueError as e:
            print("52 ValueError Exception ", e, "occurred on", xport)
            return
        except Exception as e:
            print("55 General Exception", e, "occurred on", xport)
            return
        try:
            print(__devID[device.info()['id']], end="")
        except Exception as e:
		    pass
            # print("60 General Exception ", e, "occurred on port: ", xport)
            # return
	print()

who_am_i(Port.A)
who_am_i(Port.B)
who_am_i(Port.C)
who_am_i(Port.D)
who_am_i(Port.E)
who_am_i(Port.F)

```

laurensvalk, 2021-01-19:
Confirmed, thanks for reporting. This started breaking only recently.

BertLindeman, 2021-01-19:
Thanks.

Seems like not all ports react the same each time.

laurensvalk, 2021-01-19:
This was broken by https://github.com/pybricks/pybricks-micropython/commit/01f1e17d8e1136bbc4d525758a3d90ca71d2d20a

laurensvalk, 2021-01-19:
The attempted fix is being built now and should be ready in a  few minutes:

https://github.com/pybricks/pybricks-micropython/runs/1727770616

laurensvalk, 2021-01-19:
https://github.com/pybricks/pybricks-micropython/suites/1856856559/artifacts/36056567 :tada:

BertLindeman, 2021-01-19:
> The attempted fix is being built now and should be ready in a few minutes:
> 
> https://github.com/pybricks/pybricks-micropython/runs/1727770616

Installed firmware from here (click artifacts etc..) Is the firmware on:
```
 https://github.com/pybricks/pybricks-micropython/suites/1856856559/artifacts/36056567 🎉
```
different?

Did a visual test on the primehub by inserting the Spike color sensor and later the colorDistance sensor in each port.
All lit up nicely.

If the two links point to the same firmware, this issue can be closed.

Blazing fast ;-)

laurensvalk, 2021-01-19:
They were the same. 

> All lit up nicely.

Excellent. Thanks for reporting and testing so quickly, I appreciate it. I will update the link in the installation support issue with this new version.

closed 2021-01-19 by laurensvalk

BertLindeman, 2021-01-19:
Great!
Just confirmed that the zips via both links are identical.