pybricks/support #220: [Feature] UARTdevice support for PoweredUp hubs

NStrijbosch, 2021-01-19:
Hi @laurensvalk and @dlech 

I would be very interested if the UARTDevice class, which you support for the EV3, could be made available on PoweredUp devices. Given the breakout point for third party sensors on the PoweredUp Ultrasonic sensor, this would be a feature that can not be missed in Pybricks. 

An identical implementation as for the EV3 would be great!

Thanks in advance,
Nard

laurensvalk, 2021-01-19:
Thanks for your request!

> An identical implementation as for the EV3 would be great!

I'm adding [the link here](https://docs.pybricks.com/en/latest/iodevices/uartdevice.html) for completeness.

A few high level questions / thoughts:
- Is the above feature set sufficient?
- What is the range of baudrates we should realistically support?
- What should happen with sensor auto-detect? Disable as soon as you initialize `UARTDevice`?
- What should happen when the user program ends? Reset the uart to a normal sensor or keep auto detect disabled? I imagine the auto-detect traffic might be undesired.

NStrijbosch, 2021-01-19:
*  Yes, feature set is sufficient. Although a readline() would make it complete. 
*  With the official firmware on the SPIKE Prime hub I have tested in the range (9600-115200) this would be more than fine. 
*  I guess there is a small chance that the traffic from auto detect could lead to an error in the thirdparty sensor due to not recognizing it.  The official firmware does something similar and does not turn of its auto-detect feature. I also think this error would be because of a bad software implementation on the sensor side, i.e, the chance is really small. So I am tempted to say just leave it turned on.
*  What would the behaviour be when you keep auto-detect turned on (as in the current implementation) in the cases: 
  1. Pluggin in a UART sensor (It does not get detected?)
  2. First pluggin a UART sensor then a PU sensor (does the PU sensor get detected?)

laurensvalk, 2021-01-19:
Thanks for the input!

> What would the behaviour be when you keep auto-detect turned on

When no known UART sensor is detected, there's a routine that reads/writes pins 5 and 6 in several permutations to find possible non-uart devices like the train motor. We wouldn't want this routine to run when a custom UART device is attached.

NStrijbosch, 2021-01-19:
Oké, I wrongly assumed the auto detection would consist of sending stuff over UART. In this case auto detection should be turned off for UART to work after initialisation.

In the current Pybricks version this autodetect is running when my hub is turned on and the program is not yet running.

If that is the case the custom UART device needs to be robust to these signals (I don't think this would be an issue). If the sensor cannot handle the autodetect signals you first need to run a program to init UART before plugging in the sensor?

dlech, 2021-01-19:
I would like to have this implemented using the MicroPython stream interface instead like `machine.UART`. It provides everything already in the API already (except for `clear()`?) and would give us things like `readline()` and `readinto()` for free. The the more we make it like `machine.UART`, the more portable code will be between Pybricks and the official LEGO firmware.

dlech, 2021-01-19:
> If that is the case the custom UART device needs to be robust to these signals (I don't think this would be an issue). If the sensor cannot handle the autodetect signals you first need to run a program to init UART before plugging in the sensor?

UARTs are usually pretty robust when it comes to unexpected signals, so I agree this isn't something to worry about much.

I think the current autodetect algorithm will send a few bytes at 115200 baud to UART devices when it first detects the device is plugged in. So if this is a problem, the we would need some kind of port configuration outside of the user program.

laurensvalk, 2021-01-20:
If we're going to change the API, it's worth considering having a look at [pySerial](https://pyserial.readthedocs.io/en/latest/pyserial_api.html), for which we've had several requests in the past.

> The the more we can make it like machine.UART, the more portable code will be between Pybricks and the official LEGO firmware.

In terms of portability, there is [way more](https://www.google.com/search?q=pyserial+arduino&oq=pyserial+arduino) example code for pySerial compared to a handful of hacks with the stock firmware. And it has things like `readinto` and `readline` as well.

`UARTDevice` is modeled after pySerial already, except for renaming a few QSTRs (we ran out of time for the 2.0 release to do that unfortunately, but we still can do it for 3.X.) We could still use the MicroPython implementation underneath either way.

dlech, 2021-03-16:
Here is a look at the pySerial API:

### Constructor parameters

    port – Device name or None.

This would be `Port.X` in Pybricks.

    baudrate (int) – Baud rate such as 9600 or 115200 etc.

This on is important.

    bytesize – Number of data bits. Possible values: FIVEBITS, SIXBITS, SEVENBITS, EIGHTBITS

We might be able to live without this. I've never seen a device that wasn't 8 bits. On the other hand, pretty much every serial device has an option for this.

    parity – Enable parity checking. Possible values: PARITY_NONE, PARITY_EVEN, PARITY_ODD PARITY_MARK, PARITY_SPACE

Not used often, but could be useful for some device.

    stopbits – Number of stop bits. Possible values: STOPBITS_ONE, STOPBITS_ONE_POINT_FIVE, STOPBITS_TWO

Probably used even less often than parity, but could be useful for some device.

    timeout (float) – Set a read timeout value.

This seems like a good thing to have.

    xonxoff (bool) – Enable software flow control.

Potentially useful since we don't have hardware flow control. Not sure how it works.

    rtscts (bool) – Enable hardware (RTS/CTS) flow control.
    dsrdtr (bool) – Enable hardware (DSR/DTR) flow control.

I/O ports lack the pins for hardware flow control, so we don't need this.

    write_timeout (float) – Set a write timeout value.

Not as important as read timeout. Might be able to live without it.

    inter_byte_timeout (float) – Inter-character timeout, None to disable (default).

Not sure how this is supposed to work. Probably not needed. On the other hand, this is one of the options in MicroPython, so maybe there is a use for it.

    exclusive (bool) – Set exclusive access mode (POSIX only). A port cannot be opened in exclusive access mode if it is already open in exclusive access mode.

Not applicable

---

For comparison, MicroPython `machine.UART` has a separate `init()` method. The constructor implicitly calls this method, and it can be called separately later if parameters need to be changed after the object is created. This seems like a useful way to implement it. 

Here are the parameters:

        baudrate is the clock rate.

        bits is the number of bits per character, 7, 8 or 9.

        parity is the parity, None, 0 (even) or 1 (odd).

        stop is the number of stop bits, 1 or 2.

        tx specifies the TX pin to use.

        rx specifies the RX pin to use.

        txbuf specifies the length in characters of the TX buffer.

        rxbuf specifies the length in characters of the RX buffer.

        timeout specifies the time to wait for the first character (in ms).

        timeout_char specifies the time to wait between characters (in ms).

        invert specifies which lines to invert.

We wouldn't need the pin specifiers or invert parameters. The buffer size parameters would be useful to balance memory usage with the actual applications needs. The rest correspond to pySerial parameters.

### Methods

- `open()` and `close()`: should be able to live without these (unless we need to control I/O port state after)
- `__del__()`: Not needed since UART object should be global and exist for entire duration of program. And we will have to have something else to clean up when the user program ends anyway. MicroPython has `deinit()` method that is probably similar.
- `read()`, `write()`, `flush()`, `readable()`, `writeable()`, `seekable()`, `readinto()`, `readline()`, `readlines()`, `writelines()`, `fileno()`: standard Python io methods - MicroPython UART has `read()`, `readline()`, `readinto()` and `write()`.
- `read_until()`: basically a customizable version of `readline()` - can live without it
-  `in_waiting()`, `out_waiting()`: these are essential for writing non-blocking code. MicroPython UART has `any()` which is the same as `in_waiting()`.
- `reset_input_buffer()`, `reset_output_buffer()`: potentially useful, we have something like this in PBIO already
- `send_break()`: Could be useful to "reset" after disabling port autodetection. MicroPython calls this `sendbreak()` (underscore between words is much better IMHO)
- `get_settings()`, `apply_settings()`: Can live without get_settings. Might be a use case for apply_settings, which is the equivalent of the MicroPython `init()` method.
- `__enter__()`, `__exit__()`: only make sense if there is open/close
- `set_input_flow_control()`, `set_output_flow_control()`: for manual control of software flow control, likely not needed/used
- `cancel_read()`, `cancel_write()`: only make sense if threading is enabled

### Properties

pySerial also has properties to change the settings after the object has been created. We can live without these. There are also some read-only properties to get the port/name and line states. We should be able to get by without those too.

dlech, 2021-03-16:
I was looking at this from the point of view of: would it make sense to use this for USB serial, RFComm (Bluetooth Classic) and BLE UART too, but none of those need the settings like baud rate, etc. And in the case of Bluetooth, we probably want things to be more socket-like (e.g. `listen` and `connect` methods). But some of the methods (in addition to the standard stream methods) are useful and could be shared, like `in_waiting()`.

dlech, 2021-03-17:
> * What is the range of baudrates we should realistically support?

This will just depend on the capabilities of the individual I/O port.

> * What should happen with sensor auto-detect? Disable as soon as you initialize `UARTDevice`?

Yes

> * What should happen when the user program ends? Reset the uart to a normal sensor or keep auto detect disabled?

To keep things simple, I would say reset the port. 

If this proves to be a problem, we could probably come up with some kind of port config file on hubs with external flash to prevent the port from being used for anything else.

> * What should happen if the user tries to initialize a sensor after they initialized a `UARTDevice`?

It should throw an exception since the port is in use already (the same should apply to trying to create a 2nd instance of anything on that port).

laurensvalk, 2021-03-17:
> I was looking at this from the point of view of, "would it make sense to use this for USB serial, RFComm (Bluetooth Classic) and BLE UART too?"

One thing I'm still wondering is whether our classes should "inherit" and act like a stream object themselves, or whether they should have or somehow return a stream object.

A dedicated class (`UARTDevice`, `serial.Serial`, etc.) as described in various ways above seems useful for physical ports and sensors.

But for RFComm and BLE UART it seems more common to have functions that return socket/stream objects, like e.g. `pybluez`. I'll sketch one such approach in the Bluetooth thread.

antonvh, 2021-12-07:
Yes UARTDevice! I want it! 
RFCOMM is a pretty generic stream. BLE Nordic Uart is a different beast. Making that into a `stream` object will take a bit of hacking.

antonvh, 2021-12-07:
> * `in_waiting()`, `out_waiting()`: these are essential for writing non-blocking code. MicroPython UART has `any()` which is the same as `in_waiting()`.

I would prefer to stick to Micropython standards with `any()` over pyserial nomenclature. If you are open to it. You could do a polyfill for compatibility with older pybricks code. Also the behavior of `read(n)` is different. In pyserial it blocks until you have the required bytes. In Micropython it doesn't block but also doesn't give you more bytes than were in the buffer.

laurensvalk, 2025-12-05:
This has been completed nearly 5 years after @NStrijbosch asked. Thanks for your patience 😄 🎉 

> An identical implementation as for the EV3 would be great!

We ended up taking this approach. Fully backwards compatible with some features added:
- async compatible
- wait_until method to get in sync with a stream
- adjustable baudrate at runtime

closed 2025-12-05 by laurensvalk

antonvh, 2025-12-05:
Great work! In the EV3 implementation of UART we didn't need power. It provided a steady 4v7. PUPDevices have a measly 200mA 3v3. Can the new UART also turn on DC 8v battery power? That would be nice. Or could I create a DC motor and a UART Device on the same port?

jajawunderbar, 2025-12-05:
Finally! :) Thank you very much! I've already implemented it:  [https://youtube.com/shorts/oHpGSh7YTrg?si=X0JbiOqUMaYiC3Fu](https://youtube.com/shorts/oHpGSh7YTrg?si=X0JbiOqUMaYiC3Fu)

antonvh, 2025-12-05:
Both are fine, whatever is least work and cleanest to implement. `battery_on_pin=1` Looks cleanest to me. I suppose 0 is for no power and 2 is for power on pin 2 (and therefore GND on pin 1).

I'm planning to do DC/DC conversion with the power out of battery_on_pin. I don't see myself powering electronics AND controlling a motor. It's rather going to be a motor driver board, where DC PWM will happen outside of SPIKE.

dlech, 2025-12-05:
> I suppose 0 is for no power and 2 is for power on pin 2 (and therefore GND on pin 1).

Exactly. Anything falsy (0, None, etc.) could mean no power, And anything other than 1 or 2 would raise ValueError.

laurensvalk, 2025-12-06:
Either one would be useful. I'd like to have the DCMotor for other purposes too. There are some cable splitters out there that let you run a 3v3 sensor and a DCMotor from the same port.

The one thing holding me back so far (with either approach) is safety.

For the UART case, I suppose you'd really rather keep the power on when the program ends, so your camera (or similar) doesn't stop. Just like we do with the Color Sensor.

The problem with that is that the user may connect something else at that point, and it would continue to get battery power.

On the other hand, I don't think there are any official devices that could be damaged that way since they were designed to handle this. The one thing I don't know is if it's safe to provide a non-100% duty cycle to the color sensor and ultrasonic sensor.

antonvh, 2025-12-06:
> The one thing I don't know is if it's safe to provide a non-100% duty cycle to the color sensor and ultrasonic sensor.

I don't see how it can go wrong. It's just less power, chopping it on and off at 1200Hz. I'm willing to risk both sensors to prove it.

> For the UART case, I suppose you'd really rather keep the power on when the program ends, so your camera (or similar) doesn't stop. Just like we do with the Color Sensor.

Hm. Good point. Yes indeed, we need that. It would confuse users if their cameras or displays turned off.

> The problem with that is that the user may connect something else at that point, and it would continue to get battery power.

I'm actually most worried that the user short-circuits the battery. Or puts 8V on a data line. I'm not sure the hub survives that. Big Disclaimers here? In that regard, the PUP Protocol is not that bad. It turns off the juice with a lack of heartbeat.

jajawunderbar, 2025-12-06:
A second use case. Expresslrs RC receiver over uart: [https://youtube.com/shorts/iwtTh7YCI2o](https://youtube.com/shorts/iwtTh7YCI2o)

antonvh, 2025-12-07:
> For unplugging - if we go with power options in the uart class it wouldn't be unreasonable to require that a message comes in every once in a while to keep the power on.

Hm. That could work. Just any data every second? Can you also measure power consumption per port? A drop in consumption could also signify a disconnect.

> A second use case. Expresslrs RC receiver over uart: https://youtube.com/shorts/iwtTh7YCI2o

Very nice :D I want that VTOL plane too...

ste7anste7an, 2025-12-07:
In the original Mindstorms Robot Inventor it allowed to set the motor power simulateously with using UART on the same port. We used to power our 5V circuit from the 8V motor output.

```
hub.port.A.pwm(100)           # output 8V on M+ motor output
```

If we test that existing original Lego Sensors can deal with the 8V on the motor output, we'll be fine, and it is the responsibility of the user to switch on or off the power on the motor output.