pybricks/support #221: Why I receive an error running: "left_motor.run_time(300,15000, then=Stop.COAST, wait=False)"

RobertoZaffa, 2021-01-19:
**Question**
The first time my  program executes the following statement "left_motor.run_time(300,15000, then=Stop.COAST, wait=False)", everything is fine. Second, I get the following error message:

"Traceback (most recent call last):
  File "/home/robot/MasterMind/main.py", line 307, in <module>
  File "/home/robot/MasterMind/main.py", line 55, in reset_position
OSError: [Errno 1] EPERM: 

The requested operation is not valid in the current state:
--> Check the documentation to for required conditions.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.

----------
Exited with error code 1.
"
Here the whole function:
"
def reset_position(): #porta il carrello in fondo a SX e resetta la posizione
  # Azzera posizione pick motor -  run_until_stalled(speed, then=Stop.COAST, duty_limit=None)
  pick_motor.run_until_stalled(200, then=Stop.COAST, duty_limit=50)
  # reduce the motors power
  left_motor.control.limits(actuation=25)
  right_motor.control.limits(actuation=25)
  # run_time(speed, time, then=Stop.HOLD, wait=True)
**55  left_motor.run_time(300,15000, then=Stop.COAST, wait=False)**
  right_motor.run_time(300,15000, then=Stop.COAST, wait=False)
  # Wait until it is stalled
  while not left_motor.control.stalled() and not right_motor.control.stalled():
    wait(10)
  # Stop the motor (you can also choose brake or hold here)
  left_motor.stop()
  right_motor.stop()
  left_motor.reset_angle(0)
  right_motor.reset_angle(0)
  # restore the motors power
  left_motor.control.limits(actuation=100)
  right_motor.control.limits(actuation=100)
  # Sposta il carrello tutto a sx
  x_motor.run_until_stalled(1200, then=Stop.COAST, duty_limit=50)
  x_motor.reset_angle(0)
"

**Context**
At the beginning of the program I call a function for resetting the motors. During the program I need to repeat this function.

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.
[image]

dlech, 2021-01-19:
In the screenshot, it looks like line 55 is `left_motor.control.limits(actuation=25)` rather than `left_motor.run_time(300,15000, then=Stop.COAST, wait=False)`. 

`EPREM` happens here if the control is busy doing something (running or holding motors). So you need to be sure that the motors are fully stopped before trying to change the limits.

```python
  left_motor.stop()  # or brake()
  right_motor.stop()  # or brake()
  left_motor.control.limits(actuation=25)
  right_motor.control.limits(actuation=25)
```

RobertoZaffa, 2021-01-20:
Sorry if I got confused. 
I added the instructions you suggested and now everything is ok.
The motors were in holding.
Thanks David.

closed 2021-01-20 by RobertoZaffa